1technophile/OpenMQTTGateway #764: 0.9.5-beta - WiFiManager disabled with build flag '-DESPWifiManualSetup=true' for env:nodemcuv2-rf stops sending regular SYStoMQTT messages

DigiH, 2020-09-10:
While trying out OTA updates with 0.9.5-beta, and disabling the WiFiManager for my previously well working nodemcuv2-rf to save a bit of code size, and having the WiFi details in code, with

```
build_flags =
  …
  '-DESPWifiManualSetup=true'
  …
```

I found that the regular SYStoMQTT messages are not appearing any longer; from which I usually get the uptime to be displayed in openHAB.

MQTT messages received by openHAB are interpreted and sent fine though.

Commenting out the WifiManualSetup build flag again and re-uploading via OTA, everything is back to normal with the SYStoMQTT messages.

I don't suppose this is intentional?

1technophile, 2020-09-10:
That's strange, I don't see where it can come from, I will try to reproduce

DigiH, 2020-09-10:
Yes, it puzzled me as well, after nodemcuv2-rf was running for the last few hours here, and my env is virtually identical to the [env:manual-wifi-test] setting just two down, apart from the additional OTA settings. I have reproduced it three times here going back and forth, before logging the issue.

DigiH, 2020-09-11:
> MQTT messages received by openHAB are interpreted and sent fine though.

My bad, it was actually my RF transmitter directly connected to the Raspberry Pi which switched the device, which it normally doesn't reach.

OpenMQTTGateway RF with WiFi Manager disabled seems to be …/LWT
offline

It's connected correctly to the WiFi network, but doesn't send regular SYStoMQTT messages, nor does it process received MQTT messages or RF signals. Commenting out the build flag
`'-DESPWifiManualSetup=true'`
and re-uploading brings everything back to normal.

I've just taken the default [env:nodemcuv2-rf], added the WifiManualSetup build flag, commented out the flash_mode = dout and added the OTA details

```
upload_protocol = espota
upload_port = 192.168.0.51
upload_flags =
  --auth=OTAPASSWORD
  --port=8266
```

Sorry about the earlier confusion about MQTT functionality still working.

closed 2020-09-11 by DigiH