1technophile/OpenMQTTGateway #767: Please add support for Braun Oral-B Bluetooth connected tooth brush

00pi, 2020-09-25:
**Is your feature request related to a problem? Please describe.**
Nice to have for sure, but fun anyway. The protocol is known and the implementation should be straight forward.

**Describe the solution you'd like**
Implement Braun Oral-B Bluetooth connected tooth brush protocol in ZgatewayBT.ino and provide readings
state:
pressure:
time:
mode: 
sector:

the brush is already detected and manufacturerdata provided by the ble presence function
`home/home_presence/OpenMQTTGateway_ESP32_BLE {"id":"5C:31:3E:FE:18:AA","manufacturerdata":"dc0001010502010000011f","rssi":-71,"distance":3.860897}`
(example for my Triumph Professional Care, type 3764)

**Describe alternatives you've considered**
provide code or pull request by myself - not really an option as I am hardly able to code in Visual Basic ;-)

**Additional context**
references
https://github.com/Hypfer/Cybele/blob/1a37a289b33956a3a59863ae8e135535faf1da17/docs/devices/OralBToothbrush.md
https://github.com/Hypfer/Cybele/blob/master/lib/devices/OralBToothbrushDevice.js
https://github.com/rfaelens/domotica/blob/master/mqtt-toothbrush.py
https://github.com/zewelor/bt-mqtt-gateway/blob/master/workers/toothbrush.py
https://github.com/imduffy15/esphome/commit/90c3cb62b37e9ce308ed0c9d510f54ff95c2dfec

Appreciate your consideration!

1technophile, 2022-03-01:
You can take some examples of devices and submit a Pull Request into [Theengs Decoder](https://theengs.github.io/decoder/participate/adding-decoders.html).
Alternatively you can put the toothbrush into the different mode and at the same time copy/paste the information from the application NRF Connect like below:
[image]

The interesting one Are ServiceUUID, ManufacturerData, ServiceData and Name.

Foguet, 2022-03-29:
Thank you very much for the contribution! It seems to work well. I will continue testing to see how it works.

`{"id":"A0:A0:A0:33:33:55","mac_type":0,"name":"Oral-B Toothbrush","rssi":-78,"txpower":0,"brand":"Braun","model":"Toothbrush","model_id":"Braun_BT","state":"idle","mode":"off","pressure":50,"time":0}`

NorthernMan54, 2022-03-30:
@Foguet I have the same toothbrush and am still trying to figure out the use case for this integration.  Besides tracking family members dental cleansing habits, what else?

DigiH, 2022-03-30:
> `{"id":"A0:A0:A0:33:33:55","mac_type":0,"name":"Oral-B Toothbrush","rssi":-78,"txpower":0,"brand":"Braun","model":"Toothbrush","model_id":"Braun_BT","state":"idle","mode":"off","pressure":50,"time":0}`

Thanks for the feedback @Foguet, it also showing that the toothbrush broadcasts a name. Could you also tell us which model your Oral-B toothbrush it is, iO Series, Genius Series, Smart Series ...? 

Additionally it would be great if you could also set

```
'-DpubBLEManufacturerData=true'
'-DpubBLEServiceUUID=true'
'-DpubKnownBLEServiceData=true'
```

in your build_flags for further testing and output reporting as above, to see if we get any further information for defining the decoder.

@NorthernMan54 while I understand your inquisitiveness about the purpose and don't actually have a BT toothbrush myself, I thought that 4 thumbs up for the initial request did show quite an interest ;)

NorthernMan54, 2022-03-30:
@DigiH I was racking my brain about potential automation triggers based on usage.  Set brushing your teeth mood lighting, or play an audio clip ( https://www.youtube.com/watch?v=wCio_xVlgQ0 )

DigiH, 2022-03-30:
> mood lighting, or play an audio

While I have both in the bathroom, they are for spa time not for brushing teeth ;)

I really don't know either, maybe a bit like tracking weight with the scales, tracking that whole family brushed their teeth properly all month or such. 

Or a bit like my Mi Band activity heart rate implementation showing a different floor lamp colour depending if I am below or above the desired aerobic heart rate without having to constantly raise my arm to have a look on the Mi Band when on the treadmill. Maybe this could be implemented similarly, giving light/audio cues when you should change over to the next sector or when your done cleaning all sectors properly?!?

00pi, 2022-03-30:
@DigiH 
Thanks a lot for your work! I was able to compile for an ESP-32 and got the following results for my two brushes:

Oral-B Triumph Professional Care
Type 3764
`{"id":"5C:31:3E:FE:18:AA","mac_type":0,"rssi":-59,"brand":"Braun","model":"Toothbrush","model_id":"Braun_BT","state":"idle","mode":"daily clean","sector":"sector 1","pressure":1,"time":0}`

Oral-B Genius X
Type 3771
`{"id":"D8:71:4D:E8:A5:35","mac_type":0,"name":"Oral-B Toothbrush","rssi":-50,"txpower":0,"brand":"Braun","model":"Toothbrush","model_id":"Braun_BT","state":"idle","mode":"off","pressure":50,"time":0}
`

I see updates of the readings approx. every 60 seconds. Is there a way to increase this frequency? Maybe only temporary on a condition eg. "state": "running" ?

DigiH, 2022-03-30:
Thanks for the feedback @00pi 

For further testing it would also be great if you could add

```
'-DpubBLEManufacturerData=true'
'-DpubBLEServiceUUID=true'
'-DpubKnownBLEServiceData=true'
```

to your build flags if you're using platformio. If you use the Arduino IDE you can set these to true in 

https://github.com/1technophile/OpenMQTTGateway/blob/development/main/config_BT.h#L107

> I see updates of the readings approx. every 60 seconds. Is there a way to increase this frequency? Maybe only temporary on a condition eg. "state": "running" ?

While I don't know how often the toothbrushes broadcast their data, you can also reduce the time between BT scanning with

`'-DTimeBtwRead=15000'`

(platformio) or for Arduino IDE

https://github.com/1technophile/OpenMQTTGateway/blob/development/main/config_BT.h#L75

00pi, 2022-03-30:
@DigiH  I added the three flags and get the following results
`{"id":"5C:31:3E:FE:18:AA","mac_type":0,"manufacturerdata":"dc0001010502010000013f","rssi":-75,"brand":"Braun","model":"Toothbrush","model_id":"Braun_BT","state":"idle","mode":"daily clean","sector":"sector 1","pressure":1,"time":0}`

`{"id":"D8:71:4D:E8:A5:35","mac_type":0,"name":"Oral-B Toothbrush","manufacturerdata":"dc000471050332002201020d04","rssi":-74,"txpower":0,"brand":"Braun","model":"Toothbrush","model_id":"Braun_BT","state":"running","mode":"daily clean","sector":"sector 1","pressure":50,"time":34}`

DigiH, 2022-03-30:
Thanks a lot @00pi, this will help to improve and refine the decoder.

DigiH, 2022-03-31:
Latest Oral-B decoder with changes and fixes for testing at

`decoder = https://github.com/DigiH/decoder#Oral-B`

joselito11, 2022-03-31:
@DigiH 
Oral-B 3754
`{"id":"A8:1B:6A:79:27:53","mac_type":0,"rssi":-59,"brand":"Oral-B","model":"BT Toothbrush","model_id":"Oral-B_BT","state":"idle","mode":"daily clean","sector":"sector 1","pressure":32,"time":5}`

00pi, 2022-04-01:
@DigiH Thanks again for your work! Using the latest 
`decoder = https://github.com/DigiH/decoder#Oral-B`
I ran both brushes for some time with the scan interval set to 1 second. Please find the results below. Compared to the Triumph Professional, the Genius X shows the sector you are currently brushing, but that is already implemented. 

I've found some more information on the protocol, maybe this is of any help?
`https://github.com/wise86-android/OralBlue_python/blob/master/Protocol.md`

`omg/OpenMQTTGateway_ESP32_BLE/BTtoMQTT/5C313EFE18AA {"id":"5C:31:3E:FE:18:AA","mac_type":0,"manufacturerdata":"dc0001010502010000012f","rssi":-61,"brand":"Oral-B","model":"BT Toothbrush","model_id":"Oral-B_BT","state":"idle","mode":"daily clean","pressure":1,"time":0}`

`omg/OpenMQTTGateway_ESP32_BLE/BTtoMQTT/5C313EFE18AA {"id":"5C:31:3E:FE:18:AA","mac_type":0,"manufacturerdata":"dc0001010502010000012f","rssi":-58,"brand":"Oral-B","model":"BT Toothbrush","model_id":"Oral-B_BT","state":"idle","mode":"daily clean","pressure":1,"time":0}`

`omg/OpenMQTTGateway_ESP32_BLE/BTtoMQTT/5C313EFE18AA {"id":"5C:31:3E:FE:18:AA","mac_type":0,"manufacturerdata":"dc0001010502010000012f","rssi":-55,"brand":"Oral-B","model":"BT Toothbrush","model_id":"Oral-B_BT","state":"idle","mode":"daily clean","pressure":1,"time":0}`

omg/OpenMQTTGateway_ESP32_BLE/BTtoMQTT/D8714DE8A535 {"id":"D8:71:4D:E8:A5:35","mac_type":0,"name":"Oral-B Toothbrush","manufacturerdata":"dc0004710502360000000f0004","rssi":-60,"txpower":0,"brand":"Oral-B","model":"BT Toothbrush","model_id":"Oral-B_BT","state":"idle","mode":"off","pressure":54,"time":0}

omg/OpenMQTTGateway_ESP32_BLE/BTtoMQTT/D8714DE8A535 {"id":"D8:71:4D:E8:A5:35","mac_type":0,"name":"Oral-B Toothbrush","manufacturerdata":"dc0004710502320000000f0004","rssi":-61,"txpower":0,"brand":"Oral-B","model":"BT Toothbrush","model_id":"Oral-B_BT","state":"idle","mode":"off","pressure":50,"time":0}

omg/OpenMQTTGateway_ESP32_BLE/BTtoMQTT/D8714DE8A535 {"id":"D8:71:4D:E8:A5:35","mac_type":0,"name":"Oral-B Toothbrush","manufacturerdata":"dc0004710502320000000f0004","rssi":-67,"txpower":0,"brand":"Oral-B","model":"BT Toothbrush","model_id":"Oral-B_BT","state":"idle","mode":"off","pressure":50,"time":0}

omg/OpenMQTTGateway_ESP32_BLE/BTtoMQTT/D8714DE8A535 {"id":"D8:71:4D:E8:A5:35","mac_type":0,"name":"Oral-B Toothbrush","manufacturerdata":"dc0004710502320000000f0004","rssi":-84,"txpower":0,"brand":"Oral-B","model":"BT Toothbrush","model_id":"Oral-B_BT","state":"idle","mode":"off","pressure":50,"time":0}

omg/OpenMQTTGateway_ESP32_BLE/BTtoMQTT/D8714DE8A535 {"id":"D8:71:4D:E8:A5:35","mac_type":0,"name":"Oral-B Toothbrush","manufacturerdata":"dc00047105033a000001010004","rssi":-78,"txpower":0,"brand":"Oral-B","model":"BT Toothbrush","model_id":"Oral-B_BT","state":"running","mode":"daily clean","sector":"sector 1","pressure":58,"time":0}

omg/OpenMQTTGateway_ESP32_BLE/BTtoMQTT/D8714DE8A535 {"id":"D8:71:4D:E8:A5:35","mac_type":0,"name":"Oral-B Toothbrush","manufacturerdata":"dc000471050332000801011a04","rssi":-61,"txpower":0,"brand":"Oral-B","model":"BT Toothbrush","model_id":"Oral-B_BT","state":"running","mode":"daily clean","sector":"sector 1","pressure":50,"time":8}

omg/OpenMQTTGateway_ESP32_BLE/BTtoMQTT/D8714DE8A535 {"id":"D8:71:4D:E8:A5:35","mac_type":0,"name":"Oral-B Toothbrush","manufacturerdata":"dc000471050332001301013f04","rssi":-61,"txpower":0,"brand":"Oral-B","model":"BT Toothbrush","model_id":"Oral-B_BT","state":"running","mode":"daily clean","sector":"sector 1","pressure":50,"time":19}

omg/OpenMQTTGateway_ESP32_BLE/BTtoMQTT/D8714DE8A535 {"id":"D8:71:4D:E8:A5:35","mac_type":0,"name":"Oral-B Toothbrush","manufacturerdata":"dc000471050332001501014604","rssi":-61,"txpower":0,"brand":"Oral-B","model":"BT Toothbrush","model_id":"Oral-B_BT","state":"running","mode":"daily clean","sector":"sector 1","pressure":50,"time":21}

omg/OpenMQTTGateway_ESP32_BLE/BTtoMQTT/D8714DE8A535 {"id":"D8:71:4D:E8:A5:35","mac_type":0,"name":"Oral-B Toothbrush","manufacturerdata":"dc000471050332001e01020004","rssi":-61,"txpower":0,"brand":"Oral-B","model":"BT Toothbrush","model_id":"Oral-B_BT","state":"running","mode":"daily clean","sector":"sector 2","pressure":50,"time":30}

omg/OpenMQTTGateway_ESP32_BLE/BTtoMQTT/D8714DE8A535 {"id":"D8:71:4D:E8:A5:35","mac_type":0,"name":"Oral-B Toothbrush","manufacturerdata":"dc000471050332002901022404","rssi":-73,"txpower":0,"brand":"Oral-B","model":"BT Toothbrush","model_id":"Oral-B_BT","state":"running","mode":"daily clean","sector":"sector 2","pressure":50,"time":41}

omg/OpenMQTTGateway_ESP32_BLE/BTtoMQTT/D8714DE8A535 {"id":"D8:71:4D:E8:A5:35","mac_type":0,"name":"Oral-B Toothbrush","manufacturerdata":"dc000471050332003401024904","rssi":-60,"txpower":0,"brand":"Oral-B","model":"BT Toothbrush","model_id":"Oral-B_BT","state":"running","mode":"daily clean","sector":"sector 2","pressure":50,"time":52}

omg/OpenMQTTGateway_ESP32_BLE/BTtoMQTT/D8714DE8A535 {"id":"D8:71:4D:E8:A5:35","mac_type":0,"name":"Oral-B Toothbrush","manufacturerdata":"dc000471050332010301030a04","rssi":-61,"txpower":0,"brand":"Oral-B","model":"BT Toothbrush","model_id":"Oral-B_BT","state":"running","mode":"daily clean","sector":"sector 3","pressure":50,"time":63}

omg/OpenMQTTGateway_ESP32_BLE/BTtoMQTT/D8714DE8A535 {"id":"D8:71:4D:E8:A5:35","mac_type":0,"name":"Oral-B Toothbrush","manufacturerdata":"dc000471050332010301030a04","rssi":-65,"txpower":0,"brand":"Oral-B","model":"BT Toothbrush","model_id":"Oral-B_BT","state":"ru [... cut ...]
`omg/OpenMQTTGateway_ESP32_BLE/BTtoMQTT/D8714DE8A535 {"id":"D8:71:4D:E8:A5:35","mac_type":0,"name":"Oral-B Toothbrush","manufacturerdata":"dc000471050232013400044904","rssi":-61,"txpower":0,"brand":"Oral-B","model":"BT Toothbrush","model_id":"Oral-B_BT","state":"idle","mode":"off","sector":"sector 4","pressure":50,"time":112}`

`omg/OpenMQTTGateway_ESP32_BLE/BTtoMQTT/D8714DE8A535 {"id":"D8:71:4D:E8:A5:35","mac_type":0,"name":"Oral-B Toothbrush","manufacturerdata":"dc000471050232013400044904","rssi":-62,"txpower":0,"brand":"Oral-B","model":"BT Toothbrush","model_id":"Oral-B_BT","state":"idle","mode":"off","sector":"sector 4","pressure":50,"time":112}`

DigiH, 2022-04-01:
> Please find the results below.

Thanks a lot @00pi, that gives me new information and results to go through for further refinement.

DigiH, 2022-04-02:
@00pi - updated decoder at the same URL

`decoder = https://github.com/DigiH/decoder#Oral-B`

just needs rebuilding and uploading.

If you could test the following:

• The **Triumph Professional** should also show sectors now. Are they correct in usage?
• **pressure** is now a string **"normal"** or **"high"**, which should correspond with the pressure light indicator on the toothbrush. Open for discussion if this should not show at all when **"normal"**, only publishing when **"pressure":"high"**.

Thanks