1technophile/OpenMQTTGateway #771: Sketch too big with only ZgatewayRF and ZmqttDiscovery uncommented

welcoMattic, 2020-10-02:
Before submitting a problem please check the troubleshooting section
https://docs.openmqttgateway.com/upload/troubleshoot.html

**Describe the bug**
I hava geniune Arduino Uno and Arduino IDE tells me:

```Sketch uses 32288 bytes (100%) of program storage space. Maximum is 32256 bytes.
Global variables use 1514 bytes (73%) of dynamic memory, leaving 534 bytes for local variables. Maximum is 2048 bytes.
text section exceeds available space in board
Sketch too big; see http://www.arduino.cc/en/Guide/Troubleshooting#size for tips on reducing it.
Error compiling for board Arduino Uno.```

With the following User_config.h
<details>
  <summary>Expand</summary>
```
/*  
  OpenMQTTGateway  - ESP8266 or Arduino program for home automation 

   Act as a wifi or ethernet gateway between your 433mhz/infrared IR signal  and a MQTT broker 
   Send and receiving command by MQTT
 
  This program enables to:
 - receive MQTT data from a topic and send signals corresponding to the received MQTT data
 - publish MQTT data to a different topic related to received signals
  
    Copyright: (c)Florian ROBERT
  
    This file is part of OpenMQTTGateway.
    
    OpenMQTTGateway is free software: you can redistribute it and/or modify
    it under the terms of the GNU General Public License as published by
    the Free Software Foundation, either version 3 of the License, or
    (at your option) any later version.

    OpenMQTTGateway is distributed in the hope that it will be useful,
    but WITHOUT ANY WARRANTY; without even the implied warranty of
    MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
    GNU General Public License for more details.

    You should have received a copy of the GNU General Public License
    along with this program.  If not, see <http://www.gnu.org/licenses/>.
*/
#ifndef user_config_h
#define user_config_h
/*-------------------VERSION----------------------*/
#ifndef OMG_VERSION
#  define OMG_VERSION "v0.9.5"
#endif

/*-------------CONFIGURE WIFIMANAGER-------------(only ESP8266 & SONOFF RFBridge)*/
/*
 * The following parameters are set during the WifiManager setup process:
 * - wifi_ssid
 * - wifi_password
 * - mqtt_user
 * - mqtt_pass
 * - mqtt_server
 * - mqtt_port
 * 
 * To completely disable WifiManager, define ESPWifiManualSetup.
 * If you do so, please don't forget to set these variables before compiling
 * 
 * Otherwise you can provide these credentials on the web interface after connecting 
 * to the access point with your password (SSID: WifiManager_ssid, password: WifiManager_password)
 */
/*-------------DEFINE GATEWAY NAME BELOW IT CAN ALSO BE DEFINED IN platformio.ini----------------*/
#ifndef Gateway_Name
#  define Gateway_Name "OpenMQTTGateway"
#endif
[... 319 lines cut ...]
#define TimeBetweenReadingSYS        120 // time between (s) system readings (like memory)
#define TimeLedON                    0.5 // time LED are ON
#define InitialMQTTConnectionTimeout 10 // time estimated (s) before the board is connected to MQTT
#define subjectSYStoMQTT             "/SYStoMQTT"
#define subjectMQTTtoSYSset          "/commands/MQTTtoSYS/config"

/*-------------------DEFINE LOG LEVEL----------------------*/
#define LOG_LEVEL LOG_LEVEL_NOTICE

#endif
```
</details>

**To Reproduce**
Build the last OMG version with my User_config.h

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment (please complete the following information):**
* OpenMQTTGateway version v0.9.5
* Last versions of all libraries (updated before posting this issue)

1technophile, 2020-10-02:
Hi,

It is possible that the Uno + RF doesn't support the MQTT discovery, could you try without?

welcoMattic, 2020-10-02:
It works perfectly without the discovery, so you are right :)

1technophile, 2020-10-02:
> Is there any arduino board that supports RF + discovery?

Arduino MEGA will support both and more if you need

closed 2020-10-03 by welcoMattic