1technophile/OpenMQTTGateway #775: Adding RS232?

DieKatzchen, 2020-10-10:
**Is your feature request related to a problem? Please describe.**
I need to control my projector and IR doesn't work, but it has a serial port and a documented protocol

**Describe the solution you'd like**
Using an Off-The-Shelf TTL-to-RS232 adapter, could integrate the serial port with existing system 

**Describe alternatives you've considered**
https://github.com/gm-stack/esp-mqtt-serial, is what I'm currently using but all settings are hard-coded and require reflashing to accommodate any network, device, or configuration changes

DieKatzchen, 2020-10-10:
And by "currently using" I mean trying to get to work, but it only has tx on the serial side and sub on the MQTT, so I'm yelling commands into the void with no feedback as to what I might have done wrong. I could of course try to fix it, but I'd much rather the functionality be added to this so that I have a unified control system.

1technophile, 2020-10-12:
Do you feel confortable in adding it as a pull request ?

DieKatzchen, 2020-10-12:
I could probably muddle through it. Let me fork your code and see what I come up with.

DieKatzchen, 2020-10-13:
#780 I've made a pull request, but it didn't pass one of the tests. Not sure how to read it either. Any advice? I haven't actually done a PR since college, and we didn't have these fancy code checks.

DieKatzchen, 2020-10-13:
Actually, I've looked at it again and I think it makes sense to me now. It seems to mostly be complaining about my indentation on #define and #ifdef statements. I've learned something new.

closed 2020-10-20 by 1technophile