1technophile/OpenMQTTGateway #783: Kalman filter on RSSI

joshuakoh1, 2020-10-16:
RSSI values can fluctuate between -60 to -80 even while stationary. Consider applying Kalman filter to smoothen out data points.

https://download.atlantis-press.com/article/25858154.pdf

ozett, 2023-01-03:
[image]
https://www.sciencedirect.com/science/article/pii/B978012824536100006X

ozett, 2023-01-03:
| **PDF-Document, local version:** |
| :---: |
[EasyChair-Preprint-1198.pdf](https://github.com/1technophile/OpenMQTTGateway/files/10335624/EasyChair-Preprint-1198.pdf)
| **Preview** |
[image]

ozett, 2023-01-03:
filtering seems to be only part of a solution for more accuray on RSSI

> The main problem with RSSI is that it varies too much, over time, to be used to accurately calculate distance. The direction also isn’t known when there’s only one beacon and one detector. The varying RSSI, even when nothing is moving, is caused by the Bluetooth radio signals that are reflected, deflected by physical obstacles and interfered with by other devices using similar radio frequencies. Physical factors such as the room, the beacon not uniformly emitting across a range of 360 degrees, walls, other items or even people can affect the received signal strength. How the user holds a detecting phone can affect the effectiveness of the antenna which in turn affects the signal strength

https://www.beaconzone.co.uk/blog/ibeacon-microlocation-accuracy/

>If you want to view the research paper you need to [download all the papers from the conference (zip)](https://www.sigmobile.org/mobisys/2019/MobiSys19-final-pdfs.zip) and extract p558-uranoA.pdf.

maybe only AI can help ?
https://www.beaconzone.co.uk/blog/using-ai-machine-learning-on-bluetooth-rssi-to-obtain-location/

ozett, 2023-01-03:
maybe there are some solutions under way?
https://github.com/ESPresense/ad-espresense-ips

>attempts to solve indoor position (x,y,z) with multiple ESPresense stations using multilateralization.

ozett, 2023-01-03:
best accuray = 1,5 m ?

>Hence for systems that use signal processing, trilateration and calibration tend to achieve accuracies of about 1.5m within a shorter range confined space and 5m at the longer distances.

https://www.beaconzone.co.uk/blog/ibeacon-microlocation-accuracy/