1technophile/OpenMQTTGateway #784: After experiencing network issues, OMG goes into infinite connect-disconnect loop

melyux, 2020-10-19:
**Describe the bug**
I refreshed the internal DNS settings for my router, which caused a few minutes of network instability. Other clients all recovered, but my OpenMQTTGateway box went into an infinite loop of:

1. Connecting to the Wi-Fi
2. Sending the version to the `/version` and `online` to the `/LWT` topics.
3. Sending `offline` to the `/LWT` topic within a couple of seconds.
4. Disconnecting from the Wi-Fi.

* Sending a restart command to the `/commands/MQTTtoSYS/config` topic doesn't seem to fix it.
* Restarting the MQTT broker doesn't seem to fix it.
* Forcing a reconnect to another access point doesn't seem to fix it.
After forcing this kind of reconnect twice, the OMG box stopped reconnecting and played dead. No access point was generated at any point.

The only way to get it back to normal seems to be to cut the power or reach the box (placed high out of reach) and hit the physical reset button. Not ideal

**To Reproduce**
Steps to reproduce the behavior: Not really sure... the access points were left without a routing mechanism for a few minutes, and they switched their Wi-Fi channels around several times within a short period of time.

**Expected behavior**
OMG should recover from any sort of network issue once the network itself recovers.

**Screenshots**
[image]
**Environment (please complete the following information):**
* OpenMQTTGateway version used (V0.9.3, 0.8, development): Development
* Library version related to the problem you have: N/A?

**Additional context**
Add any other context about the problem here.

* You should not have a compilation error if you use the versions of the libraries linked into the libraries folder, this badges show you the state of the compilation 
[image]
* If you are not sure this is a bug or an enhancement post your question to the forum below
[image]

1technophile, 2020-10-20:
Hi,

When you say Development version used, do you have an idea of the date on which you picked up the code?

melyux, 2020-10-21:
Hey, it was only a few commits back, sometime in September. I finally got to the box and checked the serial monitor, and saw it was trying to connect to the Wi-Fi and failing. It still failed to connect after I reset it using the button, and even after turning off and on the power. The AP wasn't showing it even trying to connect.

After I turned the power off and on, it started broadcasting a OpenMQTTGateway access point. I connected to it and tried setting the Wi-Fi network name and password again (even choosing from the list of scanned networks), but it still didn't work. I then re-flashed OMG to it, twice, and it still didn't connect.

Finally, I used the OpenMQTTGateway access point page to reset the Wi-Fi settings, and then entered the Wi-Fi settings... this time it instantly connected. This is a Wemos D1 Mini by the way. My other Wemos D1 Mini running another project that also uses Wifi Manager didn't stop working though. The network settings weren't changed at all, but even if they were, I think the ESP should recover from that as long as the network name and password haven't changed. Maybe OMG's recovery code wrote to wherever the Wi-Fi configuration is stored some settings that made it impossible to connect once the Wi-Fi came back up?

Here are some logs from when it was not connecting:

```
04:14:19.050 -> W: failed to connect and hit timeout
04:14:19.050 -> 
04:14:19.050 ->  ets Jan  8 2013,rst cause:2, boot mode:(3,7)
04:14:19.050 -> 
04:14:19.050 -> load 0x4010f000, len 3584, room 16 
04:14:19.050 -> tail 0
04:14:19.050 -> chksum 0xb0
04:14:19.050 -> csum 0xb0
04:14:19.050 -> v2843a5ac
04:14:19.050 -> ~ld
04:14:19.050 -> N:⸮⸮T: mounting FS...
04:14:19.050 -> T: mounted file system
04:14:19.050 -> T: reading config file
04:14:19.050 -> T: opened config file
04:14:19.050 -> {"mqtt_server":"***","mqtt_port":"1883","mqtt_user":"***","mqtt_pass":"***","mqtt_topic":"***","gateway_name":"OpenMQTTGateway"}T: 
04:14:19.050 -> parsed json
04:14:19.050 -> *WM: [3] allocating params bytes: 20
04:14:19.050 -> *WM: [2] Added Parameter: server
04:14:19.050 -> *WM: [2] Added Parameter: port
04:14:19.050 -> *WM: [2] Added Parameter: user
04:14:19.050 -> *WM: [2] Added Parameter: pass
04:14:19.050 -> *WM: [2] Added Parameter: name
04:14:19.050 -> *WM: [3] Updated _max_params: 10
04:14:19.050 -> *WM: [3] re-allocating params bytes: 40
04:14:19.050 -> *WM: [2] Added Parameter: topic
04:15:19.251 -> N: Attempting Wifi connection with saved AP: 0
04:16:19.456 -> N: Attempting Wifi connection with saved AP: 1
04:17:18.793 -> N: Attempting Wifi connection with saved AP: 2
04:18:14.808 -> N: Attempting Wifi connection with saved AP: 3
04:19:15.008 -> N: Attempting Wifi connection with saved AP: 4
04:20:15.209 -> N: Connect your phone to WIFI AP: OpenMQTTGateway with PWD: ...
```

tita2000, 2020-11-06:
Hello, due to the fact that my gateways (3) are distributed in my house, i can not provide any logs. But I can confirm, that when they loose WLAN connection (e.g. restart of repeater) they go offline (MQTT topic LWT = offline). When I restart them they are fine again. I'm using the BLE gateway firmware to collect temperature data.

Regards Timo

1technophile, 2020-11-06:
Hi, and if we remove the repeater from the equation (switch it off), and connect the gateway to the router, if you restart the router, do they reconnects?

melyux, 2020-11-25:
Mine don't reconnect even if the router is restarted.

I had a repeat of the same issue today when there was a power outage and the Wi-Fi went down. After the Wi-Fi was restored, OMG kept connecting and disconnecting from it every few seconds. It was even reconnecting to the MQTT broker in the split second it was connected to the Wi-Fi, but then would immediately disconnect. I had to climb up to the ledge and manually reset the power to get it working again.

I got logs this time while it was doing this:

```
 Network disconnected:
 Network disconnected:
 Network disconnected:
 Network disconnected:
```

Just thousands of lines of this over and over again. And after that, it switched to this:

```
W: wifi
W: disconnection_handlinW: wifi
W: disconnection_handlinW: wifi
W: disconnection_handlinW: wifi
W: disconnection_handlinW: wifi
W: disconnection_handlinW: wifi
W: disconnection_handlinW: wifi
W: disconnection_handlinW: wifi
W: disconnection_handlinW: wifi
W: disconnection_handlinW: wifi
W: disconnection_handlinW: wifi
W: disconnection_handlinW: wifi
W: disconnection_handlinW: wifi
W: disconnection_handlinW: wifi
W: disconnection_handlinW: wifi
W: disconnection_handlinW: wifi
W: disconnection_handlinW: wifi
W: disconnection_handlinW: wifi
W: disconnection_handlinW: wifi
W: disconnection_handlinW: wifi
W: disconnection_handlinW: wifi
W: disconnection_handlinW: wifi
W: disconnection_handlinW: wifi
W: disconnection_handlinW: wifi
W: disconnection_handlinW: wifi
W: disconnection_handlinW: wifi
W: disconnection_handlinW: wifi
W: disconnection_handlinW: wifi
W: disconnection_handlinW: wifi
W: disconnection_handling, failed 3 times
W: Attempt to g, failed 3 times
W: Attempt to g, failed 3 times
W: Attempt to g, failed 3 times
W: Attempt to g, failed 3 times
W: Attempt to g, failed 3 times
W: Attempt to g, failed 3 times
W: Attempt to g, failed 3 times
W: Attempt to g, failed 3 times
W: Attempt to g, failed 3 times
W: Attempt to g, failed 3 times
W: Attempt to g, failed 3 times
W: Attempt to g, failed 3 times
W: Attempt to g, failed 3 times
W: Attempt to g, failed 3 times
W: Attempt to g, failed 3 times
W: Attempt to g, failed 3 times
W: Attempt to g, failed 3 times
W: Attempt to g, failed 3 times
W: Attempt to g, failed 3 times
W: Attempt to g, failed 3 times
W: Attempt to g, failed 3 times
W: Attempt to g, failed 3 times
W: Attempt to g, failed 3 times
W: Attempt to g, failed 3 times
[... 148 lines cut ...]
W: Network disconnected:
W: wifi
W: disconnection_handling, failed 1 times
W: Attempt to reinit wifi: 2
W: ESP8266: Forcing to wifi 2
MQT⸮ѥ⸮⸮...
N: Connected to broker
W: Network disconnected:
W: wifi
W: disconnection_handling, failed 1 times
W: Attempt to reinit wifi: 2
W: ESP8266: Forcing to wifi 2
W:jEonnecin...
N: Connected to broker
W: Network disconnected:
W: wifi
W: disconnection_handling, failed 1 times
W: Attempt to reinit wifi: 2
W: ESP8266: Forcing to wifi 2
WEjEQQ connection...
N: Connected to broker
W: Network disconnected:
```

You can see it reconnecting to the MQTT broker and even providing BME280 sensor data every once in a while over MQTT there, but it's in an infinite connect-disconnect loop.

1technophile, 2020-11-29:
Seems that your router doesn't like to change the protocol, currently, if the wifi is not available we try to change the protocol (BGN) to check if one is working.
The issue is, in your case G works but disconnect just after.

Could you give me the brand and model of your router?

Note that I have changed the reconnection algorithm so as to browse the different protocols available every 100 attempts:
https://github.com/1technophile/OpenMQTTGateway/tree/recursive-protocol-change

This should avoid being stuck into one protocol.

melyux, 2020-11-29:
Sure, it’s a UniFi AP. Very common

Hopefully that fixes it. I don’t think I’ve seen this protocol switching code on any other project. I would think the ESP stack would take care of that, like it seems to for all the other projects I’ve tried on this chip. Was it implemented to fix someone’s corner case?

1technophile, 2020-11-29:
You can find the history below:
https://github.com/1technophile/OpenMQTTGateway/issues/563

The mechanism was inspired by ESPEasy one

melyux, 2020-12-03:
I'm still on the master branch, but I see OMG connected to `g` right now with no problems. So `g` seems to work fine, so not sure what the problem is.

Is it possible to have it upgrade the connection occasionally to make sure it's using `n` if available? It's getting stuck at `g` right now even though `n` is available.

1technophile, 2020-12-04:
> Is it possible to have it upgrade the connection occasionally to make sure it's using n if available? It's getting stuck at g right now even though n is available.

Yes, I agree we should have something like this

melyux, 2020-12-06:
Had this problem again, except this time, the Wi-Fi didn't go down at all, only the computer running the MQTT broker. Hmmm...

1technophile, 2020-12-06:
If you are still on the master lr development branch. Could you test the branch done especialy for it  ? 
https://github.com/1technophile/OpenMQTTGateway/tree/recursive-protocol-change

melyux, 2020-12-07:
Testing it now

melyux, 2020-12-07:
I noticed that there's a way to force a Wi-Fi mode in `User_config.h` using `wifiProtocol`, but it's overwritten whenever there's a connection problem. Maybe that overwriting should be disabled if the user specified `wifiProtocol`.

Further, maybe `wifiProtocol` should be made into a macro. It's one of the only non-macros defined in `User_config.h`, making it impossible to set using the normal Platform.io configuration `*_env.ini* file.

melyux, 2020-12-24:
I'm on the test branch. I turned off my MQTT broker for about 15 minutes. Upon restore, I saw that OMG was in a loop now where it would disconnect from the broker every 3 minutes. It would be in 11g mode the whole time though.

1technophile, 2020-12-24:
> I'm on the test branch. I turned off my MQTT broker for about 15 minutes. Upon restore, I saw that OMG was in a loop now where it would disconnect from the broker every 3-4 minutes, and reconnect after 8 seconds. It would be in 11g mode the whole time though.

Maybe I should enable to deactivate the automatic protocol browsing (bgn...), would you be interested in testing this?

melyux, 2020-12-24:
I’d definitely be interested. But it’s always on G anyway, both in normal operation and when it’s in this mode. Do you think it’s switching around from G to something else (B?) and that’s why it disconnects? I also found it weird that it never uses N mode, only G.

melyux, 2020-12-30:
Turned off the Wi-Fi, and upon restore it was back in the rapid connect-disconnect loop. So there’s that ~3 minute loop when the MQTT broker goes down, and also this ~10 second loop when Wi-Fi goes down.

1technophile, 2021-01-02:
Could you try to replace the lines begining by
`wifiProtocol = XXXXX`
by
`    wifiProtocol = 0;`

This way you will not have any protocol changes, and we will see if you have a better reconnection stability

melyux, 2021-01-09:
@1technophile Tried that and restarted my router and MQTT broker. Unfortunately, it still went into a connect-disconnect loop.

melyux, 2021-01-10:
Strangely, it disconnected at 11:57 pm, stayed disconnected for a few minutes, and reconnected at 12:05 am or so, and stayed connected. I wonder if it would have done this before, or if there's something about the midnight boundary that helps it.

1technophile, 2021-01-26:
The last test that we could do is to remove all the code related to wifiProtocol change, so as to do that could you remove all the code into  `disconnection_handling` method and see if you get a better behavior

csiki2, 2021-02-05:
Hi,

I have seen similar issues with the BLE gateway esp32 (rooter: Asus RT-AC65P).
I left the wifiProtocol (since it's working normally), but I practically disabled the whole protocol change code:
<pre>
#  define ATTEMPTS_BEFORE_BG 999999999 // Number of wifi connection attempts before going to BG protocol - was 10
</pre>
Still reconnects time to time (every 3-6 minutes), but doesn't seem to stuck in disconnected state.
50k uptime so far, I will check it regularly.

Regards,
  Csiki

1technophile, 2021-02-05:
If you are using `ESPWifiManualSetup` I found an issue, this is corrected on the development branch.

The retry mechanism was launched too fast without letting the time of wifi.begin to save the credentials in flash.

1technophile, 2021-03-26:
Retry mechanism removed to avoid this kind of issue

closed 2021-03-26 by 1technophile