1technophile/OpenMQTTGateway #787: Mi Scale not properly reporting wieght when set to LBS

github768484, 2020-10-22:
firmware: OpenMQTTGateway v0.9.5
device: Xiaomi scale

The value is wrong when the scale is set to lbs via a Mi Fit app. i.e. If the scale is set the Kg, my lbs conversion is accurate. if the scale is set to lbs, the raw (non calculated) value is not Kg nor is it lbs.

Here's the payload at 30lbs (Scale is set to Lbs)

{"id":"C8:47:8C:D1:0E:BE","name":"MI SCALE2","manufacturerdata":"5701c8478cd10ebe","rssi":-59,"distance":1.01076,"model":"XMTZC04HM","weight":15}

Here's the payload at 195.4lbs (Scale is set to Lbs)

{"id":"C8:47:8C:D1:0E:BE","name":"MI SCALE2","manufacturerdata":"5701c8478cd10ebe","rssi":-82,"distance":11.4949,"model":"XMTZC04HM","weight":97.7}

Here's the payload at 89.7kg (Scale is set to Kg. When set to Kg the payload is accurate)

{"id":"C8:47:8C:D1:0E:BE","name":"MI SCALE2","manufacturerdata":"5701c8478cd10ebe","rssi":-91,"distance":25.51913,"model":"XMTZC04HM","weight":89.7}

michapr, 2020-11-13:
Check this here:
https://github.com/oliexdev/openScale/wiki/Xiaomi-Bluetooth-Mi-Scale

Byte 0 seems to define the lbs setting, could be checked...

1technophile, 2021-02-11:
Done on PR linked, if you could test it guys it would be great

github768484, 2021-02-12:
I'll give it a shot this weekend.

github768484, 2021-02-15:
Would you mind linking an Arduino IDE package for an ESP32dev? I attempted to update the ZgatewayBT.ino alone, but I get errors when attempting to compile. 
`'BLE_CNCT_TIMEOUT' was not declared in this scope`

1technophile, 2021-02-19:
[esp32dev-ble-test.zip](https://github.com/1technophile/OpenMQTTGateway/files/6013036/esp32dev-ble-test.zip)
Here is the bin files

1technophile, 2021-02-19:
[esp32dev-ble.zip](https://github.com/1technophile/OpenMQTTGateway/files/6013038/esp32dev-ble.zip)
Here are the libraries

github768484, 2021-02-20:
This is a confirmation that imperial (LBS) weight is reading correctly. Thank you.

closed 2021-02-26 by 1technophile