1technophile/OpenMQTTGateway #788: Add iNode TEMPERATURE MEASUREMENT support

emwu88, 2020-10-22:
Witam,
w nawiązaniu do:
[(https://github.com/1technophile/OpenMQTTGateway/pull/755)]
jest szansa dodać czujniki z serii "iNode Care Sensor XX"
Przykład dla Raspberry podaje producent:
`#!/bin/bash
packet=""
capturing=""
count=0
kuch=""
dom=""
sal=""
gab=""
zew=""
wyslij () {
	#echo $packet >> /home/pi/d.txt
	local temp=`echo $packet | awk '{print $29$28}'`
	local tempd=`echo "ibase=16; $temp"|bc`
	local tempr=`echo "scale=2; ($tempd*175.72*4)/65536-46.85"|bc -l`
	wget -O- "192.168.1.140/temp.php?nrc=$1&typ=1&value=$tempr" > /dev/null 
	local bat=`echo $packet | awk '{print $23}'`
	local batd=`echo "ibase=16; $bat"|bc`
	local batr=`echo "scale=0; (($batd/16-2)*10)"|bc -l`
	wget -O- "192.168.1.140/temp.php?nrc=$1&typ=9&value=$batr" > /dev/null 
}
while read line
do
   count=$[count + 1]
   if [ "$capturing" ]; then
      if [[ $line =~ ^[0-9a-fA-F]{2}\ [0-9a-fA-F] ]]; then
        packet="$packet $line"
      else
			MAC=`echo $packet | awk '{print $13$12$11$10$9$8}'`
			if [ "$MAC" = "D0F018431230" ] && [ ! "$dom" ] ; then
				wyslij 3
				dom=1
			fi
			if [ "$MAC" = "D0F0184312AA" ] && [ ! "$kuch" ] ; then
				wyslij 4
				kuch=1
			fi
			if [ "$MAC" = "D0F018431232" ] && [ ! "$sal" ] ; then
				wyslij 5
				sal=1
			fi
			if [ "$MAC" = "D0F018431231" ] && [ ! "$gab" ] ; then
				wyslij 6
				gab=1
			fi
			if [ "$MAC" = "D0F01843122F" ] && [ ! "$zew" ] ; then
				wyslij 8
				zew=1
			fi
			capturing=""
			packet=""
		fi
	fi
	if [ ! "$capturing" ]; then
	if [[ $line =~ ^\> ]]; then
	  packet=`echo $line | sed 's/^>.\(.*$\)/\1/'`
	  capturing=1
	fi
	fi
done`
Pozdrawiam!

1technophile, 2020-10-31:
Hi,

Could you translate your request please and also format it properly.

emwu88, 2020-11-02:
Hi,
there is a similar ble sensor from INODE, could someone add support to it?
Producer give a example in BASH to RPi how to read data from iBeacon:

#!/bin/bash

packet=""
capturing=""
count=0
kuch=""
dom=""
sal=""
gab=""
zew=""

wyslij () {
	#echo $packet >> /home/pi/d.txt
	local temp=`echo $packet | awk '{print $29$28}'`
	local tempd=`echo "ibase=16; $temp"|bc`
	local tempr=`echo "scale=2; ($tempd*175.72*4)/65536-46.85"|bc -l`
	wget -O- "192.168.1.140/temp.php?nrc=$1&typ=1&value=$tempr" > /dev/null 
	local bat=`echo $packet | awk '{print $23}'`
	local batd=`echo "ibase=16; $bat"|bc`
	local batr=`echo "scale=0; (($batd/16-2)*10)"|bc -l`
	wget -O- "192.168.1.140/temp.php?nrc=$1&typ=9&value=$batr" > /dev/null 
}

while read line
do
   count=$[count + 1]
   if [ "$capturing" ]; then
      if [[ $line =~ ^[0-9a-fA-F]{2}\ [0-9a-fA-F] ]]; then
        packet="$packet $line"
      else
			MAC=`echo $packet | awk '{print $13$12$11$10$9$8}'`
			if [ "$MAC" = "D0F018431230" ] && [ ! "$dom" ] ; then
				wyslij 3
				dom=1
			fi
			if [ "$MAC" = "D0F0184312AA" ] && [ ! "$kuch" ] ; then
				wyslij 4
				kuch=1
			fi
			if [ "$MAC" = "D0F018431232" ] && [ ! "$sal" ] ; then
				wyslij 5
				sal=1
			fi
			if [ "$MAC" = "D0F018431231" ] && [ ! "$gab" ] ; then
				wyslij 6
				gab=1
			fi
			if [ "$MAC" = "D0F01843122F" ] && [ ! "$zew" ] ; then
				wyslij 8
				zew=1
			fi
			capturing=""
			packet=""
		fi
	fi
	
	if [ ! "$capturing" ]; then
	if [[ $line =~ ^\> ]]; then
	  packet=`echo $line | sed 's/^>.\(.*$\)/\1/'`
	  capturing=1
	fi
	fi

done

nemoo2580, 2020-11-16:
[iNode Manufacturer Specific Data document](https://docs.google.com/document/d/1hcBpZ1RSgHRL6wu4SlTq2bvtKSL5_sFjXMu_HRyWZiQ) can help if anyone would like to implement that.

DigiH, 2022-05-27:
Hi @emwu88 @nemoo2580 do either of you have any of these devices for testing when a possible decoder might be ready?

I have had a look at the iNode Manufacturer Specific Data document you linked above and seen your initial PR for the iNode energy meter 
https://github.com/1technophile/OpenMQTTGateway/pull/755
where you state

> iNode EM bluetooth advertisement is identified by the first two bytes of manufacturerdata (9082, 9282, 9482, 9682 - 4 values, because bit 1 and 2 have an aditional special meaning, not needed for this integration).

but the documentation says
			
```

byte 92 | bit 2: rtto 
          bit 3: lowBattery
```

so I'm a bit confused on the conflicting bit order counting, and also stating that these two bits are not important for the current energy meter decoder, which now only works if both bits, be it 1 and 2 _(which I agree with with my understanding of binary bit counting and your given examples of 9**0**82, 9**2**82, 9**4**82, 9**6**82)_, or 2 and 3 _(as stated in the official documentation)_ are `0`, especially ignoring the case of `lowBattery`, which I think would be very helpful to have the decoder signal to the user and still work when this is the case, and `rtto`, which could possibly regular change as well.

From the official docs and internet translation engine ;)
```
rtto to flaga wskazująca czy upłynęły 24h odostatniego odczytu pamięci danych.
rtto is a flag indicating whether the last 24 hours of data memory reading have passed.
```

If these inconsistencies can be overcome by either of you having an actual iNode device and is willing to help testing we could try adding additional iNode devices, and possibly also extending the current energy meter decoder to still work with the above mentioned eventualities. I think that these are actually quite interesting devices.

Thanks

DigiH, 2022-05-28:
Hi all,

I've submitted an initial adjusted iNode Energy Meter decoder at

https://github.com/DigiH/decoder/tree/inode-energymeter

with these changes
• allow all manufacturerdata, index 1 variations to be recognised for decoding
• added the m³ water meter functionality
• adding current average and sum units depending on electricity or water metering
• @nemoo2580 I've changed the current average electricity power from your W to kW, so as to keep the same function working properly for the m³ water reporting. Shouldn't be a real issue with the 3 decimal places still being reported
• I'm also getting slightly different (+10%) battery % results, following the doc specification of
```
battery = (batteryAndLight >> 4) & 0x0F 

if battery == 1
    batteryLevel = 100 
else
    batteryLevel = 10 * (min(battery, 11) - 1)
```
• included the `lowBattery` in the decoder, but this is probably not that necessary if a correct battery percentage can be agreed upon
• didn't include the battery voltage, as even with 100% battery the calculation in the documentation was always below the expected 3V of the CR2032/14250/CR2 batteries it uses. Works fine with a minor adjustment of the given calculation thought ;) but is it really necessary to have three battery reports?
• added more test cases, also with water meter functionality

With this amended decoder all three iNode energy meters 

https://inode.pl/ENERGY-USAGE-MEASUREMENT-c11

should be compatible, but without real word testing possibilities this is all just theoretical.

Thanks

nemoo2580, 2022-05-28:
Hi @DigiH.
I still have this device and I can test your code in my spare time. I will try to do it next week.
I think that battery voltage is not necessary.
Thanks for you commit.

DigiH, 2022-06-25:
Hi @nemoo2580,
did you get any chance at all to test my commit above with your iNode energy meter?

Hi @emwu88,
while it's possible to create decoders for all the iNode Care Sensors from the documentation, without actual device test confirmation it's not a good idea to integrate them into any OpenMQTTGateway release. So if you have any model I'd be happy to work on decoders together with your verification.

If either of you is unsure about how to test unreleased decoder commits please let us know.

Thanks