1technophile/OpenMQTTGateway #793: mqtt_user is missing or not saved into the config

gborus, 2020-11-05:
**Describe the bug**
After configuring the RF gateway through Wifi Manager, MQTT login failed, when I checked the Serial console for details, I cannot see the mqtt_user value I setup on the config page.

**To Reproduce**
Steps to reproduce the behavior:
1. After firmware upload, configure the device through the website
2. Check through serial console if you see mqtt_user info.

**Screenshots**

**Environment (please complete the following information):**
* OpenMQTTGateway v0.9.5
* I also tried, OpenMQTTGateway v0.9.4
* nodemcuv2-rf-firmware.bin I used

1technophile, 2020-11-05:
Could you indicate the length of your mqtt_user?

gborus, 2020-11-05:
Hey,

yeah, I tried 'test' and 'a' as well, but the result was the same.

thanks,

1technophile, 2020-11-05:
This is due to the length of your mqtt password, the max size is 60

gborus, 2020-11-05:
cool, that's easy to fix, let me try, will report back shortly

gborus, 2020-11-05:
this gonna take sometime, I used the auto generated homeassitant mqtt user and pwd, but that's I cannot change through the web interface, I need to dig into it.
probably this will be an issue for new homeassistant users. What do you think?

1technophile, 2020-11-05:
>probably this will be an issue for new homeassistant users. What do you think?

If you add a new user into home assistant and use it for OMG you should be ok

gborus, 2020-11-05:
Thank you, that helped. I created a homeassitant account for the bridge and with the shorter password I can see now the mqtt user password in the serial console also managed to connect to the broker.

1technophile, 2020-11-05:
If you are interested, do not hesitate to contribute to the [doc](https://github.com/1technophile/OpenMQTTGateway/blob/development/docs/integrate/home_assistant.md) by submitting a PR :-)

closed 2020-12-11 by 1technophile