1technophile/OpenMQTTGateway #794: Can't publish BTtoMQTT messages when Secure_Connection Enabled

animal6496, 2020-11-05:
Hi I'm experiencing problems when I enable Secure_Connection to the MQTT broker. The Last Will message and Version message at start are published and also the SysToMQTT is published. But when it comes to the BLE sensors (BTtoMQTT) the OMG client disconnects and reconnects again after showing all missing BTtoMQTT messages in the terminal. The error message in the terminal is (shown twice) is:

[E][ssl_client.cpp:33] _handle_error(): [data_to_read():270]: (-76) UNKNOWN ERROR CODE (004C)

I run V0.9.5 on a NodeMCU ESP32 board. Any suggestions welcome.

1technophile, 2020-11-06:
Hi,

Could you try to build and test with
```.ini
esp32_platform = espressif32@2.0.0
```

instead of 1.12.2 in platformio.ini file

animal6496, 2020-11-06:
OK I recompiled the code, erase flash memory and then reflashed the ESP32. The result is unfortunately the same. The ESP32 goes offline and comes back again after all the BLE stuff. Looks almost like that the BLE and the Wifi SSL stuff are mutually exclusive. If you guide me through the process I can help debugging. I'm not really familiar with OMG yet, just started to test it as a gateway to get the Xiaomi Scale and RF433 Kaku readings as my platform of choice Tasmota doesn't support this (yet).

animal6496, 2020-11-06:
I found the DEBUG flag and compiled with CORE_DEBUG_LEVEL=4.
I received following output in the terminal.

```
N: Connected to broker
N: Scan begin
D NimBLEScan: ">> start(duration=10)"
D NimBLEScan: "<< start()"
I NimBLEScan: "NEW DEVICE FOUND: xx:xx:xx:xx:xx:xx"
D NimBLEAdvertisedDevice: "payload: 02011a020a0c0aff4c00100505184cbb91"
I NimBLEScan: "UPDATING PREVIOUSLY FOUND DEVICE: xx:xx:xx:xx:xx:xx"
N: Device detected: XX:XX:XX:XX:XX:XX
N: Subject: /BTtoMQTT/XXXXXXXXXXXX
N: Received json : {"id":"XX:XX:XX:XX:XX:XX","rssi":-68,"txpower":12,"distance":2.799256}
[E][ssl_client.cpp:33] _handle_error(): [data_to_read():270]: (-76) UNKNOWN ERROR CODE (004C)
[E][ssl_client.cpp:33] _handle_error(): [data_to_read():270]: (-76) UNKNOWN ERROR CODE (004C)
: Wclient : not connected can't pub
MQTT connection...
I NimBLEScan: "NEW DEVICE FOUND: xx:xx:xx:xx:xx:xx"
D NimBLEAdvertisedDevice: "payload: 02011a020a0c0bff4c001006171a542ef6a8"
I NimBLEScan: "UPDATING PREVIOUSLY FOUND DEVICE: xx:xx:xx:xx:xx:xx
N: Device detected: XX:XX:XX:XX:XX:XX
N: Subject: /BTtoMQTT/XXXXXXXXXXXX
N: Received json : {"id":"XX:XX:XX:XX:XX:XX","rssi":-55,"txpower":12,"distance":0.495572}
W: client not connected can't pub
etc. etc.
```

1technophile, 2020-11-08:
It seems that the client loses the connection when the scan is in progress, we should maybe try to move the publication after the scan by implementing storage of the JSON data.

Another hint is maybe a low memory when processing the scan and SSL enable. Are you comfortable with the add of:
`Log.notice(F("FreeHeap: %d" CR), ESP.getFreeHeap());` 
after the scan [beginning ](https://github.com/1technophile/OpenMQTTGateway/blob/a184355f9bb03bb59fd67197f4ddfba184d552b0/main/ZgatewayBT.ino#L479) at the [end](https://github.com/1technophile/OpenMQTTGateway/blob/a184355f9bb03bb59fd67197f4ddfba184d552b0/main/ZgatewayBT.ino#L486).

animal6496, 2020-11-08:
Earlier today I traced the problem to getscan, in the BLEscan routine in ZgatewayBT. I was thinking along the same line using temp storage for the scane results as a workaround. Didn't think of potential memory issues. Anyway I did the log reports just before and after getscan in the BLEscan routine. Heap doesn't seem to be the problem, however, the first problems already start before getscan:
```
N: RF_EMITTER_GPIO: 12 
N: RF_RECEIVER_GPIO: 27 
N: BLE scans interval: 55555
N: BLE scans number before connect: 10
N: Publishing only BLE sensors: false
N: minrssi: 100
N: Low Power Mode: 0
N: Setup OpenMQTTGateway end
W: MQTT connection...
N: Connected to broker
N: Subject: /SYStoMQTT
N: Received json : {"uptime":5,"version":"OMG_RvS","freemem":109212,"rssi":-82,"SSID":"Nuts_locked_nomap","ip":"172.16.4.102","mac":"XX:XX:XX:XX:XX:XX","wifiprt":0,"lowpowermode":0,"interval":55555,"scanbcnct":10,"modules":"RFBTHADiscovery"}
N: Scan begin
[E][ssl_client.cpp:33] _handle_error(): [data_to_read():270]: (-76) UNKNOWN ERROR CODE (004C)
W: MQTT connection...
I NimBLEDevice: "BLE Host Task Started"
I NimBLEDevice: "NimBle host synced."
N: FreeHeap: 147504
N: FreeHeap: 147448
D NimBLEScan: ">> start(duration=10)"
D NimBLEScan: "<< start()"
I NimBLEScan: "NEW DEVICE FOUND: XX:XX:XX:XX:XX:XX"
D NimBLEAdvertisedDevice: "payload: 10161a18a4c13876e69000f42c5c0bd68a"
I NimBLEScan: "UPDATING PREVIOUSLY FOUND DEVICE: XX:XX:XX:XX:XX:XX"
D NimBLEAdvertisedDevice: "payload: 0b094154435f373645363930"
N: Device detected: A4:C1:38:76:E6:90
W: client not connected can't pub
D FreeRTOS: "Semaphore taking: name: createOrUpdateDevice (0x3ffbac84), owner: <N/A> for createOrUpdateDevice"
D FreeRTOS: "Semaphore taken:  name: createOrUpdateDevice (0x3ffbac84), owner: createOrUpdateDevice"
D FreeRTOS: "Semaphore giving: name: createOrUpdateDevice (0x3ffbac84), owner: createOrUpdateDevice"
N: Subject: /BTtoMQTT/XXXXXXXXXXXX
N: Received json : {"id":"XX:XX:XX:XX:XX:XX","name":"ATC_XXXXXX","rssi":-65,"distance":2.009448,"model":"LYWSD03MMC_ATC","tempc":24.4,"tempf":75.92,"hum":44,"batt":92,"volt":3.03}
W: client not connected can't pub
I NimBLEScan: "NEW DEVICE FOUND: 59:f1:60:f6:3c:a5"
Etc.etc.etc
```

1technophile, 2020-11-08:
Another possibility could be to don't use a task and put the scan in the main loop:
https://www.reddit.com/r/esp32/comments/gt89yn/aws_mqtt_connection_keeps_dropping_when_using/

animal6496, 2020-11-08:
Tried to narrow it down and added 
`NIMBLE_LOGI(LOG_TAG, "BLEDevice::init Started");`
just before if(!initialized){ in NimBLEDevice::init
```
N: Scan begin
[E][ssl_client.cpp:33] _handle_error(): [data_to_read():270]: (-76) UNKNOWN ERROR CODE (004C)
I NimBLEDevice: "BLEDevice::init Started"
W: MQTT connection...
I NimBLEDevice: "BLE Host Task Started"
I NimBLEDevice: "NimBle host synced."
D NimBLEScan: ">> start(duration=10)"
D NimBLEScan: "<< start()"
```
So it looks like that the BLEDevice::init("")  call itself is enough to raise the error. Or is the error message from a different "thread"? 
Let me see what I can do. I'm not really a c++ programmer so it may take some time and the result might not what you're hoping for.

1technophile, 2020-11-08:
Interesting, maybe @h2zero has some hints to help us on this matter?

animal6496, 2020-11-08:
Just noticed your comment with the link. At first glance this looks like the same issue. I'll start reading it more carefully.

h2zero, 2020-11-09:
@1technophile, I don't have many hints to provide for this issue unfortunately.

If it is caused by just initializing NimBLE then it could be something to do with the task switching that occurs at that point as the initialization creates a task for the BLE host with high priority. 

Perhaps it doesn't switch back to the MQTT client task soon enough and the connection is getting dropped?

Is it just the MQTT client thats getting disconnected or does the Wifi connection drop as well?

animal6496, 2020-11-09:
Sorry can’t check now. So I don’t know for sure but I had the idea that WiFi connection remained and only MQTT dropped. Are you suggesting that we ‘re looking at  an issue with PubSub client? I read about that library having issues with heavy loads or insufficient priority. In that case we might want to replace with aSyncMQTT or similar. The above suggestion to process and store results offline will work but there is a risk of running out of memory if there are a lot of devices to process. In my application I’ve only 15 but you can’t rely on this. I would prefer to address the real problem and not built in a future problem for others.

h2zero, 2020-11-09:
Possibly there is an issue with the client but it could possibly be resolved by running it in a task with elevated priority. Just speculation though, I don't know if that is the cause.

On similar note I use [lwmqtt](https://github.com/256dpi/lwmqtt)/[arduino-mqtt](https://github.com/256dpi/arduino-mqtt) without issues but I don't deinit/reinit BLE so I'm not sure if it would be affected the same way.

animal6496, 2020-11-11:
I've tested and can confirm that only the MQTT client loses connection. The Wifi connection is OK. This is what I could test quickly. To go further is going to take a lot more time as I never code in cpp, the ESPxx platform is new to me (for programming) and the whole IDE setup is new. So I'm on a very steep learning curve and unfortunately can't spend a lot of time on this. Just managing expectations here.

h2zero, 2020-11-11:
Thanks for that, at least we know it's not the antenna sharing causing the issue then. Unfortunately I am not familiar with OpenMQTTGateway code so I can't be certain of any conflicts with BLE and MQTT. I'll have a look though and see if anything stands out.

h2zero, 2020-11-12:
Did some digging on this and also had it reported in an issue that came up on my repo. 

Seems a similar issue was posted a year ago with the bluedroid library espressif/arduino-esp32#3449. However, the error there is different (-16) and here we are seeing (-76), I'm not sure what that error code is.

The suggested workaround for now is to do as @1technophile suggests and store the BLE data until the scan is complete then publish it. That process is working for other users.

In the mean time I will try to find a proper solution, but I need to reproduce the problem first.

animal6496, 2020-11-12:
Thank you h2zero. I 'll give it a try to get it in, will be a good learning experience.

animal6496, 2020-11-16:
OK I followed the suggestion and implemented wrappers around pub() that first check if there is an MQTT connection and if not stores the messages in a message queue. When the MQTT connection is back it checks for messages in the queue and publishes those. This seems to work, as I get messages from the various BT devices via the gateway on the broker. Unfortunately the messages a re not yet complete need to look into that. Also need to check long term stability as I do not entirely trust my pointer handling. The message store is an array created at compile time to avoid problems with heap corruption, and has (compiler environment setting) a capacity of 32 messages. I noticed however when a friend was visiting I suddenly had 5 extra BT devices, so it becomes vary important to use Black/White list options otherwise the message queue might be easily running out of space. Other option is a dynamic solution. Anyway, most important is that the approach works. Now I need to sort out how github works so that I can post the code for others to check and/or improve.

animal6496, 2020-11-17:
Florian, can you help I'm lost working with github. I created a branch tls_ble_conflict_resolution and I added the three files that I changed user_config.h, ZgatewayBT.ino and main.ino and created a pull request. This results in errors on file that I didn't work on and also in an error on main.ino that I don't have working in my own VSC environment. I used the files from 0.9.5 as base can this be an issue?

1technophile, 2020-11-17:
Thanks for the PR!

>  I used the files from 0.9.5 as base can this be an issue?

Yes you should take the development branch as the base of your work. If you take the v0.9.5 you will generate regressions when we will merge your code

> This results in errors on file that I didn't work on

This could be due to the fact that you took v0.9.5, note that you have also code format errors. So as to setup the code format check, you may take a look to this:
https://docs.openmqttgateway.com/participate/development.html#development-contributions

animal6496, 2020-11-17:
Thanks Florian.  I've now a new version now started from development branch. Also this is happily running for a little bit more than an hour now, so far so good. This implementation is a workaround for dropping the MQTT connection by the BLEscan, it is not a solution as the MQTT connection is about 25% of the time gone. Meaning if you use the gateway for other tasks that are time critical than you might want to try something else. The PR resulted in a compiler error on code I introduced, the error doesn't show in my local built so it must be environment flag. Locally I only compiled for ESP32 with BT option as I don't need anything else. Will look into that.

h2zero, 2020-11-18:
I just thought I'd chime in here and let you know that the reason the connection drops is most likely due to the scan interval/window settings. 

Looking at the code I can see they are not altered from the default values which would make the BLE scanning take 100% of the antenna time, leaving none for maintaining the MQTT connection. 

While it is still the best practice to store the scan results and send via MQTT after I would also suggest altering the scan interval to be around 100ms or more than the scan window. This will allow WiFi to use the antenna and maintain the MQTT server connection.

animal6496, 2020-11-18:
OK I just tried and set the Interval and Window time just before starting the scan. 
```
  pBLEScan->setInterval(115);
  pBLEScan->setWindow(15);
  BLEScanResults foundDevices = pBLEScan->start(Scan_duration / 1000, false);
```
That didn't make any difference. Without having a good idea about the range for these parameters, I've tried smaller window settings. This just resulted in less devices found during a scan. If I go as small as 2 ms I get GAP errors from the BLEscan process. Larger Interval settings didn't help either. 
I still have the impression that there is a code issue. I'm digging into what is triggering the errors below.
```
N: Device detected: XX:XX:XX:XX:XX:XX
[E][ssl_client.cpp:33] _handle_error(): [data_to_read():270]: (-76) UNKNOWN ERROR CODE (004C)
[E][ssl_client.cpp:33] _handle_error(): [data_to_read():270]: (-76) UNKNOWN ERROR CODE (004C)
W: MQTT connection...
```

h2zero, 2020-11-18:
That's unfortunate. Sadly there is no exact science to find values that will work for this but for a test could you try something like this?
```
pBLEScan->setInterval(1500);
pBLEScan->setWindow(450);
```
That should produce a very large window for WiFi to use the antenna and hopefully verify the theory.

animal6496, 2020-11-18:
From the positive side it didn't get worse ;) and the suggested changes work like a charm. On the downside the change in window and Interval parameters have no effect. Immediately after finding the first device the connection to the broker goes down like in the first report on the issue. Since playing with the timing parameters doesn't seem to have any effect on this (I never get to the second device) I suspect there is an interaction between the DeviceCallback routine and the PubSub library but I 've no real data to backup for this.

h2zero, 2020-11-18:
Was worth a shot at least, too bad it didn't have any effect. I'm glad you've got a working solution and thanks for testing! 

I'd like to find the answer to this puzzle still.

animal6496, 2020-11-21:
@Legion2 I agree that a true solution would fix the underlying issues which I haven't been able to trace down (yet). Would be nice to have a proper debugger. I just implemented what was suggested by h2zero and Florian as a fix but I understand also from h2zero above that storing results for delayed publication is also used by others, possibly to avoid BLE/Wifi antenna sharing problems. I've implemented this way to make as little change to the original code as possible so that it can be easily removed when a proper solution is found. Also I do not really speak cpp or even have experience with ESP's so when I have to work too much on the code it quickly becomes too complex for me. But I'm happy to learn if you have a better approach.

1technophile, 2021-04-24:
resolved on v0.9.6

closed 2021-04-24 by 1technophile