1technophile/OpenMQTTGateway #797: Best settings for running on battery

MrFloppy, 2020-11-09:
Hi,

I tried a lot to change settings to get the most runtime out of my 18650 battery that is connected to my M5stick-c (with the official shield). Unfortunately OMG started to behave strange (went offline forever or never), making me resign.

Do you have any recommendations for OMG settings just using ble to gather data from different sensors? Every 10-30 minutes would be enough in my mind.

Thanks a lot.

1technophile, 2020-11-09:
> Unfortunately OMG started to behave strangely (went offline forever or never), making me resign.

Hi,

I suppose that you are using low power mode 2?

MrFloppy, 2020-11-10:
> I suppose that you are using low power mode 2?

Yes, I do. I also increased the time between scans and the scan count.

1technophile, 2020-11-11:
> Do you have any recommendations for OMG settings just using ble to gather data from different sensors? Every 10-30 minutes would be enough in my mind.

I tested also the M5StickC battery hat but my results were not very satisfying.

> OMG started to behave strange (went offline forever or never), making me resign

Do you have the same behavior without the hat?

MrFloppy, 2020-11-12:
> 
> I tested also the M5StickC battery hat but my results were not very satisfying.
> 

Yes, that is the same for me. Maybe I change the battery and try that? 

> 
> Do you have the same behavior without the hat?
I used the precompiled firmware and a power bank there. I had no issues with that, except that the power bank went empty within a day. I usually would expect it to last way longer.

1technophile, 2020-11-14:
> I used the precompiled firmware and a power bank there. I had no issues with that, except that the power bank went empty within a day. I usually would expect it to last way longer

Could you share the capacity if your power bank ? 
I will launch a test with my M5stickC with the internal battery

1technophile, 2020-11-14:
The issue with M5StickC is that it is not designed for low power:
[image]
You can see above that in low power mode it draws 16mA, this is a lot.

If you want to have an ESP32 well suited for low power, you may take a look to the [Lolin ESP32 lite ](https://www.banggood.com/LOLIN32-Lite-V1_0_0-WiFi-bluetooth-Board-Based-ESP-32-Rev1-MicroPython-4MB-FLASH-Module-p-1237861.html?rmmds=myorder&cur_warehouse=CN&p=SZ25159101562014119R&custlinkid=1348972)
_Affiliated link_

With the Lolin32 low power mode 2:
[image]

Note that I didn't test it by the battery port, it should be less with a battery.

1technophile, 2020-11-14:
Lolin32 lite on low power mode 2 is at around 60uA on battery:
[image]

So that's interesting!

MrFloppy, 2020-11-15:
Do they all work with OMG? I am happy to switch to different hardware, but I am still not sure, which one would be the best to use as a BT gateway. 

The powerbank has a capacity of 5200mA, but it is an older one, so it may be a little less.

1technophile, 2020-11-15:
> Do they all work with OMG?

Yes the tests were done with OMG on the Lolin32 lite.

> but I am still not sure, which one would be the best to use as a BT gateway.

Look for an esp32 with a battery connector and uA consumption in deep sleep.
I'm not sure passing through a usb power bank is suitable : 
- you have more components netween the battery and the esp, driving more current draws.
- the power bank could standby itself due to the low current. To avoid this you need to have an always ON one.

The best way is to use a battery + an esp32 board designed for low power, supporting batteries.

MrFloppy, 2020-11-15:
Thank you very much. I will take a look around. The power bank was just connected to get a feeling for the power drain. Besides that you are absolutely right with all your points.

I will do some research for esp32 boards that might fit and post my results here to make this issue complete for others.

1technophile, 2020-11-15:
If you don't mind I would prefer that we continue the discussion on the [forum](https://community.openmqttgateway.com/). I would like to keep the issues for features and bugs.

1technophile, 2020-11-15:
Note that for the moment the interval is not persisted on memory, so the command must be published with retain flag to be effective.

MrFloppy, 2020-11-15:
Yes. Feel free to close the issue. I thank you very much for your support.

1technophile, 2020-11-15:
This document even if a little bit dated could help you:
https://docs.google.com/spreadsheets/d/1Mu-bNwpnkiNUiM7f2dx8-gPnIAFMibsC2hMlWhIHbPQ/edit#gid=0

florian-athingstory, 2020-11-21:
For information with a 280mAh battery and a 550s interval, I get more than 3 days and a half of scan.

1technophile, 2020-11-21:
For information with a 280mAh battery and a 550s interval, I get more than 3 days and a half of scan.

MrFloppy, 2020-11-28:
@1technophile Thank you very much for your effort. I switch back to usb adapter(s), although I do not like it. But it least it works and I added a ESP32 to cover very much the whole flat.

closed 2020-12-01 by 1technophile