1technophile/OpenMQTTGateway #798: BLE connection will not be closed

michapr, 2020-11-09:
I have noted a high battery usage of the MHO-C401 sensors and one LYWSD03MMC with original firmware.

Looking for the power usage I have seen that after connection to the deices the connection after received notification will not be closed.
So, in my case, the connection will be live for 1:06 minutes (!!) - until next scan, where a 
**BLEDevice::deinit(true);**
will kill all connections.
During this time the sensor take about 3-4mA (peaks between 1.5 and 5,5mA)  what will drain the battery of course.

For MMC is the alternate firmware a solution, but for MHO-C401 not.

Not sure how it can be solved, if connection to device cannot be closed after received notification - maybe make a 
**BLEDevice:deinit(true)** 
after a defined time (as sample max 10 seconds after request)?

Thanks
Michael

1technophile, 2020-11-09:
Good point @michapr !
Will take a look

1technophile, 2021-01-02:
I have added the disconnection after a timeout, in the BLEconnect function.

closed 2021-01-30 by 1technophile