1technophile/OpenMQTTGateway #802: Digoo DG-R8S not working with development branch

1technophile, 2020-11-15:
**Describe the bug**
Regression regarding Digoo DG-R8S support with ESPilight v0.17.0

**To Reproduce**
Steps to reproduce the behavior:
* With Pilight v0.17.0 R8S KO, R8H OK
* With Pilight v0.16.2 R8S OK, R8H KO

KO = no data received

**Expected behavior**
* With Pilight v0.17.0 R8S OK, R8H OK

**Environment (please complete the following information):**
* OpenMQTTGateway development
* Library ESPilight

**Additional context**
Reproduced also by @tmcatalin
https://community.openmqttgateway.com/t/digoo-dg-r8s-433mhz-battery-operated-temperature-and-humidity-sensor-with-embedded-display/287/72

1technophile, 2020-11-15:
@melyux, @puuu
I suppose that I should open an issue in the Pilight repository, unfortunately, it doesn't seem possible. Do you have some ideas regarding this issue?

hugokernel, 2020-12-10:
Same problem here.

derbrummbaer, 2021-03-04:
I have the same problem with my Weather Sensors using alecto_ws1700 or alecto_wsd17.
OMG 0.9.5 with PiLight 0.17.0 is not working
OMG 0.9.5 with PiLight 0.16.2 is working, 
had to made some changes in ZgatewayPilight.ino but after that i can use it with 0.16.2

melyux, 2021-12-01:
I don't have this one, I do have some that use Nexus but they're still working. I also noticed the Pilight repository has no ability to add issues, and I don't know why. There's a rewrite going on, maybe he's turned it off until then for some reason. The Pilight forums are an option, but they're sparse... the whole Pilight project is feeling too stodgy.

steadramon, 2022-05-31:
OK - I got this working with 0.17.0 as well - I've added some code to be able to enable/disable certain decoders in Pilight - 

I found that my Digoo sensor decodes best with tfa and that the nexus decoder seems to interfere - 

home/OpenMQTTGateway_ESP32_Pilight/PilighttoMQTT msg: {"limitprotocols":["tfa","ev1527"]}

This makes use of the limitProtocols call within ESPilight, of course we might want to know what protocols are available -

home/OpenMQTTGateway_ESP32_Pilight/PilighttoMQTT msg: {"availableprotocols":["x10","tfa30","tfa2017","tfa","teknihall","techlico_switch","tcm","smartwares_switch","silvercrest","selectremote","secudo_smoke_sensor","sc2262","rsl366","rc101","quigg_screen","quigg_gt9000","quigg_gt7000","quigg_gt1000","pollin","ninjablocks_weather","nexus","mumbi","logilink_switch","kerui_D026","iwds07","impuls","heitech","ev1527"]}

I'll raise a PR with some changes in it once I'm a bit happier with things... I do have questions about persistent storage - is there a standard in use for OMG at the moment? It would be best to save the limited protocols between reboots etc - also have questions about the MQTT topics and where things like the above should be posted and responded

1technophile, 2022-06-01:
Thanks for the feedback!

>  I do have questions about persistent storage - is there a standard in use for OMG at the moment? It would be best to save the limited protocols between reboots etc -

Most of users publish the configuration with the retain flag, this way the published message is saved by the broker. If you want to keep it upon broker reboot also, you can activate persistence on broker side.

closed 2023-10-23 by github-actions[bot]