1technophile/OpenMQTTGateway #806: Blood pressure sensor Silvercrest SBM69

tiagofreire-pt, 2020-11-18:
**Is your feature request related to a problem? Please describe.**
No.

This is a request for a new BLE device, such as Silvercrest SBM69 sold by LIDL in Europe.

**Describe the solution you'd like**
I need to intergrate this blood pressure product on my Home Assistant instance, regarding my biometrics usage for automations and automatic long term register.

Captured data with the android APP "nRF Connect".

RAW Data: 
0x060953424D36390201060302101804FF130601

Blood pressure service: 00001810-0000-1000-8000-00805f9b35fb
Blood pressure measurement characteristic: 00002a35-0000-1000-8000-00805f9b34fb
Properties: Indicate
Value:
Systolic: xxx.xx mmHg
Disastolic: yyy.yy mmHg
Mean AP: zzz.zz mmHg
Timestamp: hh:mm:ss dd.mm.yyyy
Pulse: aaa.a bpm
User ID: 0

**Describe alternatives you've considered**
No other solution known is available.

**Additional context**
This device requires an active BLE connection, inserting a randomly key code generated only on the first pairing moment.

tiagofreire-pt, 2020-11-22:
FCC info: https://fccid.io/OU9SBM69-B01

ZolaSat, 2022-07-26:
...and the solution?

ignisf, 2023-02-26:
The device seems to use the standard HDP profile for blood pressure monitoring. Here's and ImHex pattern that recognises the output:

```
#include <type/float16.pat>

bitfield Flags {
 blood_pressure_units: 1;
 time_stamp: 1;
 pulse_rate: 1;
 user_id: 1;
 measurement_status: 1;
 reserved_for_future_use: 3;
};

bitfield MeasurementStatus {
  body_movement_detected: 1;
  cuff_too_loose: 1;
  irregular_pulse: 1;
  pulse_rate_exceeds_lower_limit: 1;
  pulse_rate_exceeds_upper_limit: 1;
  improper_measurement_position: 1;
  reserved_for_future_use: 10;
};

struct blood_pressure_measurement {
  Flags flags;
  
  if (flags.blood_pressure_units) {
    type::float16 systolic_kpa;
    type::float16 diastolic_kpa;
    type::float16 mean_arterial_pressure_kpa;
  } else {
    type::float16 systolic_mmhg;
    type::float16 diastolic_mmhg;
    type::float16 mean_arterial_pressure_mmhg;
  }
  
  if (flags.time_stamp) {
    u16 year;
    u8 month;
    u8 day;
    u8 hours;
    u8 minutes;
    u8 seconds;
  }
  
  if (flags.pulse_rate) {
    type::float16 pulse_rate;
  }
  
  if (flags.user_id) {
    u8 user_id;
  }
  
  if (flags.measurement_status) {
    MeasurementStatus measurement_status;
  }
};

blood_pressure_measurement bpm @ 0x00;
```

ignisf, 2023-02-26:
Python PoC that reads out all data from the device:

```python
import asyncio
from bleak import BleakClient
from bleak.backends.characteristic import BleakGATTCharacteristic

address = "b8:b7:7d:XX:XX:XX"

async def eternity():
    await asyncio.sleep(3600)

async def main(address):
    async with BleakClient(address) as client:
        for service in client.services:
            print("Service: {0} ({1})".format(service.description, service.uuid))

            for characteristic in service.characteristics:
                print("  Characteristic: {0} ({1}), properties: {2}".format(characteristic.description, characteristic.uuid, ",".join(characteristic.properties)))

                for descriptor in characteristic.descriptors:
                    print("    Descriptor: {0} ({1})".format(descriptor.description, descriptor.uuid))

                if 'read' in characteristic.properties:
                    value = await client.read_gatt_char(characteristic.uuid)
                    print("    Value: {0} ({1})".format("".join(map(chr, value)), value.hex()))

                if 'indicate' in characteristic.properties:
                    await client.start_notify(characteristic.uuid, on_notification)
                    try:
                        await asyncio.wait_for(eternity(), timeout=0.5)
                    except TimeoutError:
                        print('    timeout!')

async def on_notification(handle: BleakGATTCharacteristic, data: bytearray):
    print("    Value: {0}".format(data.hex()))

asyncio.run(main(address))
```

ignisf, 2023-02-26:
Unfortunately it *is* necessary for the device to be paired at first use. It refuses to respond unless paired.

DigiH, 2023-02-26:
Unless some functions require registration for NOTIFY, all others can be read by OpenMQTTGateway's [READ command](https://docs.openmqttgateway.com/use/ble.html#read-write-ble-characteristics-over-mqtt)

For OpenMTTGateway to be able to directly publish decoded properties through the [Theengs Decoder](https://github.com/theengs/decoder) library, the Silvercrest SBM69 would need to broadcast its information in its advertisement data. If this is the case could you provide some sample data along with the relevant reading?

ignisf, 2023-02-26:
Unfortunately you need to subscribe to `indicate` events for the 'Blood Pressure Measurement (00002a35-0000-1000-8000-00805f9b34fb)' characteristic in order to get the measurement data. No relevant data is part of the advertisements of the device.

DigiH, 2023-02-26:
Unfortunately this is then currently not accessible through OpenMQTTGateway.

ignisf, 2023-02-28:
FWIW I just created https://github.com/ignisf/sbm69 and will be looking into making a home assistant component that uses it.