1technophile/OpenMQTTGateway #808: Multiple BLE-Gateway in one building and network

Nikolay-Ch, 2020-11-23:
I have two Gateways in one network (and in one building). So, if I setup this gateways with different names, HomeAssistant give one device with two MQTT-topics. And the HomeAssistant receives one device with two different MQTT-topics. The HomeAssistant does not work correctly.
If I set same name to both gateways, they appear in the HomeAssistant as one device. Also wrong.

I propose to make two different settings: one for the name of the gateway itself, and the second for the MQTT-topic, to which messages from sensors will be sent.

1technophile, 2020-12-01:
I think you are using auto-discovery function with MQTT.

If it is the case, I propose to change the auto-discovery function to publish the discovery messages with a wild card instead of the gateway name.
This way Home Assistant will create the entities with the wild cards and process all the messages coming from one device regardless of the gateway used.

1technophile, 2020-12-01:
Can be tested in #819

closed 2020-12-03 by 1technophile

Nikolay-Ch, 2020-12-24:
Hi! Please, can you explain, how to use this feature?

1technophile, 2020-12-25:
With the new development branch, by default, the mqtt auto discovery function will produce a topic with wildcards instead of the gateway name and location.
Before an MQTT discovery payload looked like this:
``` json
{"stat_t":"home/OpenMQTTGateway_ESP32_M5STICK_CP_BLE_IR/BTtoMQTT/485743015C3A","name":"Formal-batt","uniq_id":"485743015C3A-Formal-batt","dev_cla":"battery","val_tpl":"{{ value_json.batt | is_defined }}","unit_of_meas":"%","device":{"connections":[["mac","485743015C3A"]],"identifiers":["485743015C3A"],"model":"JQJCY01YM","name":"Formal","via_device":"OpenMQTTGateway_ESP32_M5STICK_CP_BLE_IR"}}
```
After the PR it looks like this:
``` json
{"stat_t":"+/+/BTtoMQTT/485743015C3A","name":"Formal-batt","uniq_id":"485743015C3A-Formal-batt","dev_cla":"battery","val_tpl":"{{ value_json.batt | is_defined }}","unit_of_meas":"%","device":{"connections":[["mac","485743015C3A"]],"identifiers":["485743015C3A"],"model":"JQJCY01YM","name":"Formal","via_device":"OpenMQTTGateway_ESP32_M5STICK_CP_BLE_IR"}}
```
This way the sensor data retrieving in HASS is not linked to a particular gateway.

Nikolay-Ch, 2020-12-26:
Thanks! But I not found where I can turn on this feature?
I must leave blank the fields "Gateway Name" and "Topic Base"?

1technophile, 2020-12-26:
With the last development branch, it is activated by default.

Nikolay-Ch, 2020-12-26:
Ok, I found this....
If I understand correctly the code, I must leave the "Gateway Name" field empty?

Nikolay-Ch, 2020-12-26:
Look... I download and compile the **development** branch, and...
Now my ESP32 send config topic with the state: **stat_t":"+/+/BTtoMQTT/XAXAXAXAXAXA**
But received data from BLE-device sended in the topic: **home/MyOpenMQTTGateway/BTtoMQTT/XAXAXAXAXAXA**
Maybe some errors in development branch?

1technophile, 2020-12-26:
> Now my ESP32 send config topic with the state: stat_t":"+/+/BTtoMQTT/XAXAXAXAXAXA
But received data from BLE-device sended in the topic: home/MyOpenMQTTGateway/BTtoMQTT/XAXAXAXAXAXA

This is the expected behavior. The gateway publishes to a particular topic the sensor data retrieved, composed of gateway ID and sensor ID. Home assistant listens to all the gateways for a particular sensor ID. 
You can find information regarding wildcard [here](https://subscription.packtpub.com/book/application_development/9781787287815/1/ch01lvl1sec18/understanding-wildcards)

anarchking, 2021-01-24:
Is there a commands config topic for renaming the gateway? I had trouble entering all the config in wifimanager before it kicks me, and i have to start all over with wifimanager

1technophile, 2021-01-26:
> Is there a commands config topic for renaming the gateway? I had trouble entering all the config in wifimanager before it kicks me, and i have to start all over with wifimanager

You can set the gateway name in your .ini file or user_config.h file before building the code or into wifimanager portal. But you cannot modify it by MQTT currently.