1technophile/OpenMQTTGateway #809: Please add support to April Brother BLE-devices

Nikolay-Ch, 2020-11-23:
Please add support to April Brother BLE Beacon devices:

ABTemp:
https://wiki.aprbrother.com/en/ABTemp.html
https://wiki.aprbrother.com/en/ABTemp_Payload_Format.html

ABSebsor N01:
https://wiki.aprbrother.com/en/ABSensor.html#absensor-n01

ABSensor N03:
https://wiki.aprbrother.com/en/ABSensor.html#absensor-n03

1technophile, 2020-12-01:
Hi,

If you have these sensors, could you take a look to how they advertize their data, servicedata, manufacturerdata, BLE connection?
The best way to do that, is to download [Nrf connect ](https://play.google.com/store/apps/details?id=no.nordicsemi.android.mcp) and look for the beacons and see if they advertise servicedata or manufacturerdata according to the measurements done.
If yes you may publish here extracts of different values/ advertized data and we could check how to read them.

Nikolay-Ch, 2020-12-01:
> Hi,
>
> If you have these sensors, could you take a look to how they advertize their data, servicedata, manufacturerdata, BLE connection?
> The best way to do that, is to download [Nrf connect [...]
Hi,
Both sensors advertise in the Eddystone format... Detailed info about packet format on the links:
https://wiki.aprbrother.com/en/ABSensor.html#n01-packet-format
https://wiki.aprbrother.com/en/ABSensor.html#n03-packet-format
If I can help to do something, I can try, but I do not code at c++ for a long time...

Here is my coe in c#, that parse data for ABN03 from OpenMQTTGateway message:
``C#
string payload = Encoding.UTF8.GetString(e.ApplicationMessage.Payload);
var payloadObj = JObject.Parse(payload);
var serviceData = payloadObj.ContainsKey("servicedata") ? payloadObj["servicedata"].Value<string>() : "";

// if serviceData not started with ab03, that we can't parse data - skip parsing
if (!serviceData.StartsWith("ab03"))
    return;

// convert hexadecimal string info byte-array
var payloadBytes = serviceData.FromHexStringToByteArray();

// create parsed messages
var temp = payloadBytes.ParseShortLE(9, 8);
var humi = payloadBytes.ParseShortLE(11, 2);
var illu = payloadBytes.ParseTwoBytesLE(13);
var batt = payloadBytes[8];
``

1technophile, 2020-12-01:
Thanks for the info.
Seems straightforward.

> If I can help to do something, I can try,

Yes, if you could give me examples of manufacturer data, service data, and the expected sensor's values. It would help.

Nikolay-Ch, 2021-01-12:
> Yes, if you could give me examples of manufacturer data, service data, and the expected sensor's values. It would help.
Look,
for the AbTemp - packet is in the iBeacon standard...
for the ABN01 and ABN03 - the packуе is formed according to the format [on this page](https://wiki.aprbrother.com/en/ABSensor.html). In gateway terminology - serviceData has started from 11th byte in the table on the page the link to which I gave above.

For ex packet for ABN03 looks like this:
`
{"id":"XX:XX:XX:XX:YY:ZZ","name":"abeacon_ECF2","rssi":-76,"distance":6.44788,"servicedata":"ab03zzyyxxxxxxxx64ebff7e000000"}
`

So, you need to parse it to something like this:
`
{
  "id": "XX:XX:XX:XX:YY:ZZ",
  "name": "abeacon_YYZZ",
  "rssi": -76,
  "distance": 6.44788,
  "temp": -2.625,
  "hum": 63.0,
  "lux": 0.0,
  "batt": 100
}
`

by XXYYZZ - I replace my device MAC-address

h2zero, 2021-12-28:
@Nikolay-Ch I may be able to add support for this but I need some more data if you're able to provide it. Some sample data of the advertising packets would be very helpful.

DigiH, 2022-03-04:
Hi @Nikolay-Ch , 

I've just created a test decoder for the April Brother N03 sensor. For testing change the decoder URL in platformio.ini to

`decoder = https://github.com/DigiH/decoder#ABN03-test`

For you above example you should now get

`{"id":"XX:XX:XX:XX:YY:ZZ","name":"abeacon_YYZZ","rssi":-76,"distance":6.44788,"servicedata":"ab03zzyyxxxxxxxx64ebff7e000000","brand":"AprilBrother","model":"N03","model_id":"ABSensor_N03","batt":100,"tempc":-2.625,"hum":63,"lux":0}`

Let us know how it works for you also with other measured values.

1technophile, 2023-04-10:
Integrated

closed 2023-04-10 by 1technophile