1technophile/OpenMQTTGateway #811: Sonoff RF Bridge R2 V1 not receiving any signal only in OMG

MarhyCZ, 2020-11-26:
**Describe the bug**
Hi, I have Sonoff RF Bridge, that I wanted to use only as a RF gateway, so this project seemed like a best option. However, in Tasmota or ESPurna, I am able to receive RF signals without any problems, for example in Tasmota I get `RfReceived":{"Sync":7700,"Low":270,"High":760,"Data":"01BDA1","RfKey":"None"}` for a simple doorbell button. 

However the OpenMQTTGateway doesnt seem to receive anything. It connects correctly to the Wi-Fi as I can see periodically the MQTT messages of ONLINE status. 

**To Reproduce**
1. Use rfbridge-firmware.bin version 0.9 (also tried 0.9.3 and 0.9.2)
2. Stock Sonoff RF Bridge, without custom firmware on the RF receiver. (Board Version 2017.11.23 R2 V1.0). Switch under the LED turned to ON after flashing.

**Expected behavior**
See some response in MQTT or Serial Console after pressing the button on simple RF doorbell as I received it in Tasmota and ESPurna without issues.

Is there some simple way to debug it? (sorry for such dumb answer) To maybe turn on higher level of debugging from the RF chip?

florian-athingstory, 2020-11-26:
Hi,

Just corrected a bug below:
https://github.com/1technophile/OpenMQTTGateway/pull/807

v0.9.4 should work

1technophile, 2020-11-26:
Hi,

Just corrected a bug below:
#807

v0.9.4 should work

MarhyCZ, 2020-11-26:
Holy cow, I am so blind (was searching for github issues with "sonoff" in the title). And this is the only version in 0.9 branch I have not tried. I will try it ASAP. and close this if it works. Sorry for spaming.

MarhyCZ, 2020-11-26:
Ok, The 0.9.4 works ok! And the dev branch with your patches also.

1technophile, 2020-11-26:
Thanks for the feedback !

closed 2020-11-28 by 1technophile

jcisio, 2020-12-12:
Hello,

I'm in the same situation. This it the first time I flash something using a serial connection. I used esptool to flash the 0.9.5 firmware (I downloaded `rfbridge-firmware.bin`) and only after that I've found this issue. I flashed again with 0.9.4 but nothing changes. Do I have to do anything before flashing?

My MQTT log:
```
home/OpenMQTTGateway_SRFB/LWT offline
home/OpenMQTTGateway_SRFB/LWT online
home/OpenMQTTGateway_SRFB/version v0.9.4
homeassistant/binary_sensor/2CF432EF564D/config {"stat_t":"home/OpenMQTTGateway_SRFB/LWT","name":"OpenMQTTGateway_SRFB","uniq_id":"2CF432EF564D","dev_cla":"connectivity","pl_on":"online","pl_off":"offline","pl_avail":"online","pl_not_avail":"offline"}
homeassistant/switch/2CF432EF564Drestart/config {"stat_t":"home/OpenMQTTGateway_SRFB/LWT","name":"restart OMG","uniq_id":"2CF432EF564Drestart","pl_on":"{\"cmd\":\"restart\"}","pl_avail":"online","pl_not_avail":"offline","cmd_t":"home/OpenMQTTGateway_SRFB/commands/MQTTtoSYS/config","device":{"name":"OpenMQTTGateway_SRFB","manufacturer":"OMG_community","sw_version":"v0.9.4","identifiers":["2CF432EF564D"]}}
homeassistant/switch/2CF432EF564Derase/config {"stat_t":"home/OpenMQTTGateway_SRFB/LWT","name":"erase OMG","uniq_id":"2CF432EF564Derase","pl_on":"{\"cmd\":\"erase\"}","pl_avail":"online","pl_not_avail":"offline","cmd_t":"home/OpenMQTTGateway_SRFB/commands/MQTTtoSYS/config","device":{"name":"OpenMQTTGateway_SRFB","manufacturer":"OMG_community","sw_version":"v0.9.4","identifiers":["2CF432EF564D"]}}
homeassistant/sensor/2CF432EF564DgatewaySRFB/config {"stat_t":"home/OpenMQTTGateway_SRFB/SRFBtoMQTT","name":"gatewaySRFB","uniq_id":"2CF432EF564DgatewaySRFB","val_tpl":"{{ value_json.value | is_defined }}","device":{"name":"OpenMQTTGateway_SRFB","manufacturer":"OMG_community","sw_version":"v0.9.4","identifiers":["2CF432EF564D"]}}
home/OpenMQTTGateway_SRFB/SYStoMQTT {"uptime":5,"version":"v0.9.4","freemem":41072,"rssi":-46,"SSID":"hainam","ip":"192.168.5.83","mac":"2C:F4:32:EF:56:4D","wifiprt":0,"modules":"SRFBHADiscovery"}
home/OpenMQTTGateway_SRFB/SYStoMQTT {"uptime":125,"version":"v0.9.4","freemem":42584,"rssi":-49,"SSID":"hainam","ip":"192.168.5.83","mac":"2C:F4:32:EF:56:4D","wifiprt":0,"modules":"SRFBHADiscovery"}
```
The RF Led is flashing but no data is sent to MQTT.

1technophile, 2020-12-12:
Hello,

Did you switched the receiver back ON with v0.9.4?

1technophile, 2020-12-12:
It may be good also to erase the flash before uploading the v0.9.4

jcisio, 2020-12-12:
The switch is at the ON position.

I did run a `esptool.py erase_flash` once before reflashing to get back the WiFi Management. However I flashed again with 0.9.5 (I didn't know about this issue). When I flashed with 0.9.4 I didn't run `erase_flash`. Now I'm struggling to connect to the device because with my poor soldering skill, the contact of the RX pinout was lost :( I tried with another RX pinout (see below) but it didn't work, maybe they are not the same RX.
[image]

Sorry, enough off-topic. I think I'm good to buy new RF bridge.

jcisio, 2020-12-13:
Well, I reopened all and tried again today. I only soldered TX / RX to save some space because the device could be powered by the micro USB. And flashing works, but no signal received.

Log:

```
  ~/data/projects/OpenMQTTGateway   development ±  esptool.py erase_flash
esptool.py v3.0
Found 1 serial ports
Serial port /dev/ttyUSB0
Connecting....
Detecting chip type... ESP8266
Chip is ESP8285
Features: WiFi, Embedded Flash
Crystal is 26MHz
MAC: 2c:f4:32:ef:56:4d
Uploading stub...
Running stub...
Stub running...
Erasing flash (this may take a while)...
Chip erase completed successfully in 1.4s
Hard resetting via RTS pin...

~/data/projects/OpenMQTTGateway   development ±  esptool.py write_flash -fs 1MB -fm dout 0x0 .pio/build/rfbridge/firmware.bin 
esptool.py v3.0
Found 1 serial ports
Serial port /dev/ttyUSB0
Connecting....
Detecting chip type... ESP8266
Chip is ESP8285
Features: WiFi, Embedded Flash
Crystal is 26MHz
MAC: 2c:f4:32:ef:56:4d
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Compressed 433248 bytes to 304889...
Wrote 433248 bytes (304889 compressed) at 0x00000000 in 27.1 seconds (effective 128.1 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```
Unplug/plug after each esptool command, then switch to ON. The binary is from the latest developpement (built with PlatformIO).

jcisio, 2020-12-13:
Then I flashed again with 0.9.4. No hope.
```
✘  ~/data/projects/OpenMQTTGateway   development ±  esptool.py erase_flash
esptool.py v3.0
Found 1 serial ports
Serial port /dev/ttyUSB0
Connecting....
Detecting chip type... ESP8266
Chip is ESP8285
Features: WiFi, Embedded Flash
Crystal is 26MHz
MAC: 2c:f4:32:ef:56:4d
Uploading stub...
Running stub...
Stub running...
Erasing flash (this may take a while)...
Chip erase completed successfully in 1.4s
Hard resetting via RTS pin...
 ~/data/projects/OpenMQTTGateway   development ±  esptool.py write_flash -fs 1MB -fm dout 0x0 ~/Downloads/rfbridge-firmware.bin 
esptool.py v3.0
Found 1 serial ports
Serial port /dev/ttyUSB0
Connecting....
Detecting chip type... ESP8266
Chip is ESP8285
Features: WiFi, Embedded Flash
Crystal is 26MHz
MAC: 2c:f4:32:ef:56:4d
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Compressed 431680 bytes to 303342...
Wrote 431680 bytes (303342 compressed) at 0x00000000 in 26.9 seconds (effective 128.4 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```

Log:
```
Dec 13 19:55:40 home/OpenMQTTGateway_SRFB/LWT online
Dec 13 19:55:40 home/OpenMQTTGateway_SRFB/version v0.9.4
Dec 13 19:55:40 rfbridge/OpenMQTTGateway_SRFB/LWT offline
Dec 13 19:55:40 rfbridge/OpenMQTTGateway_SRFB/version version_tag
Dec 13 19:56:07 home/OpenMQTTGateway_SRFB/LWT online
Dec 13 19:56:07 home/OpenMQTTGateway_SRFB/version v0.9.4
Dec 13 19:56:07 homeassistant/binary_sensor/2CF432EF564D/config {"stat_t":"home/OpenMQTTGateway_SRFB/LWT","name":"OpenMQTTGateway_SRFB","uniq_id":"2CF432EF564D","dev_cla":"connectivity","pl_on":"online","pl_off":"offline","pl_avail":"online","pl_not_avail":"offline"}
Dec 13 19:56:07 homeassistant/switch/2CF432EF564Drestart/config {"stat_t":"home/OpenMQTTGateway_SRFB/LWT","name":"restart OMG","uniq_id":"2CF432EF564Drestart","pl_on":"{\"cmd\":\"restart\"}","pl_avail":"online","pl_not_avail":"offline","cmd_t":"home/OpenMQTTGateway_SRFB/commands/MQTTtoSYS/config","device":{"name":"OpenMQTTGateway_SRFB","manufacturer":"OMG_community","sw_version":"v0.9.4","identifiers":["2CF432EF564D"]}}
Dec 13 19:56:07 homeassistant/switch/2CF432EF564Derase/config {"stat_t":"home/OpenMQTTGateway_SRFB/LWT","name":"erase OMG","uniq_id":"2CF432EF564Derase","pl_on":"{\"cmd\":\"erase\"}","pl_avail":"online","pl_not_avail":"offline","cmd_t":"home/OpenMQTTGateway_SRFB/commands/MQTTtoSYS/config","device":{"name":"OpenMQTTGateway_SRFB","manufacturer":"OMG_community","sw_version":"v0.9.4","identifiers":["2CF432EF564D"]}}
Dec 13 19:56:07 homeassistant/sensor/2CF432EF564DgatewaySRFB/config {"stat_t":"home/OpenMQTTGateway_SRFB/SRFBtoMQTT","name":"gatewaySRFB","uniq_id":"2CF432EF564DgatewaySRFB","val_tpl":"{{ value_json.value | is_defined }}","device":{"name":"OpenMQTTGateway_SRFB","manufacturer":"OMG_community","sw_version":"v0.9.4","identifiers":["2CF432EF564D"]}}
Dec 13 19:56:07 home/OpenMQTTGateway_SRFB/SYStoMQTT {"uptime":7,"version":"v0.9.4","freemem":41072,"rssi":-45,"SSID":"hainam","ip":"192.168.5.83","mac":"2C:F4:32:EF:56:4D","wifiprt":0,"modules":"SRFBHADiscovery"}
Dec 13 19:58:07 home/OpenMQTTGateway_SRFB/SYStoMQTT {"uptime":127,"version":"v0.9.4","freemem":42584,"rssi":-44,"SSID":"hainam","ip":"192.168.5.83","mac":"2C:F4:32:EF:56:4D","wifiprt":0,"modules":"SRFBHADiscovery"}
Dec 13 20:00:07 home/OpenMQTTGateway_SRFB/SYStoMQTT {"uptime":247,"version":"v0.9.4","freemem":42584,"rssi":-44,"SSID":"hainam","ip":"192.168.5.83","mac":"2C:F4:32:EF:56:4D","wifiprt":0,"modules":"SRFBHADiscovery"}
Dec 13 20:02:07 home/OpenMQTTGateway_SRFB/SYStoMQTT ```
Switch ON. Press a button on my 433 MHz remote, RF led blinking, nothing in MQTT.

herrlado, 2021-02-17:
exactly the same problem here. Flashed 0.9.5. Binary downloaded with esptool but also compiled with platformio and flashed. Also previously erased.  The red led blinks when I send some rf commands but nothing received to the mqtt. It is also R2 V1

herrlado, 2021-02-17:
OK, flashed development and it works. Found that the fix was in #807