1technophile/OpenMQTTGateway #812: Set CC1101 Frequency via MQTT

Legion2, 2020-11-26:
**Is your feature request related to a problem? Please describe.**
I have bought a CC1101 to control the blind at my home, which require a frequency of 433.42 Mhz while all other receiver require 433.92MHz. So I need to change the frequency depending on the device I want to control.

**Describe the solution you'd like**
It should be possible to specify the frequency in the json send via MQTT as an optional parameter. When the frequency is not given in the json it uses the Frequency defined by the macro `CC1101_FREQUENCY`. This also ensures backward compatibility.

**Describe alternatives you've considered**
1. Using two Gateways with different frequencies
2. Add a new command MQTT topic to set the frequency of CC1101, which then can also be used for receiving

NorthernMan54, 2021-01-16:
If have been playing with this, and am considering publishing a pull request that does this

1 - Adds a new field on received messages, the MHz of the message ( only visible when compiled for CC1101 )
2 - Adds a new optional field "MHz" when sending messages
3 - Sets the receive frequency to the most recent transmit frequency

Any thoughts?

1technophile, 2021-01-16:
> 1 - Adds a new field on received messages, the MHz of the message ( only visible when compiled for CC1101 )

Great

> 2 - Adds a new optional field "MHz" when sending messages

I would add "mhz" or "freq" as the other json command are in lower case

> 3 - Sets the receive frequency to the most recent transmit frequency

Nice idea.
Maybe it could be interesting to have also a way to configure the receiving frequency by mqtt without sending a message.

Legion2, 2021-01-16:
In #816 I implemented a frequency switching when sending with the Somfy protocol, after the transmission the frequency is reset to 443.92 MHz to allow receiving signals with PiLight. I think as long we don't have libraries that support protocols that use other frequencies we don't need a general switch via Json messages. The libraries decide based on the protocol which is the right frequency to send or receive.

@NorthernMan54 what is your use case for switching frequency?

NorthernMan54, 2021-01-16:
@Legion2 i have multiple ceiling fans on different frequencies.  Hampton Bay on 303, and GE on 315. All using zgatewayrf/rc-switch to control

As I’m working thru this I do realize that by enabling the switching I will lose the ability to receive messages for remote usage ( I was actually reviewing the cc1101 datasheet to see if it had a scanner mode, with fixed frequencies but I didn’t see that )

As a further use case I’m also pondering a mode where you can switch the receiver between protocol packages ( my use case is rc-switch and pilight ).  Use rc-switch during setup to grab the various remote codes and normally use pilight to receive sensor updates on 433.92.

Back to my proposal

am thinking the following 

1 - Adds a new field on received messages, the MHz of the message ( only visible when compiled for CC1101 ) “mhz”: 303

2 - Add a new message, to set receiver frequency ( optional, default to 433.92 ).  Ie “mhz”: 303

3 - Add a optional field to transmit message “mhz”

4 - after transmission of a message go back frequency specified in # 2

5 - at this time these changes would only work on the zgatewayrf module.

NorthernMan54, 2021-01-17:
Pull Request #847 submitted for review

1technophile, 2021-01-26:
PR integrated

closed 2021-01-26 by 1technophile