1technophile/OpenMQTTGateway #813: I only get servicedata from Xiaomi LYWSD03MMC

agoralive, 2020-11-26:
Before submitting a problem please check the troubleshooting section
https://docs.openmqttgateway.com/upload/troubleshoot.html

**Describe the bug**
I only get servicedata

**To Reproduce**
Steps to reproduce the behavior:
I really don't know, bacause I see other people are able to get information

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment (please complete the following information):**
* OpenMQTTGateway version used (V0.9.3, 0.8, development)
* Library version related to the problem you have (if you have troubles with RF provide the version of RCSwitch library)
* For IR and RF clarify if you tested with the basic examples given with these libraries

**Additional context**
Add any other context about the problem here.

This is the entire message I get from MQTT when trying to read data from a Xiaomi Temp/Hum sensor though BLE
{"id":"A4:C1:38:41:DB:7B","name":"LYWSD03MMC","rssi":-69,"distance":3.119501,"servicedata":"30585b05f97bdb4138c1a408"}

1technophile, 2020-11-27:
You will get the data every 10minutes if not flashed with the tutorial below : 
https://community.openmqttgateway.com/t/xiaomi-lywsd03mmc-temperature-and-humidity-sensor-esp32-or-hm10/1237?u=1technophile

With this tutorial you will get the data every minutes with less battery consumption.

agoralive, 2020-11-27:
Yeah, mine just goes like 20-30 minutes between publishing

THX

closed 2020-11-27 by agoralive

1technophile, 2020-11-27:
You can change it below for info : 
https://docs.openmqttgateway.com/use/ble.html#setting-the-number-of-scans-between-before-connect-attempt