1technophile/OpenMQTTGateway #821: Improve Home assistant integration

titilambert, 2020-12-03:
Before submitting a problem please check the troubleshooting section
https://docs.openmqttgateway.com/upload/troubleshoot.html

**Describe the bug**
All sensor/binary_sensor/... created by the home assistant autodiscovery are "entities" in home assistant. Those entities are not linked to a device. This is not mandatory but it's recommended.

**To Reproduce**
1. Enable HomeAssistant autodiscovery integration
2. Look at the mqtt message published ending by `/config`, ie:
```json
{
  "stat_t": "home/zarbi-b/BTtoMQTT/582D34400672",
  "name": "CLEARGRASSTRHKPA-pres",
  "uniq_id": "582D34400672-CLEARGRASSTRHKPA-pres",
  "dev_cla": "pressure",
  "val_tpl": "{{ value_json.pres | is_defined }}",
  "unit_of_meas": "kPa"
}
```

**Expected behavior**
The config json published should be
```
```json
{
  "stat_t": "home/zarbi-b/BTtoMQTT/582D34400672",
  "name": "CLEARGRASSTRHKPA-pres",
  "uniq_id": "582D34400672-CLEARGRASSTRHKPA-pres",
  "dev_cla": "pressure",
  "val_tpl": "{{ value_json.pres | is_defined }}",
  "unit_of_meas": "kPa"
  "device": {"connections": [["mac", "02:5b:26:a8:dc:12"]],
                  "manufacturer": "cleargrass",
                  "model": "CGP1W",
                  "name": "Qingping Temp RH Barometer",
                  "via_device": "OpenMQTTGateway",
}
```
Note: "via_device" is the "hostname" of the board (in my case it's a esp32 board which connect bluetooth devices)  

**Environment **
* OpenMQTTGateway version used: V0.9.5
* ZGatewayBT
* ZmqttDiscovery

**Question**
Are you open to this change/improvement ?
If yes, I will post a PR for that.
Thanks for this project

1technophile, 2020-12-03:
Thanks for proposing your help!

Yes I welcome this PR.

Here is a first attempt from @francisp
https://community.openmqttgateway.com/t/creating-devices-in-home-assistant/1121

Among others I think we need to change the variable name `child_device` to `gateway_entity` or equivalent

titilambert, 2020-12-03:
> Among others I think we need to change the boolean `child_device` to `gateway_entity` or equivalent. THis boolean indicates if the entity must be linked to the gateway.

Agreed !

1technophile, 2021-01-02:
done

closed 2021-01-02 by 1technophile