1technophile/OpenMQTTGateway #824: Home Assistant shows binary.sensor as disconnected

jlaitinen81, 2020-12-04:
I'm experiencing issues with my Open MQTT Gateway on an ESP32 Wroom with firmware 0.9.5.

The exposed binary.sensor shows as disconnected since last night and even if I power cycled the ESP, the status does not change. However, I can see mqtt messages coming nicely through in my Mosquitto MQTT broker, and the ESP also looks to be perfectly well connected to my Unifi network (static IP) and packets flow both ways.

How can I force an update on the binary sensor status? I have a node red automation alerting me constantly of a disconnected device.

1technophile, 2020-12-04:
If you take a look to the broker, does it publish regularly a SYStoMQTT message?
If yes does this one fit with your switch definition?

jlaitinen81, 2020-12-04:
Yes it does:

{"uptime":37082,"version":"v0.9.5","freemem":174800,"rssi":-38,"SSID":"xxx","ip":"yyy.yyy.yyy.yyy","mac":"zz:zz:zz:zz:zz","wifiprt":0,"lowpowermode":0,"interval":55555,"scanbcnct":10,"modules":"BTHADiscovery"}

How do I compare this to a switch definition? In my HA I can see binary_sensor.openmqttgateway_esp32_ble_7 as disconnected.

1technophile, 2020-12-04:
So the gateway is publishing regularly.

The switch status is based on this message:
[image]
LWT = online

So if you see this one online, all is fine on the gateway to broker side.

We should look to the binary sensor definition published for auto discovery below:
[image]

And see if it fits with the online message (topic/payload)

jlaitinen81, 2020-12-04:
Thanks! This is under homeassistant/binary_sensor:

[image]

and this is the one under home:

[image]

jlaitinen81, 2020-12-05:
I'm confused but on a higher level 😀. The above pasted info looks correct to me but the binary sensor is still disconnected. Where is the link from this data to the actual binary sensor in HA? My binary sensor is called binary_sensor.openmqttgateway_esp32_ble_7 and I can not find any reference to that.

1technophile, 2020-12-05:
In Home Assistant->Configuration->Devices
Click on the gateway
Click on MQTT info
[image]

You should be able to check the topic definition for the gateway and see if it fits with the broker inputs.
If not I advise you to delete the device, delete the topics with MQTT explorer and restart the gateway.

closed 2022-01-09 by stale[bot]