1technophile/OpenMQTTGateway #827: HASS auto discovery is always enabled

Legion2, 2020-12-05:
**Describe the bug**
HASS auto discovery is turned on by default in all platformio envs.
https://github.com/1technophile/OpenMQTTGateway/blob/417eb99d40d4b61b2487641b954813ea8d7ebf8f/platformio.ini#L128-L135

Disabling HASS auto discovery is only possible by creating an own env without using the existing ones as base.

**To Reproduce**
Steps to reproduce the behavior:
1. Upload any predefined configuration or use one as base
2. HASS auto discovery is enabled

**Expected behavior**
HASS auto discovery should be disabled by default and should not be enabled in the base configurations of platformio.
HASS auto discovery should only be enabled when needed.

**Environment (please complete the following information):**
* OpenMQTTGateway version: development

**Additional context**
I don't use Home Assistant.

1technophile, 2020-12-05:
This comes from the hypothesis that most of the users are using home assistant.
maybe we should activate/deactivate it at runtime

kuchel77, 2021-01-28:
So as to enable HASS auto discovery with MQTT you have to uncomment ZmqttDiscovery in User_config.h And enable discovery on your MQTT integration definition in HASS.

https://docs.openmqttgateway.com/integrate/home_assistant.html#auto-discovery

Just for some context, the current situation isn't consistent with the documentation. I don't want to use auto discovery for my OpenMQTTGateway device, because for some reason my RF device gets random signals, it then fills my HA with random RF switches, which then maxes out my HomeKit integration (150) devices, whereas I use it with a static MQTT setup and when it matches the data I want, it acts accordingly.

If either the documentation or the -D in the platform.ino could be adjust to be consistent that would be great.

1technophile, 2021-01-29:
> If either the documentation or the -D in the platform.ino could be adjust to be consistent that would be great.

You mean aligning the doc so as to indicate that the discovery is enabled by default.

Added to that I will add a configuration parameter to deactivate autodiscovery at runtime.

1technophile, 2021-01-30:
Implemented in the linked PR

Legion2, 2021-01-30:
Can we disable the discovery using a custom platformio config?

1technophile, 2021-01-31:
I will try with  `'-UZmqttDiscovery="HADiscovery"'`

Legion2, 2021-01-31:
I turned it off in my platformio config, by using `${env.build_flags}` instead of `${com-esp.build_flags}`, this also turns fo simpleReceiving, but I don't need this too.

@1technophile do I need `${libraries.wifimanager}` if I have a static Wifi config with DHCP?

1technophile, 2021-01-31:
> @1technophile do I need ${libraries.wifimanager} if I have a static Wifi config with DHCP?

No you don't need it

1technophile, 2021-01-31:
> I will try with '-UZmqttDiscovery="HADiscovery"'

This works also

closed 2021-01-31 by 1technophile