1technophile/OpenMQTTGateway #832: (RF Bridge) Problem with OTA update via PlatformIO

jcisio, 2020-12-12:
Hi,

I flashed my RF Bridge v2 first with 0.9.5 binary, then with 0.9.4 binary. Now I'm switching to PlatformIO to do OTA update but it does not seem to work.

```
...
Building in release mode
Compiling .pio/build/rfbridge/src/main.ino.cpp.o
Retrieving maximum program size .pio/build/rfbridge/firmware.elf
Checking size .pio/build/rfbridge/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [====      ]  39.5% (used 32388 bytes from 81920 bytes)
Flash: [======    ]  56.3% (used 429100 bytes from 761840 bytes)
Configuring upload protocol...
AVAILABLE: espota, esptool
CURRENT: upload_protocol = espota
Uploading .pio/build/rfbridge/firmware.bin
00:20:16 [DEBUG]: Options: {'esp_ip': '192.168.5.83', 'host_ip': '0.0.0.0', 'esp_port': 8266, 'host_port': 19328, 'auth': 'OTAPASSWORD', 'image': '.pio/build/rfbridge/firmware.bin', 'spiffs': False, 'debug': True, 'progress': True}
00:20:16 [INFO]: Starting on 0.0.0.0:19328
00:20:16 [INFO]: Upload size: 433248
00:20:16 [INFO]: Sending invitation to: 192.168.5.83
00:20:26 [ERROR]: No Answer
*** [upload] Error 1
============================================== [FAILED] Took 13.09 seconds ==============================================

Environment    Status    Duration
-------------  --------  ------------
rfbridge       FAILED    00:00:13.086
========================================= 1 failed, 0 succeeded in 00:00:13.086 =========================================
The terminal process "pio 'run', '--target', 'upload'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.
```

Any idea? Thanks.

1technophile, 2020-12-12:
Hi,

I encountered the same issue a few weeks ago with pio. Could you try with arduino ide ?

jcisio, 2020-12-12:
I tried PlatformIO because I don't have any idea how to do with Arduino IDE, I never used it (I never used PlatformIO, neither, but it was straightforward).

I'm going to try with it then, but not right now. It's past midnight (I remark that you are from France, too ;-) ).

jcisio, 2020-12-13:
Well, finally I tried Arduino IDE too. But it seems that without using that IDE for a serial port flash, impossible to do OTA.

```
Executable segment sizes:
IROM   : 391204          - code in flash         (default or ICACHE_FLASH_ATTR) 
IRAM   : 28224   / 32768 - code in IRAM          (ICACHE_RAM_ATTR, ISRs...) 
DATA   : 1628  )         - initialized variables (global, static) in RAM/HEAP 
RODATA : 3328  ) / 81920 - constants             (global, static) in RAM/HEAP 
BSS    : 26864 )         - zeroed variables      (global, static) in RAM/HEAP 
Sketch uses 424384 bytes (44%) of program storage space. Maximum is 958448 bytes.
Global variables use 31820 bytes (38%) of dynamic memory, leaving 50100 bytes for local variables. Maximum is 81920 bytes.
Serial port not selected.
```

jcisio, 2020-12-13:
Well, it looks like the device does not have any port open. Could it be a reason? I flashed with the OMG binaries. I am in upload mode (in normal mode, there is a "not enough space" error).

1technophile, 2020-12-13:
With Arduino IDE you should see the boards available by OTA on Tools->Port menu
[image]

You may not see them due to firewall/network issues

1technophile, 2020-12-13:
>  It's past midnight (I remark that you are from France, too ;-) ).

Yes I'am, have a good night!

jcisio, 2020-12-13:
"Port" was disabled, it appears now (I don't know why, I tried several thing). But I think I got the same result as with PIO.

```
Executable segment sizes:
IROM   : 391204          - code in flash         (default or ICACHE_FLASH_ATTR) 
IRAM   : 28224   / 32768 - code in IRAM          (ICACHE_RAM_ATTR, ISRs...) 
DATA   : 1628  )         - initialized variables (global, static) in RAM/HEAP 
RODATA : 3328  ) / 81920 - constants             (global, static) in RAM/HEAP 
BSS    : 26864 )         - zeroed variables      (global, static) in RAM/HEAP 
Sketch uses 424384 bytes (44%) of program storage space. Maximum is 958448 bytes.
Global variables use 31820 bytes (38%) of dynamic memory, leaving 50100 bytes for local variables. Maximum is 81920 bytes.
01:56:10 [ERROR]: No Answer
01:56:25 [ERROR]: No Answer.
```
That's the message in upload mode. The same response "No Answer" as in PIO. In normal mode, there is "Not Enough Space" error.

I think I have tried enough stuffs. Good night. "La nuit porte conseil".

jcisio, 2020-12-13:
Well, in the MQTT:
```
Dec 13 01:12:35 home/OpenMQTTGateway_SRFB/SYStoMQTT {"uptime":610,"version":"v0.9.4","freemem":42240,"rssi":-40,"SSID":"hainam","ip":"192.168.5.83","mac":"2C:F4:32:EF:56:4D","wifiprt":0,"modules":"SRFBHADiscovery"}
```
freemem 42K if it is flash memory, it could be an explication.

jcisio, 2020-12-13:
Ok freemem is the remaining heap, so it has nothing to do with. I was thinking that the 0.9.5 firmware stayed next to 0.9.4 and there was only 42K flash memory left, but it should not be possible as we write to 0x0000 every time. 

On the way for another input.

jcisio, 2020-12-13:
Maybe the log when I flashed the binaries is useful:

```
$ esptool.py  write_flash -fs 1MB -fm dout 0x0  rfbridge/rfbridge-firmware.bin
esptool.py v3.0
Found 1 serial ports
Serial port /dev/ttyUSB0
Connecting....
Detecting chip type... ESP8266
Chip is ESP8285
Features: WiFi, Embedded Flash
Crystal is 26MHz
MAC: 2c:f4:32:ef:56:4d
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Compressed 432448 bytes to 304301...
Wrote 432448 bytes (304301 compressed) at 0x00000000 in 27.0 seconds (effective 128.2 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...

$ esptool.py erase_flash
esptool.py v3.0
Found 1 serial ports
Serial port /dev/ttyUSB0
Connecting....
Detecting chip type... ESP8266
Chip is ESP8285
Features: WiFi, Embedded Flash
Crystal is 26MHz
MAC: 2c:f4:32:ef:56:4d
Uploading stub...
Running stub...
Stub running...
Erasing flash (this may take a while)...
Chip erase completed successfully in 1.6s
Hard resetting via RTS pin...

$  esptool.py  write_flash -fs 1MB -fm dout 0x0  rfbridge/rfbridge-firmware.bin
esptool.py v3.0
Found 1 serial ports
Serial port /dev/ttyUSB0
Connecting....
Detecting chip type... ESP8266
Chip is ESP8285
Features: WiFi, Embedded Flash
Crystal is 26MHz
MAC: 2c:f4:32:ef:56:4d
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Compressed 432448 bytes to 304301...
Wrote 432448 bytes (304301 compressed) at 0x00000000 in 27.0 seconds (effective 128.2 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...

$ esptool.py  write_flash -fs 1MB -fm dout 0x0  ~/Downloads/rfbridge-firmware.bin
esptool.py v3.0
Found 1 serial ports
Serial port /dev/ttyUSB0
Connecting....
Detecting chip type... ESP8266
Chip is ESP8285
Features: WiFi, Embedded Flash
Crystal is 26MHz
MAC: 2c:f4:32:ef:56:4d
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Compressed 431680 bytes to 303342...
Wrote 431680 bytes (303342 compressed) at 0x00000000 in 26.9 seconds (effective 128.4 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```
With OTA flash, the bridge should be in normal mode, right? I think it's specific to the firmware, and in upload mode (or "pairing mode", because the switch is "ON") it can't connect to the device.

1technophile, 2020-12-13:
In OTA flash I do not think you need to change the switch, let me try to OTA mine with Arduino IDE

1technophile, 2020-12-13:
[image]

I'm having authentication errors, I need to dig into it

jcisio, 2020-12-13:
Maybe use esptool with `--trace` to see if it really is the authentication step?

Sorry that I no longer can give more test, because after several fails with OTA, I managed to flash with serial port (see https://github.com/1technophile/OpenMQTTGateway/issues/811) but still failed with OMG I switched to another firmware.

1technophile, 2020-12-14:
Good you got it working, but did not get why not with OMG, that's strange. I will verify it again before releasing the [v0.9.6](https://github.com/1technophile/OpenMQTTGateway/milestone/17)

1technophile, 2021-03-12:
OTA is now working on RFBridge:

```
Uploading .pio\build\rfbridge\firmware.bin
12:45:01 [DEBUG]: Options: {'esp_ip': '192.168.', 'host_ip': '0.0.0.0', 'esp_port': 8266, 'host_port': 50633, 'auth': '', 'image': '.pio\\build\\rfbridge\\firmware.bin', 'spiffs': False, 'debug': True, 'progress': True}
12:45:01 [INFO]: Starting on 0.0.0.0:50633
12:45:01 [INFO]: Upload size: 340144
12:45:01 [INFO]: Sending invitation to: 192.168.
Authenticating...OK
12:45:01 [INFO]: Waiting for device...

Uploading: [                                                            ] 0% 
Uploading: [                                                            ] 0%
Uploading: [=                                                           ] 0% 
Uploading: [=                                                           ] 1%
Uploading: [=                                                           ] 1% 
Uploading: [=                                                           ] 2% 
Uploading: [==                                                          ] 2% 
Uploading: [==                                                          ] 3% 
Uploading: [==                                                          ] 3% 
Uploading: [==                                                          ] 3% 
Uploading: [===                                                         ] 4% 
Uploading: [===                                                         ] 4% 
Uploading: [===                                                         ] 5% 
Uploading: [===                                                         ] 5% 
Uploading: [====                                                        ] 6% 
Uploading: [====                                                        ] 6% 
Uploading: [====                                                        ] 6% 
Uploading: [====                                                        ] 7% 
Uploading: [=====                                                       ] 7% 
Uploading: [=====                                                       ] 8% 
Uploading: [=====                                                       ] 8% 
Uploading: [=====                                                       ] 9% 
Uploading: [======                                                      ] 9% 
Uploading: [======                                                      ] 9% 
Uploading: [======                                                      ] 10% 
Uploading: [======                                                      ] 10% 
Uploading: [=======                                                     ] 11% 
Uploading: [=======                                                     ] 11% 
Uploading: [=======                                                     ] 12% 
Uploading: [=======                                                     ] 12% 
Uploading: [========                                                    ] 12% 
Uploading: [========                                                    ] 13% 
Uploading: [========                                                    ] 13% 
Uploading: [========                                                    ] 14% 
Uploading: [=========                                                   ] 14% 
Uploading: [=========                                                   ] 15% 
Uploading: [=========                                                   ] 15%
Uploading: [==========                                                  ] 15% 
Uploading: [==========                                                  ] 16% 
Uploading: [==========                                                  ] 16% 
Uploading: [==========                                                  ] 17%
Uploading: [===========                                                 ] 17% 
Uploading: [===========                                                 ] 18% 
Uploading: [===========                                                 ] 18% 
Uploading: [===========                                                 ] 18% 
Uploading: [============                                                ] 19% 
Uploading: [============                                                ] 19% 
Uploading: [============                                                ] 20% 
Uploading: [============                                                ] 20% 
Uploading: [=============                                               ] 21% 
Uploading: [=============                                               ] 21% 
Uploading: [=============                                               ] 21% 
Uploading: [=============                                               ] 22% 
Uploading: [==============                                              ] 22% 
Uploading: [==============                                              ] 23% 
Uploading: [==============                                              ] 23% 
Uploading: [==============                                              ] 24% 
Uploading: [===============                                             ] 24% 
Uploading: [===============                                             ] 24% 
[... 154 lines cut ...]
Uploading: [=======================================================     ] 91% 
Uploading: [=======================================================     ] 91% 
Uploading: [=======================================================     ] 92% 
Uploading: [========================================================    ] 92% 
[... cut ...]
Uploading: [=========================================================== ] 97% 
Uploading: [=========================================================== ] 98% 
Uploading: [=========================================================== ] 98% 
Uploading: [=========================================================== ] 99% 
Uploading: [============================================================] 99% 
Uploading: [============================================================] 100% Done...

12:45:09 [INFO]: Waiting for result...
12:45:10 [INFO]: Result: OK
```

closed 2021-03-12 by 1technophile

mb-software, 2021-04-01:
Hi, I have (still) the same problem: No answer to Authentication. I am using the sourcecode of the 0.96 beta realease with platformIO. Uploaded successfully via Serial connection to the rf bridge. But OTA fails:

10:30:50 [DEBUG]: Options: {'esp_ip': '192.168.178.112', 'host_ip': '0.0.0.0', 'esp_port': 8266, 'host_port': 13934, 'auth': 'OTAPASSWORD', 'image': '.pio\\build\\sonoff-rfbridge-direct\\firmware.bin', 'spiffs': False, 'debug': True, 'progress': True}
10:30:50 [INFO]: Starting on 0.0.0.0:13934
10:30:50 [INFO]: Upload size: 420000
10:30:50 [INFO]: Sending invitation to: 192.168.178.112
Authenticating...FAIL
10:30:50 [ERROR]: No Answer to our Authentication

Am I missing something? Do I have to change some configuration, like OTAPASSWORD or something? I left everything on default and only added this to my env:

board_build.flash_mode = dout
upload_protocol = espota
upload_port = 192.168.178.112
upload_flags =
  --auth=OTAPASSWORD
  --port=8266

1technophile, 2021-04-01:
Could you check your firewall? It may be an issue if it is blocking the connection.

mb-software, 2021-04-01:
Already tried to disable the Windows firewall completely, but still the same

1technophile, 2021-04-02:
Could you try without the port?

mb-software, 2021-04-02:
It makes no difference. Same "no answer to our authentification"

1technophile, 2021-04-02:
If uou have an arduino IDE with OTA installed. Can you see the rfbridge in the port list ?

mb-software, 2021-04-02:
Currently I do not have Arduino IDE installed. Maybe I can install it to give it a try, I will let you know the result. 
Thanks for your support and great work with omg

mb-software, 2021-04-06:
Just installed Arduino IDE and the rfbridge appears right away in the ports list.