1technophile/OpenMQTTGateway #833: Hassio 433toMQTT - put value to topic suggestion

AlexMKX, 2020-12-13:
The Home assistant doesn't have conditions for a values. 
When I describe sensors like this : 
```
  - platform: mqtt
    state_topic: "homeassistant/rfgw/433toMQTT"
    off_delay: 10
    payload_on: "999999"
    payload_available: "999999"
    name: "motion_sensor"
    value_template: "{{ value_json.value | string }}"

  - platform: mqtt
    state_topic: "homeassistant/rfgw/433toMQTT"
    off_delay: 10
    payload_on: "888888"
    payload_available: "888888"
    name: "door sensor"
    value_template: "{{ value_json.value | string }}"
```

i'm getting errors in log : " No matching payload found for entity " 
Which caused by the same topic is used for 433toMQTT.
In order to solve this - the value (which identifies the specific 433 button) should be appended to topic. To have topics like these : 
homeassistant/rfgw/433toMQTT/999999 for sensor 1
and 
homeassistant/rfgw/433toMQTT/888888 for sensor 2 respectively

proposed patch : 
ZgatewayRF.ino@void RFtoMQTT() : 
change        
`pub(subjectRFtoMQTT, RFdata);  `
to 
```
#  ifdef HASSIO_433_MQTT
      char subject [1024];
      memset(&subject,0,sizeof(subject));
      char mqttval[64];
      memset (mqttval,0,sizeof(mqttval));
      itoa(MQTTvalue,mqttval,10);
      sprintf(subject,"%s/%s",subjectRFtoMQTT,mqttval);
      pub(subject, RFdata);
# else
      pub(subjectRFtoMQTT, RFdata);
# endif 
```

1technophile, 2020-12-13:
This is available here https://github.com/1technophile/OpenMQTTGateway/blob/a7575b2c3a6852614a64f045a31bd21af2b4a82f/main/User_config.h#L356

By uncommenting the line.

We should add some docs about it.

1technophile, 2021-01-02:
Added to the docs

closed 2021-01-02 by 1technophile