1technophile/OpenMQTTGateway #836: Official combability for ESP32-WROOM-32 (Adafruit ESP32 Feather)

samip5, 2020-12-14:
**Is your feature request related to a problem? Please describe.**
No, but I want it to be officially supported.

**Describe the solution you'd like**
The Feather to be added to the board listing.

**Describe alternatives you've considered**
N/A

1technophile, 2020-12-14:
Would you be interested in proposing it as a PR, you need to add the board definition to platformio.ini, you may take the [esp32dev-ble](https://github.com/1technophile/OpenMQTTGateway/blob/a83f725bada72eb0c5e055f526c5c1a9a4a47871/platformio.ini#L319) as an example

samip5, 2020-12-15:
Yes, I think I could do that, but I still need to try to get the platformio IDE installed.

1technophile, 2021-01-29:
Could you test with the board the PR linked please?

samip5, 2021-01-30:
> Could you test with the board the PR linked please?

Could you please eloborate as to what do you mean? Test it how?

1technophile, 2021-01-30:
Verify that after the upload to the board following the new env, it worse properly.
* No crash
* The led flash when receiving BLE signal

samip5, 2021-01-31:
> * The led flash when receiving BLE signal

I have nothing to test that with?

1technophile, 2021-01-31:
There is a big chance that you have BLE devices around you that will be detected by the gateway.
Even if they are not sensors, the gateway detect smartphone or other bLE device

closed 2021-02-21 by 1technophile