1technophile/OpenMQTTGateway #838: How is 868Mhz supported?

gerardo15, 2020-12-28:
Hello.

I got myself one of these

[image]

I connected and wired as its explained and flashed nodemcuv2-rf-cc1101, set up wifi and mosquitto.

I get "home/OpenMQTTGateway_ESP8266_RF-CC1101/LWT online" on my broker

but nothing else, does not react to any of my 868mhz, am i missing something or this is simple not supported?

1technophile, 2021-01-02:
> does not react to any of my 868mhz devices, am i missing something or cc1101 868mhz is not supported

Hello,

For decoding a signal you need to consider:
* the frequency itself
* and the protocol

A module can be able to listen/emit on your device frequency, but if there is no algorithm for decoding the protocol of your remote, it will not be coded.
The protocol can be very easy to decode (PT2262 protocol) to very complex protocol (LORA), they are most of the time associated with a frequency but they are not dependent.

So in your case I will look after your device protocol first, what are they models/references?

gerardo15, 2021-01-02:
> Hello,
>
> For decoding a signal you need to consider:
>
> * the frequency itself
> * and the protocol
>
> A module can be able to listen/emit on your device frequency, but if there is no algorithm for [...]

Thanks for your aswer, Basicly what i want to do is.. i just moved to a new place and the owner left around 15 PIR sensors outdoors and indoor, those look like really high end PIR sensors, not like the 433mhz i bought on aliexpress.. i have been looking online, and it seems i have been the only person trying to do this haha.. i thought it was going to be as easier as the 433mhz, just changing the cc1101 for a 868mhz as i don't need to send any commands just read when those signals are fired...

Maybe if its to complicated i just leave this untouched and move on... what do you think? am i even close?

1technophile, 2021-01-02:
Could you share a picture of the model/make and eventually a picture of the inside?

gerardo15, 2021-01-02:
Of course!

Here you go!, Really appreciate taking your time for this!

[image]
[image]

closed 2022-01-09 by stale[bot]