1technophile/OpenMQTTGateway #839: UART / SERIAL to MQTT

DanBendavid, 2020-12-29:
I am using a Arduino Nano to measure 3 CT sensors.
A Serial (@ 115200 baud) message is generated by the nano as following : 
"MSG:2565, Vrms:232.56, P1:565, E1:456, P2:565, E2:456, P3:565, E3:456"

I would like to see a Esp8266 or Esp32 being able to generate the MQTT message.
Connection between Nano and Esp is done via RX-> TX and TX->RX  (with 3.3V shift ).

Thank you

DanBendavid, 2020-12-30:
Hello,
Here is a piece of code that can be helpfull  function to get the serial 

#define EMONTX_PORT Serial

boolean input_get(JsonDocument &data)
{
  boolean gotLine = false;
  boolean gotData = false;
  String line;

  if(input_string.length() > 0) {
    line = input_string;
    input_string = "";
    gotLine = true;
  }
  // If data received on serial
  else if (Serial.available()) {
    // Could check for string integrity here
    line = Serial.readStringUntil('\n');
    gotLine = true;
  }

  if(gotLine)
  {
    // Get rid of any whitespace, newlines etc
    line.trim();

    int len = line.length();
    if(len > 0) 
    {
      DEBUG.printf_P(PSTR("Got '%s'\n"), line.c_str());

      for(int i = 0; i < len; i++)
      {
        String name = "";

        // Get the name
        while (i < len && line[i] != ':') {
          name += line[i++];
        }

        if (i++ >= len) {
          break;
        }

        // Get the value
        String value = "";
        while (i < len && line[i] != ','){
          value += line[i++];
        }

        
        if(name.length() > 0 && value.length() > 0)
        {
          // IMPROVE: check that value is only a number, toDouble() will skip white space and and chars after the number
          data[name] = value.toDouble();
          gotData = true;
        }
      }
    }
  }

  // Append some system info
  if(gotData) {
    data[F("freeram")] = ESPAL.getFreeHeap();
    data[F("srssi")] = WiFi.RSSI();
    data[F("psent")] = packets_sent;
    data[F("psuccess")] = packets_success;

    last_datastr.clear();
    serializeJson(data, last_datastr);
  }

  return gotData;
}

void loop {

// handle connections ....

 StaticJsonDocument<512> data;
 boolean gotInput = input_get(data);
  
    if(gotInput) {
      event_send(data);
    }
  }

void event_send(String &json)
{
  StaticJsonDocument<512> event;
  deserializeJson(event, json);
  event_send(event);
}

void event_send(JsonDocument &event)
{
   mqtt_publish(event);
}

DanBendavid, 2020-12-30:
@1technophile Florian,  I am currently trying to copy/adapt  the RS232 template. Not yet succeeded

1technophile, 2020-12-31:
ok, if you have questions, do not hesitate.

DanBendavid, 2021-01-05:
Hereby my implementation : The RS232 works fine at 9600baud from Arduino Nano to Gateway. I have not tried to do the MQTT to Arduino (it could be interesting to recalibrate the Emon)
In the Arduino Nano Emon I have this Serial Print : Serial.print(F("0"));// The prefix for the Serial message

    Serial.print(F("{\"Msg\" : ")); Serial.print(emontx.Msg);
    Serial.print(F(", \"Vrms\": ")); Serial.print(emontx.Vrms*0.01);
    
    if (CT1) { Serial.print(F(",\"CT1\":{\"RealPower\": ")); Serial.print(emontx.P1); }
    if (CT1) { Serial.print(F(", \"Wh\": ")); Serial.print(emontx.E1); }
    
    if (CT2) { Serial.print(F("}, \"CT2\":{\"RealPower\": ")); Serial.print(emontx.P2); }
    if (CT2) { Serial.print(F(", \"Wh\": ")); Serial.print(emontx.E2); }

    if (CT3) { Serial.print(F("}, \"CT3\":{\"RealPower\": ")); Serial.print(emontx.P3); }
    if (CT3) { Serial.print(F(", \"Wh\": ")); Serial.print(emontx.E3); }
    
    Serial.print(F("} }"));

Then in the OpenGateway, hereby  my parameters : 

Config_RS232.h

//Setup for RS232
#define MAX_INPUT   300 // how much serial data we expect
#define RS232Baud   9600 // The serial connection Baud
#define RS232Pre    "00" // The prefix for the RS232 message
#define RS232Post   "\n" // The postfix for the RS232 message
#define RS232InPost '\r' // Hacky way to get last character of postfix for incoming

I have  added in ZgatewayRS232.ino : 
#include "config_RS232.h"
#include <ArduinoLog.h>

dont know why it was working without

1technophile, 2021-01-26:
Would you want to submit a PR with your work?

DanBendavid, 2021-02-17:
Hi all,
One month after,the gateway is very reliable . Thank you so much Florian (@1technophile ) for this fabulous job.  
The can probably do a PR for the next release and use the Config_RS232.h parameters as default and to add the 2 lines in ZgatewayRS232.ino

1technophile, 2021-02-17:
Nice, thanks for the feedback!