1technophile/OpenMQTTGateway #841: No BLE scans from esp32-olimex-gtw-ble-eth if powered from PoE (no USB)

ndoo, 2020-12-30:
Before submitting a problem please check the troubleshooting section
https://docs.openmqttgateway.com/upload/troubleshoot.html

**Describe the bug**

I only see `home/OpenMQTTGateway_ESP32_OLM_GTWE/SYStoMQTT` publishing every 120 seconds, but no BTtoMQTT publish, if powered from PoE.

Board is Olimex ESP32-POE-EA.

BTtoMQTT works as expected when board is powered by USB, regardless of whether Ethernet is connected before or after ESP32 is fully booted up into sketch.

**WARNING: Do not connect USB while Ethernet with PoE is connected, if computer is grounded; the board is non-isolated, so board GND is half of PoE voltage, around 24V for 802.3af. Connecting a grounded PC will connect +24V to computer ground.**

**To Reproduce**
1. Download code tagged `OpenMQTTGateway-0.9.5`
2. Load folder in PlatformIO
3. Edit platformio.ini: Uncomment `default_envs = esp32-olimex-gtw-ble-eth`; Under `[env:esp32-olimex-gtw-ble-eth]` section, add `'-DMQTT_SERVER="a.b.c.d"'`
4. I added `'-DZmqttDiscovery="HADiscovery"'` as well
5. Disconnect USB, connect Ethernet with PoE power supply
6. Observe SYStoMQTT messages starting at 6s, then updates every approx. 120s

**Expected behavior**
Behaviour with or without USB power should be the same, i.e. BLE scans take place

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment (please complete the following information):**
* OpenMQTTGateway version used: v0.9.5
* Library version related to the problem you have: As per tagged release platformio.ini
* For IR and RF clarify if you tested with the basic examples given with these libraries

**Additional context**
As above

* You should not have a compilation error if you use the versions of the libraries linked into the libraries folder, this badges show you the state of the compilation 
[image]
* If you are not sure this is a bug or an enhancement post your question to the forum below
[image]

1technophile, 2021-01-02:
> I only see home/OpenMQTTGateway_ESP32_OLM_GTWE/SYStoMQTT publishing every 120 seconds, but no BTtoMQTT publish, if powered from PoE.

Interesting, unfortunately I don't have the board to reproduce. Would it be possible to put the log level to [LOG_LEVEL_TRACE](https://github.com/1technophile/OpenMQTTGateway/blob/e4865749b97cfd8761e80fbf9bd26ff825f3be13/main/User_config.h#L386) and post here a serial monitor extract ?

ndoo, 2021-01-05:
Attached a log as requested, MAC addresses were replaced with xx:xx:xx:xx:xx:xx

[OpenMQTTGateway_ESP32_OLM_GTWE_LOG_LEVEL_TRACE_20210106.txt](https://github.com/1technophile/OpenMQTTGateway/files/5772760/OpenMQTTGateway_ESP32_OLM_GTWE_LOG_LEVEL_TRACE_20210106.txt)

Apologies it took so long, had to switch over to my laptop (to avoid shorting the ESP32 to ground)

1technophile, 2021-01-26:
Seems that the scan is getting trouble when on POE, could you try a basic BLE scan example to see if you get something when on POE:
https://github.com/h2zero/NimBLE-Arduino/tree/master/examples/Refactored_original_examples/BLE_scan

mgx0, 2021-09-30:
has this ever been identified?

closed 2022-01-09 by stale[bot]