1technophile/OpenMQTTGateway #845: Current version of the Gateway from develop branch does not support ClearGrass CGG1 and CGP1W sensors

Nikolay-Ch, 2021-01-11:
Hi,

I bought two sensors from Clearglass: CGG1 and CGP1W
But now my OpenMqttGateway with version from develop branch does not parse Advertised Service Data from this sensors.
Now it sends raw ServiceData to MQTT server:

`
{"id":"xx:xx:xx:xx:xx:xx","name":"Qingping Temp & RH M","rssi":-74,"distance":5.270232,"servicedata":"8816xxxxxxxxxxxx0104f000da00020164"}
{"id":"xx:xx:xx:xx:xx:xx","name":"Qingping Temp & RH M","rssi":-68,"distance":2.799256,"servicedata":"5858480b8cxxxxxxxxxxxxd54f777b2a010000fe6b2188"}
{"id":"xx:xx:xx:xx:xx:xx","name":"Qingping Temp & RH M","rssi":-69,"distance":3.119501,"servicedata":"5858480b99xxxxxxxxxxxxf5893478e3010000892e142c"}
{"id":"xx:xx:xx:xx:xx:xx","name":"Qingping Temp & RH M","rssi":-67,"distance":2.509097,"servicedata":"5858480b9fxxxxxxxxxxxx452035087b0100006a6e04ae"}
`

`
{"id":"xx:xx:xx:xx:xx:xx","name":"Qingping Temp RH Barometer","rssi":-78,"distance":7.85288,"servicedata":"0809xxxxxxxxxxxx01040b01ce000702ea2602015a"}
{"id":"xx:xx:xx:xx:xx:xx","name":"Qingping Temp RH Barometer","rssi":-64,"distance":1.795564,"servicedata":"0809xxxxxxxxxxxx01040801cf000702e92602015a"}
{"id":"xx:xx:xx:xx:xx:xx","name":"Qingping Temp RH Barometer","rssi":-72,"distance":4.287841,"servicedata":"0809xxxxxxxxxxxx01041201d5000702e92602015a"}
`

where XX and xx - I replace MAC-addresses of my devices

Regards,
Nikolay

jlaitinen81, 2021-01-13:
I have the same issue. Newly ordered Qingping Temp RH sensors are not being parsed, whereas older ones with same model (CGG1) just work straight out of the box.

1technophile, 2021-01-13:
Hi,

Could you share your servicedata examples?

Nikolay-Ch, 2021-01-13:
> Hi,
> 
> Could you share your servicedata examples?

I show it in the my first message... I thing you try to setup sensor by first bytes in the "SerbiceData" param, but you parse MAC-address of the device. In my message I replace device MAC-address by 'XX' chars...

1technophile, 2021-01-14:
Sorry, was a question for @jlaitinen81

jlaitinen81, 2021-01-14:
> Hi,
> 
> Could you share your servicedata examples?

{"id":"xx:xx:xx:xx:xx:xx","name":"Qingping Temp & RH M","rssi":-86,"distance":16.54559,"servicedata":"5858480b1156b511342d582afe36f3a10000005d54a9b6"}

jlaitinen81, 2021-01-14:
`8810 089211342d58 01 04 e2 00  79 01 0201 62
xxxx   MAC-addr.  xx xx  temp   hum  fw?  batt
temp = 0*256+e2H = 226 -> 22.6 degree
hum  = 01*256 + 79H = 377 -> 37.7%
batt = 62H = 98%
about firmware relase not sure (current: 1.02) `

Nikolay-Ch, 2021-01-14:
> Found this:
> 
> [image]
> 
> In: [atc1441/ATC_MiThermometer#135](https://github.com/atc1441/ATC_MiThermometer/issues/135)

8810 - it is a different device... our device - 8816

Nikolay-Ch, 2021-01-22:
I add pullrequest to fix recognition ClearGlass sensors...
https://github.com/1technophile/OpenMQTTGateway/pull/849

closed 2021-01-29 by 1technophile