1technophile/OpenMQTTGateway #852: Trying to get OWSOO Door Sensor to work - parsing error ?

supertick, 2021-01-27:
**Description**
Hello, and thanks for your exceptional work ! 

I am trying to figure out what I think might be a bug or mis-configuration.
I have Esp32 dev board running the latest (master) OpenMQTTGateway.
I'm trying to get the following door sensors to work properly.
https://www.amazon.com/gp/product/B07MG3LYPC/ref=ppx_yo_dt_b_asin_title_o02_s00?ie=UTF8&psc=1
[image]

using a rx3400 receiver - https://www.sparkfun.com/datasheets/Wireless/General/MO-RX3400.pdf
[image]

The switch is supposed to be using the  EV1527 protocol.
From what I read its a preamble + 20 address bit + 4 state bits.

What I receive on the mqtt topics is the following:

`homeassistant/switch/30AEA4F33CB016383/config payload:{"stat_t":"home/OpenMQTTGateway_ESP32_RF/433toMQTT","name":"16383","uniq_id":"30AEA4F33CB016383","pl_on":"16383","cmd_t":"home/OpenMQTTGateway_ESP32_RF/commands/MQTTto433","device":{"name":"OpenMQTTGateway_ESP32_RF","model":"[\"RF\",\"HADiscovery\"]","manufacturer":"OMG_community","sw_version":"version_tag","identifiers":["30AEA4F33CB0"]}} 
home/OpenMQTTGateway_ESP32_RF/433toMQTT payload:{"value":16383,"protocol":29,"length":14,"delay":211} 
homeassistant/switch/30AEA4F33CB04095/config payload:{"stat_t":"home/OpenMQTTGateway_ESP32_RF/433toMQTT","name":"4095","uniq_id":"30AEA4F33CB04095","pl_on":"4095","cmd_t":"home/OpenMQTTGateway_ESP32_RF/commands/MQTTto433","device":{"name":"OpenMQTTGateway_ESP32_RF","model":"[\"RF\",\"HADiscovery\"]","manufacturer":"OMG_community","sw_version":"version_tag","identifiers":["30AEA4F33CB0"]}} 
home/OpenMQTTGateway_ESP32_RF/433toMQTT payload:{"value":4095,"protocol":29,"length":12,"delay":211} 
homeassistant/switch/30AEA4F33CB08191/config payload:{"stat_t":"home/OpenMQTTGateway_ESP32_RF/433toMQTT","name":"8191","uniq_id":"30AEA4F33CB08191","pl_on":"8191","cmd_t":"home/OpenMQTTGateway_ESP32_RF/commands/MQTTto433","device":{"name":"OpenMQTTGateway_ESP32_RF","model":"[\"RF\",\"HADiscovery\"]","manufacturer":"OMG_community","sw_version":"version_tag","identifiers":["30AEA4F33CB0"]}} 
home/OpenMQTTGateway_ESP32_RF/433toMQTT payload:{"value":8191,"protocol":29,"length":13,"delay":211} 
homeassistant/switch/30AEA4F33CB04095/config payload:{"stat_t":"home/OpenMQTTGateway_ESP32_RF/433toMQTT","name":"4095","uniq_id":"30AEA4F33CB04095","pl_on":"4095","cmd_t":"home/OpenMQTTGateway_ESP32_RF/commands/MQTTto433","device":{"name":"OpenMQTTGateway_ESP32_RF","model":"[\"RF\",\"HADiscovery\"]","manufacturer":"OMG_community","sw_version":"version_tag","identifiers":["30AEA4F33CB0"]}} 
home/OpenMQTTGateway_ESP32_RF/433toMQTT payload:{"value":4095,"protocol":29,"length":12,"delay":211} 
homeassistant/switch/30AEA4F33CB016383/config payload:{"stat_t":"home/OpenMQTTGateway_ESP32_RF/433toMQTT","name":"16383","uniq_id":"30AEA4F33CB016383","pl_on":"16383","cmd_t":"home/OpenMQTTGateway_ESP32_RF/commands/MQTTto433","device":{"name":"OpenMQTTGateway_ESP32_RF","model":"[\"RF\",\"HADiscovery\"]","manufacturer":"OMG_community","sw_version":"version_tag","identifiers":["30AEA4F33CB0"]}} 
home/OpenMQTTGateway_ESP32_RF/433toMQTT payload:{"value":16383,"protocol":29,"length":14,"delay":211} 
homeassistant/switch/30AEA4F33CB08191/config payload:{"stat_t":"home/OpenMQTTGateway_ESP32_RF/433toMQTT","name":"8191","uniq_id":"30AEA4F33CB08191","pl_on":"8191","cmd_t":"home/OpenMQTTGateway_ESP32_RF/commands/MQTTto433","device":{"name":"OpenMQTTGateway_ESP32_RF","model":"[\"RF\",\"HADiscovery\"]","manufacturer":"OMG_community","sw_version":"version_tag","identifiers":["30AEA4F33CB0"]}} 
home/OpenMQTTGateway_ESP32_RF/433toMQTT payload:{"value":8191,"protocol":29,"length":12,"delay":211} 
homeassistant/switch/30AEA4F33CB04095/config payload:{"stat_t":"home/OpenMQTTGateway_ESP32_RF/433toMQTT","name":"4095","uniq_id":"30AEA4F33CB04095","pl_on":"4095","cmd_t":"home/OpenMQTTGateway_ESP32_RF/commands/MQTTto433","device":{"name":"OpenMQTTGateway_ESP32_RF","model":"[\"RF\",\"HADiscovery\"]","manufacturer":"OMG_community","sw_version":"version_tag","identifiers":["30AEA4F33CB0"]}} 
home/OpenMQTTGateway_ESP32_RF/433toMQTT payload:{"value":4095,"protocol":29,"length":12,"delay":211} 
homeassistant/switch/30AEA4F33CB08191/config payload:{"stat_t":"home/OpenMQTTGateway_ESP32_RF/433toMQTT","name":"8191","uniq_id":"30AEA4F33CB08191","pl_on":"8191","cmd_t":"home/OpenMQTTGateway_ESP32_RF/commands/MQTTto433","device":{"name":"OpenMQTTGateway_ESP32_RF","model":"[\"RF\",\"HADiscovery\"]","manufacturer":"OMG_community","sw_version":"version_tag","identifiers":["30AEA4F33CB0"]}} 
home/OpenMQTTGateway_ESP32_RF/433toMQTT payload:{"value":8191,"protocol":29,"length":12,"delay":211} 
homeassistant/switch/30AEA4F33CB04095/config payload:{"stat_t":"home/OpenMQTTGateway_ESP32_RF/433toMQTT","name":"4095","uniq_id":"30AEA4F33CB04095","pl_on":"4095","cmd_t":"home/OpenMQTTGateway_ESP32_RF/commands/MQTTto433","device":{"name":"OpenMQTTGateway_ESP32_RF","model":"[\"RF\",\"HADiscovery\"]","manufacturer":"OMG_community","sw_version":"version_tag","identifiers":["30AEA4F33CB0"]}} 
home/OpenMQTTGateway_ESP32_RF/433toMQTT payload:{"value":4095,"protocol":29,"length":12,"delay":211} 
homeassistant/switch/30AEA4F33CB04095/config payload:{"stat_t":"home/OpenMQTTGateway_ESP32_RF/433toMQTT","name":"4095","uniq_id":"30AEA4F33CB04095","pl_on":"4095","cmd_t":"home/OpenMQTTGateway_ESP32_RF/commands/MQTTto433","device":{"name":"OpenMQTTGateway_ESP32_RF","model":"[\"RF\",\"HADiscovery\"]","manufacturer":"OMG_community","sw_version":"version_tag","identifiers":["30AEA4F33CB0"]}} 
home/OpenMQTTGateway_ESP32_RF/433toMQTT payload:{"value":4095,"protocol":29,"length":12,"delay":211} 
home/OpenMQTTGateway_ESP32_RF/SYStoMQTT payload:{"uptime":108963,"version":"version_tag","freemem":242592,"rssi":-42,"SSID":"xdentrite","ip":"10.0.0.193","mac":"30:AE:A4:F3:3C:B0","wifiprt":0,"modules":["RF","HADiscovery"]} 
homeassistant/switch/30AEA4F33CB016383/config payload:{"stat_t":"home/OpenMQTTGateway_ESP32_RF/433toMQTT","name":"16383","uniq_id":"30AEA4F33CB016383","pl_on":"16383","cmd_t":"home/OpenMQTTGateway_ESP32_RF/commands/MQTTto433","device":{"name":"OpenMQTTGateway_ESP32_RF","model":"[\"RF\",\"HADiscovery\"]","manufacturer":"OMG_community","sw_version":"version_tag","identifiers":["30AEA4F33CB0"]}} 
home/OpenMQTTGateway_ESP32_RF/433toMQTT payload:{"value":16383,"protocol":29,"length":13,"delay":196} 
homeassistant/switch/30AEA4F33CB04095/config payload:{"stat_t":"home/OpenMQTTGateway_ESP32_RF/433toMQTT","name":"4095","uniq_id":"30AEA4F33CB04095","pl_on":"4095","cmd_t":"home/OpenMQTTGateway_ESP32_RF/commands/MQTTto433","device":{"name":"OpenMQTTGateway_ESP32_RF","model":"[\"RF\",\"HADiscovery\"]","manufacturer":"OMG_community","sw_version":"version_tag","identifiers":["30AEA4F33CB0"]}} 
home/OpenMQTTGateway_ESP32_RF/433toMQTT payload:{"value":4095,"protocol":29,"length":12,"delay":196} `

**This is only one switch**, should I expect the "name" field to be consistent ?
Is the protocol supposed to be consistent too  ?  Is this a parsing error ?
Do I need to do more configuration ?  (currently I've only uncommented **default_envs = esp32dev-rf** in platformio.ini

Thanks for your response.

Regards,
  Noob

1technophile, 2021-01-29:
Hello,

Is it several open signal followed, or only one, if several could you post the published payloads for one opening action ?
Would be nice to have it also with time, [MQTT explorer ](http://mqtt-explorer.com/) may help you for this.

supertick, 2021-01-30:
I'll get more structured data and correlate it to events ...
But I can verify it is a single switch, and only sending a heartbeat, no open/close data..
And that's why I think the RF parsing is having problems, because the "name" changes.
I'll create a more structured and detailed log shortly.
Thanks for the response.

supertick, 2021-01-30:
[image]

I think this might illustrate the problem.
There is only one switch.

The first pink box is the door opening ...
9 seconds later the second green box the door closes...

Opening or closing currently generates four MQTT messages each..

My OpenMQTTGateway's identifier is 30AEA4F33CB0

- "name" I would expect the same value for the same switch.  
- not sure why their are two duplicate messages published to two separate topics
  2 msgs go to topic homeassistant/switch/30AEA4F33CB05145193/config
  2 msgs go to topic home/OpenMQTTGateway_ESP32_RF/433toMQTT
- value I would expect to change, "name" and "protocol" I would not expect to change  
  
EV1527 protocol should have preamble + 20 address bit + 4 state bits
the 20 bit address should not be changing if the 4 state bits changes 

It makes me think this is a parsing problem - just not sure where to dig next

1technophile, 2021-01-31:
From these screenshots there is one value 5145193 for door opening and one for door closing 511, I don't know why they are repeated each twice, but at least you have one value per action.
Seems more a problem of duplicate than an issue with the library parsing.

> 2 msgs go to topic homeassistant/switch/30AEA4F33CB05145193/config

Note that these messages are used for Home assistant auto discovery, they could be deactivated by commenting Zmqttdiscovery lines into platformio.ini. Or in a near future by an MQTT command #856

supertick, 2021-01-31:
Shouldn't the name be the same if the "address" == "name" ?

1technophile, 2021-01-31:
> Shouldn't the name be the same if the "address" == "name" ?

ZgatewayRF/RCSwitch are just reporting a value corresponding to the signal received without any interpretation of address, status (contrary to Pilight or RF2), so this is normal even if it seems confusing. This is due to the fact that we are trying to fit into the HASS convention with raw informations.

supertick, 2021-01-31:
Thanks for your responses 1technophile ;)
I'm appreciating the fact there are probably a bajillion protocols and details of smart devices.  The fact this project tries manages to present some form of standard useful interface is a testament to all the great work that has been done.

In order to provide a useful interface it would seem (in my noobie mind) you would want to uniquely identify "a device" or is that not the case ?  Is it only important to uniquely identify events ?

If its important to identify devices the messages to me seem wrong because the two messages where
"name":"5145193"  and "name":"511"
came from the same sensor.

Am I thinking about it incorrectly?

supertick, 2021-02-03:
Another example ....
[image]
I only have one switch ;)
Thanks for the mqtt explorer tip

closed 2022-01-09 by stale[bot]