1technophile/OpenMQTTGateway #854: Strange Total-Crash Failure after a couple of days/weeks

Dattel, 2021-01-30:
Hi, 

i have a strange behaviour with OMG in Version 0.9.5 on a ESP8266 NodeMCU Board
My attached devices are:
-BME280
-RF-Sender
-RF-Receiver

I'm using my own prod_env.ini with the following compiler settings:

```
[com-esp01]
lib_deps =
  ${com-esp.lib_deps}
build_flags =
  ${env.build_flags}
  ${platformio.build_flags}
  -DMQTT_MAX_PACKET_SIZE=1024
  '-DsimpleReceiving=true'
  '-DTRACE=0'

[env:nodemcuv2-OpenMediaGateway]
platform = ${com.esp8266_platform}
board = nodemcuv2
lib_deps =
  ${com-esp01.lib_deps}
  ${libraries.rc-switch}
  ${libraries.bme280}
build_flags = 
  '-DESPWifiManualSetup="1"'
  '-DWifiManager_password="******"'
  '-DOMG_VERSION="v0.9.5"'
  '-Dwifi_ssid="*********"'
  '-Dwifi_password="*********"'
  '-DMQTT_USER="*********"'
  '-DMQTT_PASS="*********"'
  '-DMQTT_SERVER="*********"'
  '-DMQTT_PORT="1884"'
  '-Dota_password="*********"'  
```

After a variable time period (days/weeks) the board hangs... If i try to capture some informations using the serial port in need to adjust the baudrate to 74880 and all i got is:

```
 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 3584, room 16 
tail 0
chksum 0xb0
csum 0xb0
v2843a5ac
~ld
⸮⸮⸮
```

The only way to fix these issue-state is to flash a sample blink sketch with Arduino-IDE using the **"Erase all Flash Content"** and after that reflashing OMG. If i flash without "Erase all Flash Content", i can't leave these state of death.

I also have these issue with other ESP-01 based boards only with BME280 Sensors attached. But here the frequence of death is much less frequently.

Any Idea's how to fix that?? Or any clue, where i have to dig a bit deeper??

Thanks
D

Dattel, 2021-01-30:
saw some similar problems here with a suggest for solution:

https://github.com/esp8266/Arduino/issues/2414#issuecomment-696125575

1technophile, 2021-01-30:
Could you try to isolate the crash reason by deactivating the modules one by one please?
I would start first by the BME280.

Alternatively you could try to update the BME280 library (platformio.ini file):
`bme280 = SparkFun BME280@2.0.4`
to
`bme280 = SparkFun BME280@2.0.9`

Dattel, 2021-01-30:
i will have a try with BME@2.0.9
disable the BME will not make much sence for me because my ESP01 boards only uses the BME Sensors... so if i disable these sensor, the OMG instances will be useless for me :-)

i just compiled and uploaded the version to my NODEMCU board, which is also using the RF-Receiver/sender... i'll try to keep an eye on that and post feedback.
Thanks for your quick reply

1technophile, 2021-01-31:
> disable the BME will not make much sence for me because my ESP01 boards only uses the BME Sensors... so if i disable these sensor, the OMG instances will be useless for me :-)

This is just to target the defective module, not to ask for a permanent removing

Dattel, 2021-01-31:
Thanks, i completely understand... 

i've updated my NodeMCU with RF and BME to BME-Lib2.0.9 and give it a try. If it dosn't work, i'll remove the BME from these board completely for testing.

Dattel, 2021-02-10:
Updating to BME @2.0.8 seems not to fix.... 
I've got some power outage yesterday - i'm not sure, if that accelerates the occurance of the failure.

BME@2.0.9 was not possible in a quickly manner since there are some changes in the initialization process that seems to require a bit of changes in OMG to open the I2C.

Nevertheless, i removed the BME-Lib and the sensor completely from the board and give it a try for testing purpose.
keep you updated here...

Albertowue, 2021-02-17:
Hi, 
I have exactly the same problem (After a variable time period the board hangs). My setup is:
 
OMG in Version 0.9.5 development
ESP8266 NodeMCU Board
espressif8266@2.6.3
- DHT22
-RF-Sender
-RF-Receiver

Also for me the only way to unlock the esp is to use the Dattel technique, 
I have a different sensor, so the problem isn't the dht or BME module.

Dattel, 2021-02-24:
Today my device crashes again - this time only with RF-Send/Receive Modules..
I assume, it's nothing wrong with the BME Lib...

so 14 days to crash this time...

1technophile, 2021-02-24:
Thanks for the feedback.

Did you checked your power supply voltage and stability?

Also I found out that there is an issue with wifi manual setup and the v0.9.5,  could you erase the flash, comment [this line ](https://github.com/1technophile/OpenMQTTGateway/blob/4431af462f46336d391c67c8121e45167aa57db5/main/main.ino#L899) of code, and upload it.

The issue is that the wifi.begin() function may be called too close to another wifi.begin resulting in a continuous reconnect loop
I don't know if this is producing your problem or not but it may be a track to follow.

1technophile, 2021-02-24:
@Albertowue did you get a serial monitor output (115200) to see what's going on.
Also for you both it could be nice to put the [log level ](https://github.com/1technophile/OpenMQTTGateway/blob/4431af462f46336d391c67c8121e45167aa57db5/main/User_config.h#L388) to LOG_LEVEL_TRACE

Dattel, 2021-02-25:
Hi @1technophile, thanks for your fast reply..

i just updated my sources to the latest development-sources, commented out your line, but didn't change the loglevel, since the device is running offsite only with power supply. As i mentioned earlier, if the device crashes, i only got rubbish on the serial and the rate changes magically to 74880 - so i assume that there is nothing more usefull coming in then.

Will give it a try now...

1technophile, 2021-03-02:
Hello,

I have done an update of the RCSwitch libraries, 
could you change the [line 87 of platformio.ini](https://github.com/1technophile/OpenMQTTGateway/blob/ebd6bb8031e59847e66895f45335ed764a89348b/platformio.ini#L87) file
from

`rc-switch = https://github.com/1technophile/rc-switch.git#385a7e0`

to

`rc-switch = https://github.com/1technophile/rc-switch.git#0e0d210`

and see if you get a better stability?

Dattel, 2021-03-03:
sure... 
I updated my device with the latest git-sources and changed to rc-switch #0e0d210...
But this time, i don't commented out your [suggested](https://github.com/1technophile/OpenMQTTGateway/blob/4431af462f46336d391c67c8121e45167aa57db5/main/main.ino#L899)
line from your last post.

Albertowue, 2021-03-04:
I just updated to the latest git sources and changed line 87 of platformio.ini as indicated. The system just crashed after 10 days of operation.

hvorragend, 2021-03-10:
I am using OMG with BLE discovery. I've tried the current code on several ESP32 and I recognized crashes after several days. 
 I will try https://github.com/1technophile/OpenMQTTGateway/commit/be5a93a91a90b3d3b5b4fff022c691c8b3d89d71#diff-4446afd728a4f34cbcddc306a9cb6be845d1a61c216076a295683bcc9c106724 soon.

1technophile, 2021-03-10:
>  I've tried the current code on several ESP32 

Could you indicate what do you mean by the current code?

1technophile, 2021-03-10:
> I just updated to the latest git sources and changed line 87 of platformio.ini as indicated. The system just crashed after 10 days of operation.

Hmm, I don't understand how you can have a system crash after 10 days of operation if you took a change that was provided 2 days before your message :-)

hvorragend, 2021-03-10:
> > I've tried the current code on several ESP32
> 
> Could you indicate what do you mean by the current code?

I am using the current dev-branch. Not the release.

1technophile, 2021-03-10:
> I am using the current dev-branch. Not the release.

ok, there were several corrections made these last days, could you try the last dev?

Albertowue, 2021-03-16:
> > I just updated to the latest git sources and changed line 87 of platformio.ini as indicated. The system just crashed after 10 days of operation.
> 
> Hmm, I don't understand how you can have a system crash after 10 days of operation if you took a change that was provided 2 days before your message :-)

Excuse my bad English, I meant that before making the change you suggested it was blocked again. For now with your changes I have reached 12 days of operation and without blocks

1technophile, 2021-03-16:
No problem, thanks for the feedback!

1technophile, 2021-03-16:
I'm closing the issue if it reappears do not hesitate to comment here

closed 2021-03-16 by 1technophile

Dattel, 2021-04-10:
sorry, the problem isn't fixed... two devices reached lockdown-state a couple of days before - i didn't realized that so i can't say, how long they really last without crashing.

1technophile, 2021-04-12:
@Dattel which version are you using? By lockdown state, you mean that it didn't restart itself?
Do you have still the BME detached?

Dattel, 2021-04-13:
with "lockdown state" i mean, i have to reflash the device including "Erase UserData & Wifi settings"..

i have two "locked down" devices:

- One with BME attached. (ESP01)
- and the other (NODEMCU Dev-Board ESP8266) without the BME attached and also not included the libs in the Build-section. (just with rc-switch)

I'm using the latest git-checkout from the 03.03.2021 with rcswitch#0e0d210

Dattel, 2021-04-15:
I reflashed both devices 2 day's ago - the NODEMCU Dev-Board lasts for 2 days and stucks again....
So i just pulled the latest git sources and reflashed again and will have a look, if these version is more stable now.