1technophile/OpenMQTTGateway #858: Concurrency bug on esp32 at publishing ble mqtt messages

csiki2, 2021-01-31:
**Describe the bug**
On ESP32 the BLE publishing core can use the same PubSubClient object at the same time as the "main" core that regularly publishes the status of the gateway.
Typical result is that half ble - half status mqtt message arrives to the MQTT server:

2021-01-31 19:43:13 WARNING (MainThread) [homeassistant.components.mqtt] Can't decode payload b'{"uptime":3362,"version":"version_tag","freemem":142364,"rssi":-51,"SSID":"ASUS_CS_2G","ip":"192.168.50.227","mac":"3C:61:05:1D:BC:98","wifiprt":0\x9a\x01\x00,home/home_presence/OpenMQTTGateway_ESP32_BLE{"id":"C4:58:1F:E1:04:51","name":"tkr","man' on home/OpenMQTTGateway_ESP32_BLE/SYStoMQTT with encoding utf-8 (for <Job HassJobType.Callback <function MqttSensor._subscribe_topics.<locals>.message_received at 0x7f66e84611f0>>)

I am using Home Assistant with Mosquitto (I don't think that this is relevant, still might be needed).

**To Reproduce**
Steps to reproduce the behavior:
<pre>
#  define ESPWifiManualSetup true // don't know whether this is important, probably not
</pre>
Using ESP32 (esp32dev-ble only), for config_BT.h (TimeBtwRead is the imporatant one):
<pre>
#  define Scan_duration 5000 // probably not important, I checked whether it has any impact on the memory usage
#  define TimeBtwRead 1 // this __forces__ the gateway for non-stop scanning, greatly increase the chance of the concurrency issue
#  define ScanBeforeConnect 60 // to reflect the above changes and have a connect only at every 5 mins
#  define pubKnownBLEServiceData true // don't know whether this is important, but it's true at me
#  define pubBLEManufacturerData true // don't know whether this is important, but it's true at me
#  define pubBLEServiceUUID true // don't know whether this is important, but it's true at me
</pre>

Probably not important, but in ZgatewayBT.ino, BLEscan (why don't we have define for this, I don't know):
<pre>
  pBLEScan->setActiveScan(false); //active scan uses more power, but get results faster -- was true
</pre>

I have 4 MiJia LYWSDCGQ close to the gateway and the tracked devices make the possibily of the occurence about once per 3 hours, probably some more frequent sender would further increase the chance.

**Expected behavior**
Always meaningful mqtt messages on the server side. :)

**Environment (please complete the following information):**
* OpenMQTTGateway version used (origin/development)
* Simpe esp32dev board

**Additional context**
If I add a "pthread_mutex_t m_pthread_mutex;" to the PubSubClient class and defend:
boolean PubSubClient::publish(const char* topic, const uint8_t* payload, unsigned int plength, boolean retained)
boolean PubSubClient::publish_P(const char* topic, const uint8_t* payload, unsigned int plength, boolean retained) // probably not important
with it, the problem is resolved.
This is probably more of a proof of concept than a final good solution though:
<pre>
boolean PubSubClient::publish(const char* topic, const uint8_t* payload, unsigned int plength, boolean retained) {
    pthread_mutex_lock(&m_pthread_mutex);
    if (connected()) {
        if (this->bufferSize < MQTT_MAX_HEADER_SIZE + 2+strnlen(topic, this->bufferSize) + plength) {
            // Too long
            pthread_mutex_unlock(&m_pthread_mutex);
            return false;
        }
        // Leave room in the buffer for header and variable length field
        uint16_t length = MQTT_MAX_HEADER_SIZE;
        length = writeString(topic,this->buffer,length);

        // Add payload
        uint16_t i;
        for (i=0;i<plength;i++) {
            this->buffer[length++] = payload[i];
        }

        // Write the header
        uint8_t header = MQTTPUBLISH;
        if (retained) {
            header |= 1;
        }
        bool res = write(header,this->buffer,length-MQTT_MAX_HEADER_SIZE);
        pthread_mutex_unlock(&m_pthread_mutex);
        return res;
    }
    pthread_mutex_unlock(&m_pthread_mutex);
    return false;
}
</pre>
No ESP32 check, didn't check whether there are other possible concurrency issues around this class, a cheap spinlock probably better (don't know the esp32 good enough) etc.

Regards,
  Csiki

1technophile, 2021-02-05:
Thanks for reporting such a detailled report, it will help.
I will dig into it

1technophile, 2021-02-05:
I have added a task suspend when the SYStoMQTT is processed, if you want you can test the linked PR

csiki2, 2021-02-06:
I might miss something here:
As far as I can see this change will make sure that the SYStoMQTT message goes out, but after that we might get a first half SYStoMQTT second half BT message if the suspend happened in the middle of a BT message creation:
the SYStoMQTT overwrites the full buffer then the BT task will continue to write to the buffer after the resume.

csiki2, 2021-02-06:
Since PubSubClient is not part of this lib directly, you could make a bridge class like:
<pre>
class OMGPubSubClient : protected PubSubClient {
protected:
#ifdef(ESP32)
  pthread_mutex_t _mutex;
#endif
public:
// bridge only the really used contructors and methods here
...
};

// publish sample
boolean OMGPubSubClient::publish(const char* topic, const uint8_t* payload, unsigned int plength, boolean retained) {
#ifdef(ESP32)
pthread_mutex_lock(&_mutex);
#endif
bool res = PubSubClient::publish(topic, payload, plength, retained);
#ifdef(ESP32)
pthread_mutex_unlock(&_mutex);
#endif
  return res;
}
</pre>

and then use that with the necessary guards.
You might also need to reach the mutex Lock and Unlock from outside of the object, using a suspend to a core and then sending a message can end up at a deadlock (so, Lock - Suspend - Unlock will be the necessary order).

csiki2, 2021-02-06:
Alternatively, a small (8KB) lockless queue could be used and only the main core would use PubSubClient.
Probably this is the cleanest solution, but also somewhat memory heavy (esp32 tend to have enough memory) and would be the biggest change code-wise.

1technophile, 2021-02-06:
Well seen, there is still a risk of suspending an "in progress" ble scan publication with this solution.

> Let me know, if you are interested in this solution, probably I will try it out.

I'm really interested, a queue was experimented on  #804 for info so as to solve issues on TLS + BLE.

csiki2, 2021-02-08:
I will create a pull request this week.
I will need extra test for you as I don't have non-esp32 boards, hopefully I won't brake anything.
Also, I found another (smaller) issues, should I report them separately?

1technophile, 2021-02-08:
> I will need extra test from you as I don't have non-esp32 boards

No problem, I will be happy to help you

> Also, I found another (smaller) issues, should I report them separately?

Yes please

csiki2, 2021-02-11:
The pull request
https://github.com/1technophile/OpenMQTTGateway/pull/867

I am new with this platfromio thing. Is there a way to run your lint checks at my client (so I won't need exta commits next time :) )?

1technophile, 2021-02-12:
>The pull request
#867

Thanks!

> I am new with this platfromio thing. Is there a way to run your lint checks at my client (so I won't need exta commits next time :) )?

yes, here is the tip:

So as to format automaticaly your document you have to add the "clang-Format" extension to VSCode, once done, you can format the code by doing a right click into the code file window and clicking "Format document".

closed 2021-02-26 by 1technophile