1technophile/OpenMQTTGateway #864: Can't disable HADiscovery at compile

ProfEngr, 2021-02-08:
**Describe the bug**
HADiscovery is commented "//" in user_config.h but still gets compiled in by VS Code.

**Expected behavior**
HADiscovery should not be compiled into the bin.

**Screenshots**

[image]
[image]

**Environment (please complete the following information):**
* OpenMQTTGateway version used (V0.9.5)

1technophile, 2021-02-08:
Yes, the doc has been updated in the [development branch ](https://github.com/1technophile/OpenMQTTGateway/commit/2d377a1c3f8350bf423452fc2dcadea5f9fe9ed5) but not yet published.

Could you try to add:
`'-UZmqttDiscovery="HADiscovery"'`

To your environment or remove 

`  '-DZmqttDiscovery="HADiscovery"'`

from [platformio.ini](https://github.com/1technophile/OpenMQTTGateway/blob/14fedfdf53dd1e4d9b28ff186f8b64a280d62206/platformio.ini#L144)

1technophile, 2021-04-24:
Closing, for info HAdiscovery can be disable also at [runtime ](https://docs.openmqttgateway.com/use/gateway.html#auto-discovery)

closed 2021-04-24 by 1technophile