1technophile/OpenMQTTGateway #870: FRQ: MQTT-Controlled watchdog

AlexMKX, 2021-02-16:
Hello, I've noticed that sometimes device hangs or non-responsible.

It would be good idea to have a MQTT-controlled watchdog, which works like this:
1. When it enabled, the hardware watchdog for ESP32 is setup with minutes-grade interval  (10-minutes for example).
2. If watchdog is not reset in due time - the board is reset by hardware WDT

In order to reset watchdog timer - the certain MQTT command should be published. 
So in any cases (including WIFI issues) - the board will definitely auto-reset in 10 minutes to give a new try.

Those boards, which has no hardware WDT can be expanded with 555-based WDT, tied to GPIO. In this case, the MQTT command should pulse GPIO in order to pat the WDT.

1technophile, 2021-04-03:
Could you test the [v0.9.6beta](https://github.com/1technophile/OpenMQTTGateway/releases/tag/v0.9.6beta) and see if the device still hangs?

1technophile, 2021-04-24:
Please test v0.9.6 and if the gateway still hang, we will see how to find the root cause (watchdog by mqtt seems a bit like a temporary bypass for me)

closed 2021-04-24 by 1technophile