1technophile/OpenMQTTGateway #875: Cant compile dev build since Jan 31st

pharper-aus, 2021-02-24:
Since the introduction of auto discovery changes I can no longer build and receive the following error

**ZgatewayBT:615:13: error: 'disc' was not declared in this scope
         if (disc)
             ^**

I think disc is in scope in main but not in this module.

1technophile, 2021-02-24:
the continuous integration [build](https://github.com/1technophile/OpenMQTTGateway/actions), so there may be something depending on the environment of the build.

Could you share your environment parameters, Arduino IDE/platformio with respectively user_config.h or .ini file. Don't forget to remove your credentials.

pharper-aus, 2021-02-26:
ok I will do that shortly, 

I simply did a git pull of latest dev , opened main.ini in arduino IDE then uncommented this line 

//#define ZgatewayBT     "BT"       //ESP8266, ESP32

by removing the // then choosing compile and I get the error

by commenting the line         

//if (disc)

the code will compile but obviously discovery happens every time.

*also do you have a guide to setting up the project in platform IO, it looks like a great environment and I work in VSCode everyday now so any help would be great.

1technophile, 2021-02-26:
Thanks for the info, I will take a look.

> also do you have a guide to setting up the project in platform IO, it looks like a great environment and I work in VSCode everyday now so any help would be great.

I began with this one:
https://www.youtube.com/watch?v=0poh_2rBq7E

1technophile, 2021-02-26:
Hi @pharper-aus 

I have done a correction on the linked branch, could you test that it solves your problem please

closed 2021-02-26 by 1technophile

pharper-aus, 2021-02-27:
Thanks , that worked like a charm , also thanks for the youtube link it was perfect