1technophile/OpenMQTTGateway #885: Can't upload FW to M5StickC - Platformio

misko903, 2021-03-07:
Before submitting a problem please check the troubleshooting section
https://docs.openmqttgateway.com/upload/troubleshoot.html

**Describe the bug**
Hello, I can't upload the fw to my new M5StickC through Platformio. Done all the steps as described in manual.

```
> Executing task: C:\Users\misko903\.platformio\penv\Scripts\platformio.exe run --target upload --environment esp32-m5stick-c-ble <

Processing esp32-m5stick-c-ble (platform: espressif32@1.12.2; board: m5stick-c; framework: arduino)
--------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/m5stick-c.html
PLATFORM: Espressif 32 (1.12.2) > M5Stick-C
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 3.10004.210126 (1.0.4)
 - tool-esptoolpy 1.20600.0 (2.6.0)
 - tool-mkspiffs 2.230.0 (2.30)
 - toolchain-xtensa32 2.50200.97 (5.2.0)
Converting main.ino
C:/Users/misko903/Downloads/#RPi/M5Stick/OpenMQTTGateway/OpenMQTTGateway-0.9.5/main/main.ino:41:0: warning: "struct_size" redefined
 #    define struct_size 4
 ^
C:/Users/misko903/Downloads/#RPi/M5Stick/OpenMQTTGateway/OpenMQTTGateway-0.9.5/main/main.ino:38:0: note: this is the location of the previous definition
 #    define struct_size 12
 ^
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Library Manager: Installing git+https://github.com/1technophile/Arduino-Log.git#d13cd80
FileNotFoundError: [WinError 2] The system cannot find the file specified:
  File "C:\Users\misko903\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 177:
    env.SConscript("$BUILD_SCRIPT")
  File "C:\Users\misko903\.platformio\packages\tool-scons\scons-local-4.1.0\SCons\Script\SConscript.py", line 591:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\misko903\.platformio\packages\tool-scons\scons-local-4.1.0\SCons\Script\SConscript.py", line 280:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\misko903\.platformio\platforms\espressif32@1.12.2\builder\main.py", line 213:
    target_elf = env.BuildProgram()
  File "C:\Users\misko903\.platformio\packages\tool-scons\scons-local-4.1.0\SCons\Util.py", line 658:
    return self.method(*nargs, **kwargs)
  File "C:\Users\misko903\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 62:
    env.ProcessProjectDeps()
  File "C:\Users\misko903\.platformio\packages\tool-scons\scons-local-4.1.0\SCons\Util.py", line 658:
    return self.method(*nargs, **kwargs)
  File "C:\Users\misko903\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 140:
    project_lib_builder = env.ConfigureProjectLibBuilder()
  File "C:\Users\misko903\.platformio\packages\tool-scons\scons-local-4.1.0\SCons\Util.py", line 658:
    return self.method(*nargs, **kwargs)
  File "C:\Users\misko903\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 1062:
    project.install_dependencies()
  File "C:\Users\misko903\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 898:
    lm.install(spec)
  File "c:\users\misko903\.platformio\penv\lib\site-packages\platformio\package\manager\_install.py", line 49:
    spec, silent=silent, skip_dependencies=skip_dependencies, force=force
  File "c:\users\misko903\.platformio\penv\lib\site-packages\platformio\package\manager\library.py", line 86:
    force=force,
  File "c:\users\misko903\.platformio\penv\lib\site-packages\platformio\package\manager\_install.py", line 97:
    pkg = self.install_from_url(spec.url, spec, silent=silent)
  File "c:\users\misko903\.platformio\penv\lib\site-packages\platformio\package\manager\_install.py", line 141:
    vcs = VCSClientFactory.new(tmp_dir, url)
  File "c:\users\misko903\.platformio\penv\lib\site-packages\platformio\package\vcsclient.py", line 55:
    src_dir, remote_url, tag, silent
  File "c:\users\misko903\.platformio\penv\lib\site-packages\platformio\package\vcsclient.py", line 137:
    self.configure()
  File "c:\users\misko903\.platformio\penv\lib\site-packages\platformio\package\vcsclient.py", line 146:
    result = proc.exec_command([cls.command, "--exec-path"])
  File "c:\users\misko903\.platformio\penv\lib\site-packages\platformio\proc.py", line 117:
    p = subprocess.Popen(*args, **kwargs)
  File "C:\Users\misko903\.platformio\python3\lib\subprocess.py", line 800:
    restore_signals, start_new_session)
  File "C:\Users\misko903\.platformio\python3\lib\subprocess.py", line 1207:
    startupinfo)
============================================================================= [FAILED] Took 7.04 seconds =============================================================================
Environment          Status    Duration
-------------------  --------  ------------
esp32-m5stick-c-ble  FAILED    00:00:07.045
======================================================================= 1 failed, 0 succeeded in 00:00:07.045 ======================================================================= 
The terminal process "C:\Users\misko903\.platformio\penv\Scripts\platformio.exe 'run', '--target', 'upload', '--environment', 'esp32-m5stick-c-ble'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.
```

I have made also PlaformIO:Clean, but it does the same thing.
Thanks!

1technophile, 2021-03-12:
Could you try with the last development branch please?

kajmaj, 2021-03-12:
The same result with the latest development branch and `nodemcuv2-pilight`

1technophile, 2021-03-12:
Seems to be the same as 
https://community.openmqttgateway.com/t/cannot-compile-for-pilight/1472/5

Could you go there, I have made a proposal to test

misko903, 2021-03-13:
Same result - failed, going to try the proposal on the link.

misko903, 2021-03-13:
tested with:

```
Several tracks to follow:
Launch the command pio update
Replace
arduinolog = https://github.com/1technophile/Arduino-Log.git#d13cd80

by

arduinolog = thijse/ArduinoLog
```

still the same.

1technophile, 2021-03-13:
> still the same.

It should not be exactly the same, you should have now the error for wifimanager instead of Arduino log. The problem could be located into platformio library manager.

kajmaj, 2021-03-13:
Hm  I will try to do complete uninstall, leftovers cleaning and new installation of VSC and Platformio. 
No other idea what else could do.

1technophile, 2021-03-14:
@misko903 please install git

closed 2021-03-16 by 1technophile