1technophile/OpenMQTTGateway #891: Stability issue with esp32dev-ble-test (2 day max runtime)

github768484, 2021-03-11:
> 
> 
> [esp32dev-ble-test.zip](https://github.com/1technophile/OpenMQTTGateway/files/6013036/esp32dev-ble-test.zip)
> Here are the bin files

I can only get the ESP to run for a maximum of 2 days. The only change to the default setting is that the interval is set to 4000 (4 seconds). I logged the Freemem and didn't observe any significant change while ESP was running. Additionally, I've installed onto another ESP32 and had the same results.

1technophile, 2021-03-11:
I will try to reproduce with these parameters

1technophile, 2021-03-11:
[esp32dev-ble-test.zip](https://github.com/1technophile/OpenMQTTGateway/files/6125707/esp32dev-ble-test.zip)
If you want to test the last binaries also, here they are

csiki2, 2021-03-16:
If you have the chance, you could try to use a slightly modified BLEScan()
<pre>
void BLEscan() {
  TIMERG0.wdt_wprotect = TIMG_WDT_WKEY_VALUE;
  TIMERG0.wdt_feed = 1;
  TIMERG0.wdt_wprotect = 0;
  Log.notice(F("Scan begin" CR));
  BLEDevice::init("");
  BLEScan* pBLEScan = BLEDevice::getScan(); //create new scan
  MyAdvertisedDeviceCallbacks myCallbacks;
  pBLEScan->setAdvertisedDeviceCallbacks(&myCallbacks);
  pBLEScan->setActiveScan(false); //active scan uses more power, but get results faster <-- Change 1
  BLEScanResults foundDevices = pBLEScan->start(Scan_duration / 1000, false);
  scanCount++;
  Log.notice(F("Found %d devices, scan number %d end deinit controller" CR), foundDevices.getCount(), scanCount);
  //BLEDevice::deinit(true); <-- Change 2
}
</pre>
The first disables the active scan, the second comments out the deinit function, so the ESP32 will stop allocate and reallocate the whole BLE stack again and again and again...
(probably the second change alone will help you though)

github768484, 2021-03-17:
> 
> 
> [esp32dev-ble-test.zip](https://github.com/1technophile/OpenMQTTGateway/files/6125707/esp32dev-ble-test.zip)
> If you want to test the last binaries also, here they are

This version is 4 days running. So far so good.

github768484, 2021-03-17:
> 
> 
> If you have the chance, you could try to use a slightly modified BLEScan()
> 
> void BLEscan() {
>   TIMERG0.wdt_wprotect = TIMG_WDT_WKEY_VALUE;
>   TIMERG0.wdt_feed = 1;
>   TIMERG0.wdt_wprotect = 0;
>   Log.notice(F("Scan begin" CR));
>   BLEDevice::init("");
>   BLEScan* pBLEScan = BLEDevice::getScan(); //create new scan
>   MyAdvertisedDeviceCallbacks myCallbacks;
>   pBLEScan->setAdvertisedDeviceCallbacks(&myCallbacks);
>   pBLEScan->setActiveScan(false); //active scan uses more power, but get results faster <-- Change 1
>   BLEScanResults foundDevices = pBLEScan->start(Scan_duration / 1000, false);
>   scanCount++;
>   Log.notice(F("Found %d devices, scan number %d end deinit controller" CR), foundDevices.getCount(), scanCount);
>   //BLEDevice::deinit(true); <-- Change 2
> }
> 
> The first disables the active scan, the second comments out the deinit function, so the ESP32 will stop allocate and deallocate the whole BLE stack again and again and again...
> (probably the second change alone will help you though)

I'm happy to. Please help me by either adding to an Arduino IDE or binary and posting on this thread.

csiki2, 2021-03-17:
> I'm happy to. Please help me by either adding to an Arduino IDE or binary and posting on this thread.
Sorry, I use platformio and I don't know what setup is used for the firmware that Florian gave to you (development branch with simpe esp32dev-ble environment without any extra settings?).

csiki2, 2021-03-18:
> I'm doing some test regarding scan_duration and interval so as to identify the relevant settings for continuous scanning. Will publish a binary here when done.

I played around with such parameters at ESPHome in the past (OMG replaced it), it's quite easy to kill the OTA...
Also, the active scan with big window is a strong wifi killer (not a surprise though).
<pre>
esp32_ble_tracker:
  scan_parameters:
    duration: 150 s
    interval: 52 ms
    window: 30 ms
    active: false
</pre>

1technophile, 2021-03-22:
@github768484 I reproduce your issue on the linked PR  with these parameters:
```ini
  '-DTimeBtwRead=0'
  '-DScan_duration=1000'
  '-DAttemptBLECOnnect=false'
  '-DActiveBLEScan=true'
```
I'm doing some test with the active BLE scan deactivated as suggested by @csiki2

1technophile, 2021-03-23:
The gateway is stuck into a loop trying to reconnect even if the wifi and broker is up:
```
[E][WiFiClient.cpp:232] connect(): connect on fd 58, errno: 118, "Host is unreachable"
W: failure_number_mqtt: 624
W: failed, rc=-2
```
Seems to be a low level issue around the wifi stack.

I update the ESP32 Arduino framework and see if it gets better.

csiki2, 2021-03-25:
In general, what do we win with the active scan?
As far as I understand, the active scan "questions back" for extra data at a device broadcast. For my devices I got back nothing or "very important" things like model name.
Is there a case when the active scan is necessary?

1technophile, 2021-03-25:
From my tests, active scan brings more scan results compared to passive scan, 2.6/s versus 2.3/s

Note that I have updated to the ESP32 Arduino PIO framework 3.1.1 and the esp32 is not anymore stuck with the parameters above. Nevertheless, I'm getting some regular restarts that I will analyze.

csiki2, 2021-03-25:
The number of the json messages increased, but the scan results were actually the same.
As far as I can remember, what the original code did is sending 2 results if the gateway got an answer for the active asking: one with the passive result and one (or in theory more) with the passive result + the active result added (reused the json and resent it), but because of this, the meaningful part was the same. (or first the nearly empty active result was sent if the order was wrong...)
Currently at the serviceDataCount>1 tries to resolve the case when there was any active result: we merge all the data before and then the messages will be different _only_ at serviceData level.
I didn't see new temp/hum etc. only model name in my case.

1technophile, 2021-03-25:
> The number of the json messages increased, but the scan results were actually the same.

I haven't gone in deep on the message content, I stayed on a quantitative analysis level. Ideally, I should make a test bench with a  CGH1 changing status every X seconds and verify the percentage of change per status that I get. But this is another story...

1technophile, 2021-03-25:
Note that I get now good stability by disabling the watchdog ... maybe not the best way but at least it prevents random reset of the board.

1technophile, 2021-03-28:
The beta is now quite satisfying in term of stability:
[image]

I got 2 ESP32 now running continuously for more than 2 days. Do not hesitate to test it.

closed 2021-03-28 by 1technophile

csiki2, 2021-03-28:
Sidenote: BTforceScan won't work with TimeBtwRead = 0 (originally the value seemed to be intended always >1)

1technophile, 2021-03-28:
> Sidenote: BTforceScan won't work with TimeBtwRead = 0 (originally the value seemed to be intended always >1)

Indeed, but I'm not sure it is useful when we are continuously scanning

csiki2, 2021-03-28:
Force scan does connection type scanning (full BLEConnect).
I don't use either, just mentioned. :)

1technophile, 2021-03-28:
You are right, thanks for pointing this