1technophile/OpenMQTTGateway #906: Query about 0.9.6 beta behaviour with ESP32 BLE gateway

DigiH, 2021-03-27:
Having update my RF, IR and BLE gateways with the latest 0.9.6 beta version - all worked nicely OTA - both the RF and IR gateways seem to work fine, but I see confusing new behaviour with the ESP32 based BLE gateway, as follows:

With 0.9.5 I previously used the different topics
**home/OpenMQTTGateway_BLE/BTtoMQTT/XXXXXXXXXXXX**
and
**home/home_presence/OpenMQTTGateway_BLE**

to get the different JSON results for the same BT device, e.g.

**home/OpenMQTTGateway_BLE/BTtoMQTT/XXXXXXXXXXXX**
with JSON
{"id":"XX:XX:XX:XX:XX:XX","name":"","rssi":-67,"distance":2.509097}

and
**home/home_presence/OpenMQTTGateway_BLE**
with JSON
{"id":"XX:XX:XX:XX:XX:XX","name":"","manufacturerdata":"030000604700000a440010ad61c688c626ad5cd132","rssi":-67,"distance":2.509097}

with the latter using parts of manufacturerdata to get the battery level status for my UE Booms.

With 0.9.6 beta both topics now result only in the same JSON, without any manufacturerdata.

{"id":"XX:XX:XX:XX:XX:XX","name":"","rssi":-67,"distance":2.509097}

Is there a setting I haven't come across during my short time of testing yet, which allows for the manufacturerdata to be displayed and processed again?

Additionally, with my nrf51822 BT Beacons the "name" tag is not included in the JSON now.

**0.9.5**
{"id":"XX:XX:XX:XX:XX:XX","name":"HOME-02","rssi":-75,"distance":5.832737}

**0.9.6 beta**
{"id":"XX:XX:XX:XX:XX:XX":-77,"distance":7.119795}

Also there seem to multiple discoveries of the same BT devices now during a single Scan_duration, but I haven't closely looked at the other new related settings like BLEScanWindow etc. in config_BT.h yet.

Many thanks for any help and insight into the above.

1technophile, 2021-03-27:
> Is there a setting I haven't come across during my short time of testing yet, which allows for the manufacturerdata to be displayed and processed again?

Indeed there is a build setting to have the manufacturer data, here is [it](https://github.com/1technophile/OpenMQTTGateway/blob/cd55f0dfd5966abff3da442cc698ffea171ce788/main/config_BT.h#L102)

> Additionally, with my nrf51822 BT Beacons the "name" tag is not included in the JSON now.

That's strange I see the name of my Qingping devices, could it be that the message with the name is override by a message without a name?

> Also there seem to multiple discoveries of the same BT devices now during a single Scan_duration, but I haven't closely looked at the other new related settings like BLEScanWindow etc. in config_BT.h yet.

 What settings are you using?

DigiH, 2021-03-27:
I already ammended the original text above, as I just discovered pubBLEManufacturerData shortly after sending it off  ;)

> Additionally, with my nrf51822 BT Beacons the "name" tag is not included in the JSON now.

> That's strange I see the name of my Qingping devices, could it be that the message with the name is override by a > 
message without a name? 

Ok, I set ActiveBLEScan to false, as this reduced (not fully elimintaed) the multiple discoveries mentioned above. Setting it back to true I do get the "name" tag back, but now I get roughly 17 discoveries for each device during the default Scan_duration of 10000 - all the the name though ;)

1technophile, 2021-03-27:
Could you try to remove these [2 lines](https://github.com/1technophile/OpenMQTTGateway/blob/cd55f0dfd5966abff3da442cc698ffea171ce788/main/ZgatewayBT.ino#L641) to see if it removes the duplicate issue:
```c
  pBLEScan->setInterval(BLEScanInterval);
  pBLEScan->setWindow(BLEScanWindow);
```

DigiH, 2021-03-27:
> ```c
>   pBLEScan->setInterval(BLEScanInterval);
>   pBLEScan->setWindow(BLEScanWindow);
> ```

Commented them out, but still the same issue. It seems to be my UE Booms with the multiple discoveries (17-22) times within the defsult 10 second scan window, but most of the times my nrf51822 BT Beacons don't get discovered at all during each Scan_duration. Possibly due to the other multiple discoveries. 

If I delete my white list though, and have all the my other BT devices included other ones are recognised mutiple times as well.

Last try just now:

**478** BT MQTT messages within the 10 second scan window for something like 15 BT devices.

DigiH, 2021-03-27:
Ok, definitely is dependant on the BT devices being scanned.

With no white list I get these results during a single Scan_duration of 10 seconds

[image]

With the first two being the two UE Booms, the 3rd is one of my Playbulb Candles, and the last two are my nrf51822 BT Beacons, not being recognised at all during this scan.

With the whitelist being active, just for the UE Booms and the BT Beacons I get the following when I comment out 

pBLEScan->setInterval(BLEScanInterval);
pBLEScan->setWindow(BLEScanWindow);

[image]

The results vary slightly from scan to scan, sometimes the BT beacons get recognised 1-2 times, or not at all, but the UE Booms always have miltiple recognitions.

1technophile, 2021-03-27:
Could you do a test with this configuration please?
```ini
  '-DAttemptBLECOnnect=false'
  '-DActiveBLEScan=true'
  '-DTimeBtwRead=0'
  '-DScan_duration=1000'
```

And removing the lines:
```
pBLEScan->setInterval(BLEScanInterval);
pBLEScan->setWindow(BLEScanWindow); 
```

Also if we want to get closer to v0.9.5 you may use:
```ini
  '-DAttemptBLECOnnect=false'
  '-DActiveBLEScan=true'
  '-DTimeBtwRead=0'
  '-DScan_duration=10000'
```
removing the lines:
```
pBLEScan->setInterval(BLEScanInterval);
pBLEScan->setWindow(BLEScanWindow); 
```
go back to previous NimBLE Arduino library version in platformio.ini:
`ble = https://github.com/h2zero/NimBLE-Arduino.git#1.0.0`

go back to the previous ESP32 framework in platformio.ini:
`esp32_platform = espressif32@3.0.0`

DigiH, 2021-03-27:
> ```ini
>   '-DAttemptBLECOnnect=false'
>   '-DActiveBLEScan=true'
>   '-DTimeBtwRead=0'
>   '-DScan_duration=1000'
> ```

With '-DTimeBtwRead=0' and  '-DScan_duration=1000' this just constantly scans for devices in repeating 1 second loops, but still results in the UE Booms and the Playbulb Candles being recognised many times more than the BT Beacons.

So I set everything back as previously desired, i.e.

'-DAttemptBLECOnnect=false'
 '-DActiveBLEScan=true'
 '-DTimeBtwRead=10000' (with the default  '-DTimeBtwRead=55555' for testing purposes)
 '-DScan_duration=10000'

did NOT remove 

> pBLEScan->setInterval(BLEScanInterval);
> pBLEScan->setWindow(BLEScanWindow); 

but did go back to the previous NimBLE Arduino library 1.0.0 and the previous ESP32 framework 3.0.0.

Now it's all working nicely again with each device only being recognised once during every scan.

So the NimBLE Arduino library 1.2.0 / ESP32 framework 3.1.1 combo caused the strange behaviour I was previously seeing.

Not the ideal solution I suppose, but I'll use the previous library/framework for my BLE gateway build for now. With the IR and RF gateways they did not cause any issues.

As I only saw this with the UE Booms and Playbulb Candles I'll be interested to see if anyone else finds the same behaviour with any other BT devices.

1technophile, 2021-03-27:
> but did go back to the previous NimBLE Arduino library 1.0.0 and the previous ESP32 framework 3.0.0.
Now it's all working nicely again with each device only being recognised once during every scan.

Would it be possible to isolate which one between these 2 changes is guilty please ?

DigiH, 2021-03-27:
> Would it be possible to isolate which one between these 2 changes is guilty please ?

Already tried that this afternoon, as I was expecting the NimBLE Arduino library to be the curplit, but when trying to use the old NimBLE Arduino library 1.0.0 with the current ESP32 framework 3.1.1 I got a build error. Didn't have much time to try and dig further into it this afternoon, but the error was

In file included from /OpenMQTTGateway-0.9.6beta/main/ZgatewayBT.ino:36:0:
.pio/libdeps/esp32dev-ble-ha/NimBLE-Arduino/src/FreeRTOS.h:72:31: error: 'RingbufferType_t' has not been declared
     Ringbuffer(size_t length, RingbufferType_t type = RINGBUF_TYPE_NOSPLIT);

Building with previous ESP32 framework 3.0.0 and using the current NimBLE Arduino library 1.2.0 builds fine, but is still causing the many multiple recognitions.

So very likely an issue caused by the current NimBLE Arduino library 1.2.0 change.

DigiH, 2021-03-28:
I just did some more testing, and some of my other BT devices which get multiple recognitions during a single scan are 

Apple TV 4k
Xiaomi Mi Body Composition Scale 2

along with the previously mentioned
UE Boom 2
Playbulb Candle

Might help others with these devices to help testing/verifying the occurrence.

Thanks

1technophile, 2021-03-28:
It could be related to the introduction of [setScanFilterMode](https://github.com/h2zero/NimBLE-Arduino/commit/bd64ede5e806c2216ff55c1dd17d8280fecc84df) in 1.2.0 of NimBLE Arduino.

So as to confirm it, could you install 1.1.0 instead of 1.2.0

`ble = https://github.com/h2zero/NimBLE-Arduino.git#1.1.0`

It will build ok with ESP32 framework 3.1.1

DigiH, 2021-03-28:
> So as to confirm it, could you install 1.1.0 instead of 1.2.0
> 
> `ble = https://github.com/h2zero/NimBLE-Arduino.git#1.1.0`
> 
> It will build ok with ESP32 framework 3.1.1

Perfect, NimBLE Arduino library 1.1.0 built with ESP32 framework 3.1.1 works fine, with every, previously affected, device only being recognised and reported over MQTT once.

I will keep this combo for my build, but am not too sure what other important changes might be included in the 1.1.0 to 1.2.0 update related to OMG. Just briefly came accross this - m_maxResults - which als sounds like it could have had an effect on the issue I was seeing.

https://github.com/h2zero/NimBLE-Arduino/commit/afc5e92d9d8c77675be8dd0ed006f8eb9ac3ff16

Thanks a lot

1technophile, 2021-03-28:
@h2zero, is it possible that the introduction of setScanFilterMode in 1.2.0 changed the behavior of the scan regarding filtering (even if the new method is not used)?

h2zero, 2021-03-28:
>@h2zero, is it possible that the introduction of setScanFilterMode in 1.2.0 changed the behavior of the scan regarding filtering (even if the new method is not used)?

Yes, in a way. The controller based scan filter has a cache size limit and once reached it will clear it and start reporting the same devices again. The cache is adjustable and the filtering is adjustable as you can see in the link you referenced. However there is something I missed in the API for this and that is the ability to clear the cache so anything in them from a previous scan will not be filtered when starting again. This can be done directly with a low level call to `esp_ble_scan_dupilcate_list_flush()`. 

So if you are seeing the same device too often the filtering should be adjusted to reduce it accordingly and if devices are missing in subsequent scans the list flush call above should be made before starting the scan. If that does not produce the results as expected I will look into adding additional filtering on the host side.

1technophile, 2021-03-28:
@h2zero thanks!

@DigiH could you add please
```
esp_ble_scan_dupilcate_list_flush();
```
after [this line](https://github.com/1technophile/OpenMQTTGateway/blob/f01b7184a02e43184eaa60b0c976a49dd159701c/main/ZgatewayBT.ino#L642) + NimBle arduino 1.2.0 and see if you get the BT beacons at each scan?

DigiH, 2021-03-28:
> @DigiH could you add please
> 
> ```
> esp_ble_scan_dupilcate_list_flush();
> ```
> 
> after [this line](https://github.com/1technophile/OpenMQTTGateway/blob/f01b7184a02e43184eaa60b0c976a49dd159701c/main/ZgatewayBT.ino#L642) + NimBle arduino 1.2.0 and see if you get the BT beacons at each scan?

Afraid that didn't fix it

[image]

2 UE Booms, one Playbulb Candle and the Apple TV 4K have many multiple recognitions again during a single scan. Ignore the 0 count, as it is a switched off BT beacon, and last in the list a correctly once recognised BT beacon.

1technophile, 2021-03-28:
Thanks, the most important thing to look for, is the fact that it recognizes your beacons. So the test is OK I think.

This added line was not expected to fix the duplicate but only the non-recognized items.

DigiH, 2021-03-28:
> This added line was not expected to fix the duplicate but only the non-recognized items.

My misunderstanding, sorry. Turned the second beacon on as well now and both get correctly recognised once during the scan.

h2zero, 2021-03-28:
Nice, glad to see that helps. I think I should add that function as an API call as well as doing it automatically in Scan::start. I'll be watching this for further tweaks that should be made.

1technophile, 2021-03-28:
Thanks for the feedback!

What could be the filtering parameter that will enable us to retrieve the same level of filtering as for the 1.1.0?

h2zero, 2021-03-28:
Hard to say for sure, the controller filter works a bit differently. I would think increasing the cache size to max would be the place to start and work my way down. So before `NimBLEDevice::init()` you would add 'NimBLEDevice::setScanDuplicateCacheSize(1000);` and see how that goes.

DigiH, 2021-03-28:
> So before `NimBLEDevice::init()` you would add `NimBLEDevice::setScanDuplicateCacheSize(1000);` and see how that goes.

Just testing this, and looks promising so far, with your suggested 1000 ... monitoring for a bit now.

DigiH, 2021-03-28:
Ok, with

BLEDevice::setScanDuplicateCacheSize(1000);

added before 

BLEDevice::init("");

in 

I'm getting no more multiple recognitions per scan.

[image]

I have to say I also deleted the previous 

> esp_ble_scan_dupilcate_list_flush();

as I didn't see any difference in the BT beacons being recognised or not.

@h2zero - you mentioned

> and work my way down … and see how that goes.

So do you think I should keep testing with other values than the current 1000, and that different setup with different amounts of BT devices might need a changed setting here, or is the current 1000 ok if the results are fine?

Thanks

h2zero, 2021-03-28:
>I have to say I also deleted the previous
    esp_ble_scan_dupilcate_list_flush();
as I didn't see any difference in the BT beacons being recognised or not.

That's curious, I would think that would still be necessary. Something for me to look into.

>So do you think I should keep testing with other values than the current 1000, and that different setups with different amounts of BT devices might need a changed setting here, or is the current 1000 ok if the results are fine?

If you're up to it, I'd love to get some benchmark data for reference, if not that's cool too, glad it's working for you.

DigiH, 2021-03-28:
> If you're up to it, I'd love to get some benchmark data for reference, if not that's cool too, glad it's working for you.

 BLEDevice::setScanDuplicateCacheSize(500);       OK
 BLEDevice::setScanDuplicateCacheSize(300);       OK
 BLEDevice::setScanDuplicateCacheSize(100);       OK
 BLEDevice::setScanDuplicateCacheSize(50);         OK
 BLEDevice::setScanDuplicateCacheSize(10);         FAIL - multiple recognitions appearing again

working my way up again
 BLEDevice::setScanDuplicateCacheSize(20);         FAIL - multiple recognitions appearing again
 BLEDevice::setScanDuplicateCacheSize(30);         FAIL - multiple recognitions appearing again
 BLEDevice::setScanDuplicateCacheSize(40);         OK

I have to add though that I only monitored 4 devices during that testing out of the 12 devices which previously showed the multiple recognitions. The two UE Booms, which showed the most were included.

> > I have to say I also deleted the previous
> > esp_ble_scan_dupilcate_list_flush();
> > as I didn't see any difference in the BT beacons being recognised or not.
> 
> That's curious, I would think that would still be necessary. Something for me to look into.

The line was commented out in all the above tests, and the BT beacons showed up fine.

h2zero, 2021-03-28:
Awesome, thanks for doing that. Gives me some idea for future issues.

1technophile, 2021-03-30:
Thanks for the cooperation in solving this.

If I add:

DigiH, 2021-03-30:
> If I add:
> 
> `BLEDevice::setScanDuplicateCacheSize(500);`
> 
> By default, we should be ok I think?

For my experience so far I would assume 500 should be more than enough for most setups. Running here with 100 for the last two days and with some neighbours BT devices included there are around 20 device altogether, all of them only showing up once per scan if no white-list is active.

One thing I saw in the last two days though is that the BLE Gateway restarts about 4-5 times every 24 hours. Wasn't able to catch the restart reason so far, only ever saw it when seeing the uptime, compared to the IR and RF gateways, which both were up constantly in the 2 days.

Was wondering if me taking out

> esp_ble_scan_dupilcate_list_flush();

might have something to do with it, so currently put it back in and restarted all gateways for another compare.

So far so good ...

[image]

I suppose I could do some further logging of the restart reason and the memory history to pin it down. 

@1technophile - any idea what might cause these numerous restarts or suggestions what I should look out for and possibly log? Since the BLE gateway reconnects fine to the WiFi there isn't really that much of an impact, but nevertheless it became very apparent, as with 0.9.5 I usually had uptimes of days and weeks with my same setup.

h2zero, 2021-03-30:
I don't see any issue with that. For reference the Arduino default value is 20 and IDF default is 200.

1technophile, 2021-03-30:
> @1technophile - any idea what might cause these numerous restarts or suggestions what I should look out for and possibly log? Since the BLE gateway reconnects fine to the WiFi there isn't really [...]

Is it your configuration?
```
'-DAttemptBLECOnnect=false'
'-DActiveBLEScan=true'
'-DTimeBtwRead=10000'
'-DScan_duration=10000'
```

DigiH, 2021-03-30:
> Is it your configuration?
> 
> ```
> '-DAttemptBLECOnnect=false'
> '-DActiveBLEScan=true'
> '-DTimeBtwRead=10000'
> '-DScan_duration=10000'
> ```

Yes, these are my settings.

DigiH, 2021-03-30:
Just as an update. I have increased

BLEDevice::setScanDuplicateCacheSize(200); 

from the previous 100. So far the BLE gateway is up an running for over 6 hours now, whereas with 100 the first restart always happened after 3 to 4 hours at the latest. 

I'll keep monitoring and if a restart happens again I'll change it to your planned 500, @1technophile, to see what the behaviour is then.

Not really sure how setting the cache size could affect or cause restarts, just observing here ;)

1technophile, 2021-03-30:
Thanks for the update, 
I will run a test with 200 also

DigiH, 2021-03-31:
> I will run a test with 200 also

Still looking promising with 200 so far

[image]

Will see what it says tomorrow during the day.

DigiH, 2021-03-31:
Oh no :(

Just when going to bed I saw the following

[image]

Not quite extly the double as could be expected when going from 100 to 200, but close enough to be possibly relate it to setting the Duplicate Cache Size!?!

Setting to 300 now, just to see if the next restart will be around 12 - 16 hours from now.

DigiH, 2021-03-31:
So far so good, for me with

> BLEDevice::setScanDuplicateCacheSize(200);

[image]

Previously with 100 and 200 I found that before a restart happened that 

(100)
"btqblck":4

(200)
"btqblck":1

if that maybe gives some indication as to why the gateway restarted.

Hopefully tomorrow the uptime will be up and running as well.

h2zero, 2021-04-01:
Pushed the above mentioned updates to NimBLE-Arduino master, not that it seems to affect this issue any longer. However, it may provide more consistency if there is some timing related event causing the crashing.

DigiH, 2021-04-02:
So my test with 300 came up with the following

[image]

so together with the previous results, this somehow suggested an exponential relationship between setting the cache size and the time until a restart happens.

With trying to confirm this I set it to 100 again

BLEDevice::setScanDuplicateCacheSize(100);

Now a restart happened almost immediately after 12 minutes of the upload, and since then the BLE gateway is up and running for over 13 hours.

Definitely not confirming my initial assumption, but still a lot more frequent restarts than accustomed to from the previous version.

@1technophile - did you come across such restarts with your test with 200?

Thanks

1technophile, 2021-04-02:
I have spotted the reset in serial:
```
T: getDeviceByMac 
T: manufacturerdata 
T: Is it a iNode Energy Meter?
T: Remove manufacturer data
N: Subject: /BTtoMQTT/
N: Received json : {"id":"","rssi":-37,"distance":0.009408}
T: jsonPublishing
T: Pub json :{"id":"","rssi":-37,"distance":0.009408} into custom topic: home/home_presence/OpenMQTTGateway_ESP32_interval
ASSERT_PARAM(-218959118 0), in arch_main.c at line 343
Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0)
Core 0 register dump:
PC      : 0x4008489c  PS      : 0x00060b34  A0      : 0x80136114  A1      : 0x3ffc00b0  
A2      : 0x00000001  A3      : 0x00000000  A4      : 0x60008048  A5      : 0x00000000  
A6      : 0x00000004  A7      : 0x3ffbdbb4  A8      : 0x8008489c  A9      : 0x3ffc0090  
A10     : 0x0000f2f2  A11     : 0x00000037  A12     : 0x00000014  A13     : 0xffffffff  
A14     : 0x00000000  A15     : 0xfffffffc  SAR     : 0x00000004  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x400847d4  LEND    : 0x400847db  LCOUNT  : 0x00000000  
Core 0 was running in ISR context:
EPC1    : 0x40009203  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x4008489c

ELF file SHA256: 0000000000000000

Backtrace: 0x4008489c:0x3ffc00b0 0x40136111:0x3ffc00d0 0x40019fb5:0x3ffc00f0 0x4001ab1f:0x3ffc0120 0x40087b4a:0x3ffc0140 0x40084e0a:0x3ffc0180 0x4008a2ce:0x3ffc01a0 0x4008ad7b:0x3ffc01c0 0x4008ca99:0x3ffc01e0 0x4013653c:0x3ffe1b40 0x40136b05:0x3ffe1b60 0x40090de2:0x3ffe1b90

Core 1 register dump:
PC      : 0x4019a4d2  PS      : 0x00060a34  A0      : 0x8011a2b1  A1      : 0x3ffbc870  
A2      : 0x00000000  A3      : 0x80000001  A4      : 0x00000000  A5      : 0x00000001  
A6      : 0x00060a20  A7      : 0x00000000  A8      : 0x80091fac  A9      : 0x3ffbc860  
A10     : 0x00000003  A11     : 0x00060a23  A12     : 0x00060a20  A13     : 0x3ffbc770  
A14     : 0x00000000  A15     : 0x3ffbc560  SAR     : 0x00000000  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

ELF file SHA256: 0000000000000000

Backtrace: 0x4019a4d2:0x3ffbc870 0x4011a2ae:0x3ffbc890 0x4009278d:0x3ffbc8b0 0x40090de2:0x3ffbc8d0

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5828
entry 0x400806a8
N: 
************* WELCOME TO OpenMQTTGateway **************
```

It occurs after the home presence detection, if it is something that you don't use, could you try to comment [this line](https://github.com/1technophile/OpenMQTTGateway/blob/55dd571c49e27b661b903e944634d7bcaacdca5a/main/config_BT.h#L111)? 

And see if it removes the reset.

DigiH, 2021-04-02:
> I have spotted the reset in serial:
> It occurs after the home presence detection, if it is something that you don't use, could you try to comment [this line](https://github.com/1technophile/OpenMQTTGateway/blob/55dd571c49e27b661b903e944634d7bcaacdca5a/main/config_BT.h#L111)?

I'm actually using home presence for all my battery checks for the UE Booms as well as the BT Beacons, getting this information from the manufacturerdata. As I stated at the top, the manufacturerdata was only present in the home_presence JSON, but not in the individual BTtoMQTT/XXXXXXXXXXXX JSON for OMG 0.9.5. That's the only reason why I went for home_presence.

Now that with 0.9.6 beta the JSON is identical for home_presence and BTtoMQTT messages I'm planning on rewriting my code, as there is no need for the gateway to send such duplicate MQTT messages all the time.

With BLEDevice::setScanDuplicateCacheSize(100) the gateway is still up and running 19 hours +, and we now seem to know that this doesn't affect the restarts.

So with me having home presence detection ON and the gateway scanning, recognising and sending home_presence MQTT mesages for at least 3 devices every 20 seconds, I'm seeing restarts from 12 minutes to 36 hours after after upload.

Any further tests I can do to help pin down when and why sometimes home presence detection causes these restarts?

I'll need a bit of time to rewrite my openHAB code to get all the same information without using home_presence at all. With the Easter holidays now it shouldn't take too long ;)

1technophile, 2021-04-02:
> Any further tests I can do to help pin down when and why sometimes home presence detection causes these restarts?

If you have an ESP32 laying around that you can use for test, you could upload the code without the home presence. It should not impact your production if you set it up to a different product structure.

> Easter holidays now it shouldn't take too long ;)
:-)

DigiH, 2021-04-02:
> If you have an ESP32 laying around that you can use for test, you could upload the code without the home presence. It should not impact your production if you set it up to a different product structure.

I know, but I literally meant it shouldn't take too long to rewrite my openHAB code :) Since both JSON messages are now the same, doing this was at the top of my list already, to avoid unnecessary duplicate messages. My MQTT broker already has enough MQTT messages to process from the other OMGs and the mysensors gateway etc.

Rewrote openHAB code, commented out home presence integration and reset all 3 openMQTTgateways for a new count down ...

[image]

DigiH, 2021-04-03:
Just when I was thinking that all was going smootly, another restart happened within the first 24 hours :(

[image]

So it seems that removing home presence integration didn't do the trick.

DigiH, 2021-04-03:
For all intents and purposes though, the gateway always restarts very quickly and always reliably reconnects to the WiFi. My white-list is retained and reapplied after every restart. So apart from seeing the restarts in the uptime they don't actually affect the functionality.

Thanks a lot to @1technophile and @h2zero for helping to solve the issue with the duplicate recognition and MQTT messages, also my openHAB code has been rewritten now and the MQTT messages been optimised with taking out home presence integration.

I think this issue can be closed now and the restart issue looked at separately, possibly with others seeing the same and being able to chime in to help narrow down the cause!?!

Any further testing I can do, please let me know.

1technophile, 2021-04-03:
Thanks for your help, I'm digging into the issue with restart and will publish a PR if I find something.

closed 2021-04-03 by 1technophile

DigiH, 2021-04-16:
@1technophile - FYI, just as a test I put the line

> esp_ble_scan_dupilcate_list_flush();

back in again. Since then my BLE gateway has been without restarts for over 24 hours now. Which hasn't happened the last 2 weeks before, without the flushing.

Asked the user vossivossi in the community forum if he feels like testing this as well, as he seems to have similar regular restarts with his BLE gateway. Maybe then we see if the line actually makes a difference in regards to the restarts, or if it's just varying results of fickle ESP32 boards.

DigiH, 2021-07-25:
@1technophile @h2zero - FYI, with OMG 0.9.7 and the default setting of 

`define BLEScanDuplicateCacheSize 200`

I had the issue with multiple recognitions during a single scan again for many of my devices. 

Not sure how this might have changed from the above testing to the final implementation, but overwriting the value with the manual build flag

`'-DBLEScanDuplicateCacheSize=500'`

seems to fix it again for me.

h2zero, 2021-07-25:
If you happen to have more devices around you that are advertising than before then the cache will clear more often which happens once full.

DigiH, 2021-07-25:
> If you happen to have more devices around you that are advertising than before then the cache will clear more often which happens once full.

No new devices since the last test, and with a more restricted whitelist active now it should actually be less devices now.

DigiH, 2021-07-25:
P.S. - I only really came across this as I'm currently testing the new Read/Write BLE characteristics feature with my Playbulb Candles, so was monitoring the direct MQTT Broker stream, otherwise things worked as expected on my openHAB backend and I wouldn't even have recognised the issue. Also with the possibility of the build flag now, things are easily adjusted for each individual environment :) was really only an FYI as it behaved differently to the testing above.

h2zero, 2021-07-25:
Glad to hear it's working out for you. Out of curiosity what was your scan time set to?

DigiH, 2021-07-26:
Scan duration left at the default

`define Scan_duration 10000`

in config_BT.h

Even with my change stated above with 

`'-DBLEScanDuplicateCacheSize=500'`

I am still now seeing the odd duplicate multiple recognitions during a single scan.

Also testing the new Read/Write BLE characteristics feature I find that for each Read of the battery statuses of my nine Playbulb Candles there is always two responses for each of the single Read requests for each candle - usually both responses are identical with

`"success":true`

but every now and then one of the double responses is **true**, while the other is **false**, indicating that these are two separate responses for a single request.

So all a bit puzzling for me at the moment, especially with the above manual testing with 0.9.6 working so flawlessly after we found a good value for setting the DuplicateCacheSize.