1technophile/OpenMQTTGateway #908: Heltec ESP32 LORA v2

botts7, 2021-03-28:
I have 2 x https://compatible.openmqttgateway.com/index.php/product/heltec-lora-32-v2/ 

But I cant seem to get any messages from Sender using the sender sketch from https://github.com/HelTecAutomation/Heltec_ESP32/tree/master/examples/LoRa after installing OMG on the other

(if I install heltec sender on one device and receiver on the other, connection works and I receive the test packets)

OMG loads and connects to my MQTT broker fine. 

I get SYS to MQTT payloads
I get maybe 1-5 LORAtoMQTT payloads, which does not seem to be related to the sender messages.

eg of payload
{"rssi":-109,"snr":-12.25,"pferror":-15141,"packetSize":179,"message":"E����om�ٿ�bq����$#�\"�dL$#f&ظ1+8d!�Vs3?)��[97�ȀY�Lؗ��6��%��צNA�lC� �������(C6z�Q!D@��=d���Z�h\t��ظ@$ɲ"}

OpenMQTTGateway version used (V0.95) (and from Dev Branch from source)

Im sure that im missing something - any help would be appreciated

1technophile, 2021-03-28:
> https://github.com/HelTecAutomation/Heltec_ESP32/tree/master/examples/LoRa

I don't think this program is using the same library as OMG for LORA.

OMG is using the [one](https://github.com/sandeepmistry/arduino-LoRa) from sandeepmistry. Could you try with the example provided there instead?

botts7, 2021-04-02:
Thanks for pointing me in the right direction!

The following works including the display

`
/*
  LoRa Simple Gateway/Node Exemple
  This code uses InvertIQ function to create a simple Gateway/Node logic.
  Gateway - Sends messages with enableInvertIQ()
          - Receives messages with disableInvertIQ()
  Node    - Sends messages with disableInvertIQ()
          - Receives messages with enableInvertIQ()
  With this arrangement a Gateway never receive messages from another Gateway
  and a Node never receive message from another Node.
  Only Gateway to Node and vice versa.
  This code receives messages and sends a message every second.
  InvertIQ function basically invert the LoRa I and Q signals.
  See the Semtech datasheet, http://www.semtech.com/images/datasheet/sx1276.pdf
  for more on InvertIQ register 0x33.
  created 05 August 2018
  by Luiz H. Cassettari
*/

//Libraries for LoRa
#include <SPI.h>              // include libraries
#include <LoRa.h>

//Libraries for OLED Display
#include <Wire.h>
#include <Adafruit_GFX.h>
#include <Adafruit_SSD1306.h>

const long frequency = 915E6;  // LoRa Frequency

const int csPin = 18;          // LoRa radio chip select
const int resetPin = 14;        // LoRa radio reset
const int irqPin = 26;          // change for your board; must be a hardware interrupt pin

//OLED pins
#define OLED_SDA 4
#define OLED_SCL 15 
#define OLED_RST 16
#define SCREEN_WIDTH 128 // OLED display width, in pixels
#define SCREEN_HEIGHT 64 // OLED display height, in pixels

//packet counter
int counter = 0;

Adafruit_SSD1306 display(SCREEN_WIDTH, SCREEN_HEIGHT, &Wire, OLED_RST);

void setup() {
  Serial.begin(9600);                   // initialize serial
  while (!Serial);

  //reset OLED display via software
  pinMode(OLED_RST, OUTPUT);
  digitalWrite(OLED_RST, LOW);
  delay(20);
  digitalWrite(OLED_RST, HIGH);  

  //initialize OLED
  Wire.begin(OLED_SDA, OLED_SCL);
  if(!display.begin(SSD1306_SWITCHCAPVCC, 0x3c, false, false)) { // Address 0x3C for 128x32
    Serial.println(F("SSD1306 allocation failed"));
    for(;;); // Don't proceed, loop forever
  }
  
  display.clearDisplay();
  display.setTextColor(WHITE);
  display.setTextSize(1);
  display.setCursor(0,0);
  display.print("LORA SENDER ");
  display.display();

  Serial.println("LoRa Receiver Test");

  LoRa.setPins(csPin, resetPin, irqPin);

  if (!LoRa.begin(frequency)) {
    Serial.println("LoRa init failed. Check your connections.");
    while (true);                       // if failed, do nothing
  }

  Serial.println("LoRa init succeeded.");
  Serial.println();
  Serial.println("LoRa Simple Node");
  Serial.println("Only receive messages from gateways");
  Serial.println("Tx: invertIQ disable");
  Serial.println("Rx: invertIQ enable");
  Serial.println();

  LoRa.onReceive(onReceive);
  LoRa.onTxDone(onTxDone);
  LoRa_rxMode();
}

void loop() {
  if (runEvery(1000)) { // repeat every 1000 millis

    String message = "HeLoRa World! ";
    message += "I'm a Node! ";
    message += millis();

    LoRa_sendMessage(message); // send a message

    Serial.println("Send Message!");

    display.clearDisplay();
    display.setCursor(0,0);
    display.println("LORA SENDER");
    display.setCursor(0,20);
    display.setTextSize(1);
    display.print("LoRa packet sent.");
    display.setCursor(0,30);
    display.print("Counter:");
    display.setCursor(50,30);
    display.print(counter);      
    display.display();

    counter++;
  
    delay(10000);    
  }
}

void LoRa_rxMode(){
  LoRa.enableInvertIQ();                // active invert I and Q signals
  LoRa.receive();                       // set receive mode
}

void LoRa_txMode(){
  LoRa.idle();                          // set standby mode
  LoRa.disableInvertIQ();               // normal mode
}

void LoRa_sendMessage(String message) {
  LoRa_txMode();                        // set tx mode
  LoRa.beginPacket();                   // start packet
  LoRa.print(message);                  // add payload
  LoRa.endPacket(true);                 // finish packet and send it
}

void onReceive(int packetSize) {
  String message = "";

  while (LoRa.available()) {
    message += (char)LoRa.read();
  }

  Serial.print("Node Receive: ");
  Serial.println(message);
}

void onTxDone() {
  Serial.println("TxDone");
  LoRa_rxMode();
}

boolean runEvery(unsigned long interval)
{
  static unsigned long previousMillis = 0;
  unsigned long currentMillis = millis();
  if (currentMillis - previousMillis >= interval)
  {
    previousMillis = currentMillis;
    return true;
  }
  return false;
}
`

1technophile, 2021-04-02:
Glad to hear it works !

closed 2021-04-02 by botts7