1technophile/OpenMQTTGateway #909: Using CC1101 with pilight works fine, adding somfy nothing is received anymore

AlmightyFrog, 2021-03-28:
I used the exact same hardware about one year or so (early pilight) before, so i was pretty sure it still worked.

So not sure what the issue is, first i thought it is about my hardware when i just flashed the esp32dev-pilight-somfy-cc1101 config but no data was received anymore.

Then i commented out ```'-DZactuatorSomfy="Somfy"'``` and Pilight was working properly again.
Debugging that CC1101 is quite hard still, as there is no good debugging info given from the lib itself - known issue.
Anyhow, when using CC1101 i'd suggest to add a ```ELECHOUSE_cc1101.getCC1101()``` somewhere in code to validate whether connection to CC1101 seems to work, e.g. in setupSomfy() after that ```ELECHOUSE_cc1101.Init();```.
Using following code i could evaluate that CC1101 seems to be recognized correctly:
```
  if (ELECHOUSE_cc1101.getCC1101())
  {
    Log.notice(F("$$$ getCC1101 passed " CR));
  }
  else
  {
    Log.notice(F("$$$ getCC1101 FAILED " CR));
  }
```

I tried both last release and development version - for both it is not working.

Question: Is my use case supported at all? I want to do on CC1101 following:
- Receive and send Pilight
- Send Somfy (seldom enough that i can miss at that time receiving anything else no worries)
- and maybe in future if rtl443 is working stable also receive that

Or can CC1101 only be exclusively used for either pilight or somfy? I tried also rf and rf2 in combination with somfy, but there was same behavior - as soon as i compile somfy code in there is no receive anymore.

To be complete, the ESP32 also does BLE receiving, but i guess that should not cause any interference...

1technophile, 2021-03-30:
@Legion2 could you jump in and provide some idea about this?

Legion2, 2021-03-31:
I'm not sure if I tested the `esp32dev-pilight-somfy-cc1101` configuration. But I also don't know how the somfy module interferes with receiving in pilight, because it only supports sending. @AlmightyFrog didn't the pilight receiving work at all or did it stop working after sending a command with somfy or pilight?

AlmightyFrog, 2021-03-31:
@Legion2 i couldn't test the sending feature, as i currently have no unused device to to test that properly with, but receiving does work.

I can receive data with pilight e.g. like some cheap rf transmitter for power outlet.
It works as long, until i enable in the pio config the somfy module and then it does not receive anything anymore.

Same behavior is there if i use rf instead of pilight, so i guess problem lies somewhere in somfy code.

Legion2, 2021-04-25:
Sorry for the late reply, but I had no time until now. I updated to v0.9.6 and realized that I also use Pilight in combination with Somfy. It works perfectly fine and I can send and receive via Pilight and can also send with somfy.

closed 2022-01-09 by stale[bot]