1technophile/OpenMQTTGateway #910: OpenMQTTGateway_rtl_433 does not show any received pakage

franki29, 2021-03-30:
Before submitting a problem please check the troubleshooting section
https://docs.openmqttgateway.com/upload/troubleshoot.html

**Describe the bug**
OpenMQTTGateway_rtl_433 does not show any received pakage

**To Reproduce**
Steps to reproduce the behavior:
1. Compile OpenMQTTGateway_rtl_433
2. Press any key on a 433 remote
3. See LED Blinking that a signal is receiving
4. Connect to Serial or MQTT Broker
5. No output of the signal or information that a signal is received

**Expected behavior**
Clear message on Seriel or MQTT

**Screenshots**
: Received json : {"uptime":2,"version":"version_tag","freemem":131904,"freestack":4168,"rssi":-58,"XXXXXXXXXX","mac":"24:0A:C4:EF:82:64","mhz":433,"RTLminRssi":-82,"RTLRssi":0,"RTLCnt":0,"modules":["rtl_433"]}␊

**Environment (please complete the following information):**
* OpenMQTTGateway version used (newest development version)
* 

**Additional context**
default_envs = esp32dev-pilight-cc1101 is working correctly

1technophile, 2021-03-30:
Do you have another RF device to test with, it may be that your remote is not compatible.

Also, you can try to activate the [verbose mode ](https://github.com/1technophile/OpenMQTTGateway/blob/f01b7184a02e43184eaa60b0c976a49dd159701c/platformio.ini#L549)

franki29, 2021-04-01:
Hi, yes I tried a differnt RF device with same result.
I also tried to get the status over MQTT but getting:  RTL_433toMQTT = {"Status": "Error"}.
I also tried to compile with:
 '-DRTL_DEBUG=4' 
 '-DRAW_SIGNAL_DEBUG=true'
   '-DOMG_VERSION="$PIOENV"'
   '-DRF_RECEIVER_GPIO=27'
During start I get all available protocols.
But sill nothing  is displayed in serial or MQTT in terms of received signals, I have two LED and one is blinking as soon the CC1101 get a signal.

1technophile, 2021-04-01:
> I also tried to get the status over MQTT but getting: RTL_433toMQTT = {"Status": "Error"}.

I think this is a good hint. Could you try to reverse back to pilight again and verify that this is still working with it. The goal is to eliminate a wiring/hardware defect that could have occurred.

franki29, 2021-04-01:
Yes Pilight-CC1101 still running:
In this case it is a motion sensor that is sending a movement:

PilighttoMQTT = {"message":{"unit":703146,"state":"tamped"},"protocol":"iwds07","length":"","value":"703146","repeats":2,"status":2}

1technophile, 2021-04-01:
Maybe @NorthernMan54 has some pointer to give you?

franki29, 2021-04-01:
I was testing the original code and Example of  NorthernMan54 /rtl_433_ESP and it did not work. But I got a strange message, that at rtl_433_ESP/src/rtl_433_ESP.h the receiver PIN was already set to PIN 4. So I change this to PIN 27 and than it works.
Than I did the same at /OpenMQTTGateway/.pio/libdeps/esp32dev-rtl_433/rtl_433_ESP/src/rtl_433_ESP.h and:
It works:
_433toMQTT = {"model":"LaCrosse-TX141W","id":382461,"channel":0,"battery_ok":1,"wind_avg_km_h":6.1,"wind_dir_deg":0,"test":0,"mic":"CRC","protocol":"LaCrosse TX141-Bv2, TX141T

So I think the default PIN in the source is not overwritten.

NorthernMan54, 2021-04-01:
Looking at the rtl_433_ESP package and I'm not using 'RF_RECEIVER_GPIO' so will I delete the definition from the library and republish.

Give me an hour or 2

NorthernMan54, 2021-04-01:
Pull request #913 clean's up the redundant compiler definitions in the rtl_433_ESP library

NorthernMan54, 2021-04-01:
@1technophile I just added this repo to my watch list, so I should now see these when they come up

closed 2021-04-01 by 1technophile