1technophile/OpenMQTTGateway #914: Part of the RF signal is lost in OMG

sleepwalkera, 2021-04-03:
Trying to use a rf doorbell with OMG, the doorbell is very simple, looks like [this](https://www.lazada.com.my/products/newvoye-v027f-wireless-door-bell-doorbell-visitor-chime-loceng-pintu-i1565008426.html). 

The original signal looks like this.
[image]

Modified RCSwitch:: Protocol and added doorbell protocol, which works well in the rcswitch receiving demo.

`  { 270,  0, { 0, 0 }, 1, {  1, 23 }, { 1,  2 }, { 2, 1 }, false,  0 },  // 01 (VOYE V027F WIRELESS DOOR BELL)`

22:47:39.454 -> Decimal: 14168321 (24Bit) Binary: 110110000011000100000001 Tri-State: not applicable PulseLength: 268 microseconds Protocol: 1
22:47:39.454 -> Raw data: 6179,634,169,632,172,231,566,629,176,627,170,223,580,226,572,231,573,231,574,225,571,632,169,629,174,227,572,230,576,225,571,634,167,227,573,229,576,226,577,221,576,225,576,228,573,229,572,626,181,
22:47:39.454 -> 
22:47:41.250 -> Decimal: 14168321 (24Bit) Binary: 110110000011000100000001 Tri-State: not applicable PulseLength: 268 microseconds Protocol: 1
22:47:41.250 -> Raw data: 6183,629,174,632,169,227,570,633,169,638,164,232,572,228,570,234,568,227,572,230,573,624,179,625,175,232,569,223,575,232,569,625,179,228,574,227,572,223,578,225,576,225,577,231,569,233,566,633,171,
22:47:41.250 -> 
22:47:42.661 -> Decimal: 14168321 (24Bit) Binary: 110110000011000100000001 Tri-State: not applicable PulseLength: 268 microseconds Protocol: 1
22:47:42.661 -> Raw data: 6180,629,171,630,173,231,571,632,171,627,171,229,571,228,575,229,571,226,576,229,571,629,172,631,170,228,572,230,573,229,571,629,173,230,570,228,573,230,572,228,571,232,571,227,572,226,576,628,175,

But in OMG, it does not work properly, when I press the button, whether in Seriel or MQTT, it only has a 1/10 chance to capture the signal in OMG.

22:50:36.771 -> N: Subject: /433toMQTT
22:50:36.771 -> N: Received json : {"value":14168321,"protocol":1,"length":24,"delay":270}
22:50:36.771 -> N: no pub. dupl
22:50:36.771 -> N: Received json : {"value":14168321,"protocol":1,"length":24,"delay":270}
22:50:36.878 -> N: no pub. dupl
22:50:36.913 -> N: no pub. dupl

(This is a log of nearly ten button presses, and only one capture.)

1technophile, 2021-04-03:
> Modified RCSwitch:: Protocol and added doorbell protocol,

Could you point to the repo/version used please

> But in OMG, it does not work properly, when I press the button, whether in Seriel or MQTT, it only has a 1/10 chance to capture the signal in OMG.

Could you try to remove the protocols you don't use and see if you get better repeatability?

sleepwalkera, 2021-04-03:
@1technophile Thanks for the quick reply. I use the latest development version of OMG, have removed all other protocols, but the result is still the same.

1technophile, 2021-04-03:
> have removed all other protocols

I suppose that you are talking about the other RCSwitch protocols isn't it?

When you've done your test with receivedemo, was it the example for the [original](https://github.com/sui77/rc-switch) repo or the [fork ](https://github.com/1technophile/rc-switch)?

sleepwalkera, 2021-04-03:
I removed the protocols, except for the protocol I added in RCSwitch.cpp
I did not install the original RCSwitch version, only the fork was installed.

1technophile, 2021-04-03:
Note also that OMG removes duplicate signals, so if you press 10 times within 3s you will get only one press.

sleepwalkera, 2021-04-03:
I knew that there is a mechanism to deal with duplication, and I did not press the button continuously. The repetition you see is the repetition of the signal itself.

sleepwalkera, 2021-04-06:
@1technophile I want to help explore this issue in depth. Can you give me some hints, such as how to output lower-level logs? Thanks for your patience.

1technophile, 2021-04-11:
You can activate trace in user_config.h, 
https://github.com/1technophile/OpenMQTTGateway/blob/e1aef48296b9205f57a6d3a3a4bef2e8a52a6ff7/main/User_config.h#L378

Set it to LOG_LEVEL_TRACE

sleepwalkera, 2021-04-12:
@1technophile I turned on LOG_LEVEL_TRACE and ran the same test. When a problem occurred, I still couldn't capture any logs. I recorded two short [videos](https://www.bilibili.com/video/BV1t54y1b7js). The first 30 seconds of the video was tested on OMG, and only the last time the signal was captured. The following is the receivedemo test, every time the signal is captured.

sleepwalkera, 2021-04-12:
OMG log looks like this

21:12:05.044 -> T: Rcv. RF
21:12:05.044 -> T: RF Task running on core :1
21:12:05.044 -> T: isAdupl?
21:12:05.044 -> N: Subject: /433toMQTT
21:12:05.044 -> N: Received json : {"value":14168321,"protocol":1,"length":24,"delay":200}
21:12:05.044 -> T: jsonPublishing
21:12:05.044 -> T: Store val: 14168321
21:12:05.078 -> T: Min ind: 1
21:12:05.078 -> T: store code : 14168321 / 20558
21:12:05.078 -> T: Col: val/timestamp
21:12:05.078 -> T: mem code : 14168321 / 8679
21:12:05.078 -> T: mem code : 14168321 / 20558
21:12:05.078 -> T: mem code : 0 / 0
21:12:05.078 -> T: mem code : 0 / 0
21:12:05.078 -> T: mem code : 0 / 0
21:12:05.078 -> T: mem code : 0 / 0
21:12:05.078 -> T: mem code : 0 / 0
21:12:05.078 -> T: mem code : 0 / 0
21:12:05.078 -> T: mem code : 0 / 0
21:12:05.078 -> T: mem code : 0 / 0
21:12:05.078 -> T: mem code : 0 / 0
21:12:05.078 -> T: mem code : 0 / 0
21:12:05.078 -> T: Hey I got a callback 
21:12:05.078 -> T: isAdupl?
21:12:05.117 -> N: no pub. dupl
21:12:05.117 -> N: Received json : {"value":14168321,"protocol":1,"length":24,"delay":200}
21:12:05.117 -> T: Rcv. RF
21:12:05.117 -> T: RF Task running on core :1
21:12:05.117 -> T: isAdupl?
21:12:05.117 -> N: no pub. dupl

1technophile, 2021-04-12:
Thanks for sharing these infos, could you add your user_config.h without credentials and the board used ?

sleepwalkera, 2021-04-14:
@1technophile This is my [user_config.h](https://pastebin.com/raw/n1JXWYCr), based on 0.9.6 release

1technophile, 2021-04-15:
Seems that I will need to test with your protocol and a sender + a receiver.

Will try to reproduce

1technophile, 2021-04-24:
Tried to reproduce your issue,
I replaced all the protocol  by yours on one sender (ESP8266).
Put OMG on an ESP32 + an RF receiver with all the protocol (modified the number 17 to a delay of 270 (to equal your definition)
Sent several MQTT commands from the ESP8266:

`{"value":14168321,"protocol":1,"length":24,"delay":270}`

Each time the command is caught by the ESP32 (published or not depending if it is considered as a duplicate)

If you press your doorbell with a longer time does it gets recognised each time?

sleepwalkera, 2021-04-25:
@1technophile Thank you for your patience. I tried pressing for a longer time, but the result was the same. What I am currently doing is also using other RF transmitters to reproduce this problem and compare whether the actual signal sent is the same as the doorbell signal.

closed 2022-01-09 by stale[bot]