1technophile/OpenMQTTGateway #917: Compiler warning when build with PIO, redefine of struct_size in main.ino

AlmightyFrog, 2021-04-05:
It is not nice to see a warning at each fresh build, but it seems to be a wrong warning...

**Describe the bug**
It seems that there is something strange ongoing at converting that main.ino to main.ino.cpp giving a warning:

```
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (3.1.1) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 3.10005.210308 (1.0.5) 
 - tool-esptoolpy 1.30000.201119 (3.0.0) 
 - toolchain-xtensa32 2.50200.97 (5.2.0)
Converting main.ino
/mypath/OpenMQTTGateway-REF/main/main.ino:41:0: warning: "struct_size" redefined
 #    define struct_size 4
 ^
/mypath/OpenMQTTGateway-REF/main/main.ino:38:0: note: this is the location of the previous definition
 #    define struct_size 12
 ^
```

Here you can see the corresponding code after converting, which seems totally fine to me:
```
# if defined(ESP8266) || defined(ESP32) || defined(__AVR_ATmega2560__) || defined(__AVR_ATmega1280__)
#define struct_size 12
ReceivedSignal receivedSignal[struct_size] = {{0, 0}, {0, 0}, {0, 0}, {0, 0}, {0, 0}, {0, 0}, {0, 0}, {0, 0}, {0, 0}, {0, 0}, {0, 0}, {0, 0}};
# else
#define struct_size 4
ReceivedSignal receivedSignal[struct_size] = {{0, 0}, {0, 0}, {0, 0}, {0, 0}};
# endif
#endif
```
It pretty much just removes the comments and some excessive whitespaces.

**To Reproduce**
Just type `pio run` and when it is compiling look at the output.
If you want to compare the original main.ino with the temporary main.ino.cpp just make sure to save or use it until it is remved when compile is finished.

I directly opened it with a diff tool like `meld main.ino main.ino.cpp` in another shell.

**Environment (please complete the following information):**
Latest version development.

dkneisz, 2021-05-06:
Looks like this is more a PIO issue:
https://community.platformio.org/t/warning-symbol-redefined-probably-because-of-fpreprocessed/6362

NorthernMan54, 2021-05-07:
This is a potential workaround

```
#    ifdef struct_size
#      undef struct_size
#    endif
```

ie

```
#    define struct_size 12
ReceivedSignal receivedSignal[struct_size] = {{0, 0}, {0, 0}, {0, 0}, {0, 0}, {0, 0}, {0, 0}, {0, 0}, {0, 0}, {0, 0}, {0, 0}, {0, 0}, {0, 0}};
#  else // boards with smaller memory
#    ifdef struct_size
#      undef struct_size
#    endif
#    define struct_size 4
ReceivedSignal receivedSignal[struct_size] = {{0, 0}, {0, 0}, {0, 0}, {0, 0}};
#  endif
```

PR #1010 (2021-07-06): Cleanup compiler warning.
## Description:
Resolves #917

## Checklist:
  - [x] The pull request is done against the latest development branch
  - [x] Only one feature/fix was added per PR and the code change compiles without warnings
  - [x] I accept the [DCO](https://github.com/1technophile/OpenMQTTGateway/blob/development/docs/participate/development.md#developer-certificate-of-origin).
commit 2ae8f0c0 (2021-07-06): Cleanup compiler warning. (#1010)

Resolves #917
closed 2021-07-06 by 1technophile