1technophile/OpenMQTTGateway #921: Somfy Beta

hermanndererdmann, 2021-04-10:
When using the Somfy config i can‘t get the gateway to work.

Did anyone encounter the same problems?

ecoenjaerts, 2021-04-16:
Yes...the same issue here. I have tried it with both ESP32 and NodeMCU, but to no avail.

ecoenjaerts, 2021-04-17:
Found in the code that ZradioC1101 needs to be enabled for the C1101 library to be used. There is some difference in connections as described in the OpenMQTTGateway Documentation compared to the SmartRC-CC1101-Driver-Lib documentation. The OpenMQTTGateway Documentation e.g. mentions to connect C1101 GDO2 to ESP32 raw pin number 27 (which corresponds to GPIO 17). The SmartRC-CC1101-Driver-Lib documentation however mentions to connect GDO2 to ESP32 GPIO4). 

The connection diagram of the C1101 Driver library has been working for me when using the C1101 example that is included in the Somfy_Remote_Lib

closed 2022-01-09 by stale[bot]