1technophile/OpenMQTTGateway #922: Unavailable values after home assistant restart - retain flag?

thehijacker, 2021-04-12:
**Is your feature request related to a problem? Please describe.**

All the messages from my temperature sensors arrive in topic:

esp32/OpenMQTTGateway_ESP32_BLE/BTtoMQTT

When I restart Home Assistant initial value of sensors is unavailable. It should be last read value before restart.

**Describe the solution you'd like**

Values after Home Assistant restart should be last read value and not unavailable. Now they are back to value only when they are first read after Home Assistant restart.

**Describe alternatives you've considered**

I am guessing that there should be a retain flag on the messages. Correct me if I am wrong.

Thank you.

1technophile, 2021-04-12:
> I am guessing that there should be a retain flag on the messages. Correct me if I am wrong.

If we do this you will get a value with a timestamp equal to the reconnection to the broker = restart. This will not represent the truth of a sensor value.

Is it not a home assistant issue rather than an OMG one?

1technophile, 2021-04-22:
Closing this, feel free to reopen if needed

closed 2021-04-22 by 1technophile