1technophile/OpenMQTTGateway #923: Add PJON compatibility

gioblu, 2021-04-14:
PJON could enable [single wire communication](https://github.com/gioblu/PJON/tree/master/src/strategies/SoftwareBitBang) for up to 255 devices over up to 2000m, as well as a very cheap [laser or optic-fibre link](https://github.com/gioblu/PJON/tree/master/src/strategies/AnalogSampling).

PJON is a network protocol with a wide data-link support, and there is no alternatives which implement new data links with comparable performance.

I noticed @1technophile starred PJON and so digging in his account I found this beautiful project. 
Keep up the good work and compliments for what you have built.

1technophile, 2021-04-14:
Hi,

Pjon is very interesting, thank you for the public contribution ! 

Outside of the wire capabilities how do you think we could use it in OMG  ?

gioblu, 2021-04-14:
@1technophile the router classes: https://github.com/gioblu/PJON/blob/master/documentation/routing.md may be interesting for you being rather simple and portable.

closed 2021-04-29 by gioblu