1technophile/OpenMQTTGateway #925: No auto-discoverd sensors in HomeAssistant for rtl_433-devices?

hvorragend, 2021-04-14:
Before submitting a problem please check the troubleshooting section
https://docs.openmqttgateway.com/upload/troubleshoot.html

**Describe the bug**
I have flashed the current Git code on an ESP32 with platform IO.
The ESP32/CC1101 is working fine and I see a lot of payloads in:  home/OpenMQTTGateway_rtl_433_ESP/RTL_433toMQTT
Several system entities were automatically created in Home Assistant: 
- SYS: Auto discovery (on)
- SYS: Free memory 
- and many SYS-entities more

I am having the following payload in home/OpenMQTTGateway_rtl_433_ESP/RTL_433toMQTT:
```
{
  "model" : "TFA-Drop",
  "id" : 983179,
  "battery_ok" : 1,
  "rain_mm" : 80.01,
  "mic" : "CHECKSUM",
  "protocol" : "TFA Drop Rain Gauge 30.3233.01",
  "rssi" : -56,
  "duration" : 441749
}
```

But this device/entity wasn't created in Home Assistant.

**To Reproduce**
Please see above

**Expected behavior**
I thought that all sensore were automatically created in Home Assistant.

**Screenshots**
none

**Environment (please complete the following information):**
* OpenMQTTGateway version used: current development code (some commits after v0.9.6)
* Library used from release page

1technophile, 2021-04-14:
Not for the moment. This is something that we are discussing below if you want to join the discussion
https://community.openmqttgateway.com/t/mqtt-discovery-message-hampton-bay-fans/1485?u=1technophile

hvorragend, 2021-04-14:
Thank you for the answer.

closed 2021-04-14 by hvorragend