1technophile/OpenMQTTGateway #933: Model of BLE sensor is incorrectly reported as LYWSD03MMC

kelchm, 2021-04-27:
**Describe the bug**
Using a collection of CGG1 (M version), MHO-C401 and LYWSD03MMC BLE sensors running PVVX firmware, **all** devices are incorrectly shown with a model of `LYWSD03MMC_XXX` (with XXX being either ATC or PVVX, depending on the selected advertisement format).

**To Reproduce**
1. Setup MHO-C401 / CGG1 devices running 
2. Observe that the model field within received MQTT messages is incorrect.

**Expected behavior**
The reported model should either be generic (if the required data is not included in the BLE advertisement) or report the actual model of the device.

**Screenshots**
[image]

**Environment (please complete the following information):**
* OpenMQTTGateway: v0.9.6
* Device Firmware: PVVX firmware v3.0 / v3.1 (using either ATC or PVVX advertising format)

1technophile, 2021-04-28:
Thanks for pointing this, is it possible to set this to [true](https://github.com/1technophile/OpenMQTTGateway/blob/4d49f5b08ee45bb10bbdb2ead2788dedf147dc7e/main/config_BT.h#L101) and paste here your broker extracts (litteral json of each sensor so that I can copy/paste) please?
It would help refine the distinction criteria between the different models.

kelchm, 2021-04-30:
**MHO-C401**
```
{
  "id": "A4:C1:38:EC:DE:DF",
  "name": "MHO_ECDEDF",
  "rssi": -81,
  "servicedata": "dfdeec38c1a4ee08c612260a2c5704",
  "model": "LYWSD03MMC_PVVX",
  "tempc": 22.86,
  "tempf": 73.148,
  "hum": 48.06,
  "batt": 44,
  "volt": 2.598
}
```

**MHO-C401**
```
{
  "id": "A4:C1:38:9D:D4:74",
  "name": "MHO_9DD474",
  "rssi": -83,
  "servicedata": "74d49d38c1a4a808c012b60a3c7004",
  "model": "LYWSD03MMC_PVVX",
  "tempc": 22.16,
  "tempf": 71.888,
  "hum": 48,
  "batt": 60,
  "volt": 2.742
}
```

I've also now realized that my CGG1's flashed with PVVX firmware were not being picked up as sensors at all.

CGG1 (M)
```
{
  "id": "82:00:12:34:2D:58",
  "name": "CGG_342D58",
  "rssi": -28,
  "servicedata": "5030480b67582d341200820a10015902bd0b"
}
```

CGG1 (M)
```
{
  "id": "6A:12:34:2D:58:5A",
  "name": "CGG_2D585A",
  "rssi": -87,
  "servicedata": "5a582d34126a38081513da0b5c0304"
}
```

1technophile, 2021-07-11:
>   "model": "LYWSD03MMC_PVVX"

Maybe it makes sense to replace this with "model": "PVVX_FIRMWARE" as the model is available in the name already, thoughts?

> I've also now realized that my CGG1's flashed with PVVX firmware was not being picked up as sensors at all.

Can you try this on line 1248 ?:
```c
      if (service_len >= 30 && (strncmp(service_data + 6, "38c1a4", 6) == 0 || strncmp(service_data + 6, "0b6758", 6) == 0)) {
```

closed 2022-04-16 by stale[bot]