1technophile/OpenMQTTGateway #944: Enable Dependabot security updates

Legion2, 2021-05-13:
**Is your feature request related to a problem? Please describe.**
I'm always frustrated when I get Dependabot alerts for my OpenMQTTGateway fork, which are not fixed upstream

**Describe the solution you'd like**
Enable Dependabot security updates which automatically create Pull Requests to update the dependencies.

**Describe alternatives you've considered**
Create Pull Request to update dependencies manually.

1technophile, 2021-05-29:
Would be interesting. Do you want to set it up ?

Legion2, 2021-05-30:
I don't have the permission on this repository to do so. @1technophile It only requires three clicks https://docs.github.com/en/code-security/supply-chain-security/managing-vulnerabilities-in-your-projects-dependencies/configuring-dependabot-security-updates#enabling-or-disabling-dependabot-security-updates-for-an-individual-repository

1technophile, 2021-05-30:
Thanks, done!

closed 2021-05-30 by 1technophile