1technophile/OpenMQTTGateway #951: bme280 not working

pauldogg, 2021-05-17:
Before submitting a problem please check the troubleshooting section
https://docs.openmqttgateway.com/upload/troubleshoot.html

**Describe the bug**
A clear and concise description of what the bug is.
Using a heltec esp32 v2  with lora and oled, i've added a bme280 and the serial reports:
N: Bosch BME280 Initialized - Result of .begin(): 0x 

**To Reproduce**
Steps to reproduce the behavior:
1. connect bme280 (i tried 3 different sensors)
2. flash module with omg with BME280 enabled (i tried address 0x77 and 0x76)
3. See error in serial console

**Expected behavior**
Address identified, values reported

**Screenshots**
N: Bosch BME280 Initialized - Result of .begin(): 0x

**Environment (please complete the following information):**
* OpenMQTTGateway version used V0.9.6
* Library version related to the problem you have (if you have troubles with RF provide the version of RCSwitch library) bme280.h

**Additional context**
Add any other context about the problem here.

NorthernMan54, 2021-05-17:
To assist in troubleshooting, can you share your wiring diagram

pauldogg, 2021-05-17:
Sure.
i'm using these: https://www.amazon.com/gp/product/B01N4EHIW6

mcu      bme
3.3v -->vcc
21----->SDA
22----->SCL

I also have a TSL2561 on the same bus and that one works fine.
Thanks

pauldogg, 2021-05-17:
Wrong sensor bmp instead of bme derp

closed 2021-05-17 by pauldogg

spacemanspiff2007, 2021-05-28:
I'm having a similar issue

Connection
```
3.3v -> VIN
21   -> SDA
22   -> SCL
```

Output on start
```text
************* WELCOME TO OpenMQTTGateway **************
N: OpenMQTTGateway Version: version_tag
[E][WiFiMulti.cpp:184] run(): [WIFI] Connecting Failed.
N: WiFi ok with manual config credentials
N: Setup BME280 on adress:
N: Bosch BME280 Initialized - Result of .begin(): 0x
N: IR_EMITTER_GPIO: 14
N: IR_RECEIVER_GPIO: 26
N: OpenMQTTGateway modules: ["BME280","IR"]
N: ************** Setup OpenMQTTGateway end **************
```

What I don't understand is why the adress of the BME does not get printed on startup.
Since it's a constant it should always work.

This is what gets published:
```
N: Subject: /CLIMAtoMQTT/bme
N: Received json : {"tempc":0,"tempf":32,"hum":0,"pa":0,"altim":44330.77,"altift":145442.2}
```

mcmlxxvii, 2024-09-24:
I'm facing the exact same issue with a BME280 and Heltec v2. I'm using this one: https://www.amazon.ca/dp/B07KYJNFMD 

That sensor is an I2C only version that works both in 3.3v and 5v, as it has a built-in voltage regulator. Mine is connected to 3.3v though.

I tested 3 different sensors (using both 0x76 and 0x77 addresses) with my Heltec v2 board, using the recommended pinout: 2 of the [aforementioned model](https://www.amazon.ca/dp/B07KYJNFMD) and 1 of a [similar model](https://www.amazon.ca/dp/B0CXF89ZYQ) used by @pauldogg, which has 6 pins and is a hybrid I2C/SPI version. All of them show the same behavior: _Bosch BME280/BMP280 failed: 0x0_ and weird readings, preceded by _[E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263_ error message in the serial console.

I tested all those sensors with an ESP-WROOM-32 board using the exact same wiring, voltage and pinout. For that, I leveraged an I2C Scanner Sketch to make sure of the sensor's address and a BME280 Testing Sketch using both SparkFun (2.0.9) and Adafruit (2.2.4) libraries in Arduino IDE.

Everything worked as expected using the ESP-WROOM-32. However, when trying the exact same sensors with a Heltec v2 running OMG heltec-rtl_433-fsk compiled via Gitpod.io, they don't work.

You can see some console logs illustrating the issue below:

```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:13160
load:0x40080400,len:3036
entry 0x400805e4
N:
************* WELCOME TO OpenMQTTGateway **************
N: SYS config not found
N: SSD1306 config initialised
N: Saved SSD1306 config loaded
N: Setup SSD1306 Display end
N: OpenMQTTGateway Version: 20240918-01
N: OTA server cert hash: 02118cbaf5cc02748398d93ecdb0ef2a444c37358a617bfb2206dfe                                                 351dc7cf2
N: Config loaded from flash
N: Attempting Wifi connection with saved AP: 0
N: No WebUI config to load
N: OpenMQTTGateway URL: http://[redacted by me]/
N: ZwebUI setup done
N: RF Config not found using default
N: Enable RTL_433 Receiver: 915.00Mhz
N: Setup BME280/BMP280 on address: 0x76
N: Setup BME280/BMP280 on SDA pin: 0x15 ----> Added by me for debugging
N: Setup BME280/BMP280 on SCL pin: 0x16 ----> Added by me for debugging
[  8943][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
N: Bosch BME280/BMP280 failed: 0x0
N: ZgatewayRTL_433 setup done
N: Offline disabled
N: OpenMQTTGateway modules: ["HELTEC_SSD1306","WebUI","BME280","rtl_433"]
N: ************** Setup OpenMQTTGateway end **************
N: Reconfiguring MQTT client...
N: Connected to broker
N: [ OMG->MQTT ] topic: home/OMG_heltec_rtl_433_ESP_FSK/LWT msg: online
```

```
[462430][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
[463432][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
[464433][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
[465434][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
[466435][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
[467436][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
N: [ OMG->MQTT ] topic: home/OMG_heltec_rtl_433_ESP_FSK/CLIMAtoMQTT/bme msg: {"tempc":-3.4,"tempf":25.88,"hum":0,"pa":0,"altim":44330.77,"altift":145442.2}
```
Hope the above information helps! I'm having a hard time trying to figure out what's going on by myself - This is driving me nuts. :)

Thanks!